Freetz-NG/freetz-ng #119: [compile failed] ncurses 6.2 wont compile

harryboo, 2020-11-21:
Hi,

the package 'ncurses' is needed when selecting the package 'lcd4linux' but it wont compile.
To reproduce the problem just select '7490', 'Firmware 6.9x' and the package 'lcd4linux' (or use the attached config).
[minimal.config.txt](https://github.com/Freetz-NG/freetz-ng/files/5577690/minimal.config.txt)

Here is also a log for the package 'ncurses' (error at building 'terminfo database'):
[ncurses_log.txt](https://github.com/Freetz-NG/freetz-ng/files/5577692/ncurses_log.txt)

-Clean checkout
-NO modifications

Thanx

fda77, 2020-11-21:
This works fine for me

```
"terminfo.tmp", line 4901, col 25, terminal 'xterm+indirect': limiting value of `colors' from 0x1000000 to 0x7fff
"terminfo.tmp", line 4901, col 40, terminal 'xterm+indirect': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 7837, col 36, terminal 'dvtm-256color': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4791, terminal 'xterm-16color': error writing /home/jens/freetz-ng-org/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/share/terminfo/x/xterm-16color
```
Is you hdd full?

harryboo, 2020-11-21:
I am using a virtual machine (vmware workstation):

jens@jens-virtual-machine:~$ cat /proc/version
Linux version 4.15.0-123-generic (buildd@lcy01-amd64-017) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #126-Ubuntu SMP Wed Oct 21 09:40:11 UTC 2020

there is enought diskspace available:

jens@jens-virtual-machine:~$ df
Dateisystem    1K-Blöcke  Benutzt Verfügbar Verw% Eingehängt auf
udev             1985280        0   1985280    0% /dev
tmpfs             401596     1852    399744    1% /run
/dev/sda1       61663020 27294012  31207000   47% /
tmpfs            2007960        0   2007960    0% /dev/shm
tmpfs               5120        4      5116    1% /run/lock
tmpfs            2007960        0   2007960    0% /sys/fs/cgroup
tmpfs             401592       28    401564    1% /run/user/1000

What do you use for building the image?

fda77, 2020-11-21:
> Dateisystem 1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
Warum schreiben wir auf englisch?

Es kann sein dass dein "terminfo entry-description compiler" zu alt ist (sieh Link oben).
Mit Fedora 32:
```
$ tic -V
ncurses 6.1.20191109
```

Ein Update von Ubuntu 18.04 auf 20.04 oder 20.10 könnte den Fehler "behebt".

harryboo, 2020-11-21:
Danke für den Link in Beitrag 2.
Habe daraufhin Ubuntu 20.04 installiert und es compiliert wieder.

Mann sollte die Version von 'tic' prüfen, wenn das Paket 'ncurses' selektiert wurde.

Danke nochmals.

fda77, 2020-11-21:
Es gibt soweit ich weiss keinen Mechanismus in Freetz um einen bestimmte Version zu prüfen, nur ob etwas installiert ist.
Welche Version von tic bzw ncurses hattest du denn instaliert?

harryboo, 2020-11-22:
Mit dieser ging es nicht: ncurses 6.1.20180127

fda77, 2020-11-23:
Es wird jetzt ein eigenes ncurses/tic gebaut das die gleiche Version wie für die Fritzbox ist:
```
$ tools/tic -V
ncurses 6.2.20200212
```

closed 2020-11-23 by fda77