Freetz-NG/freetz-ng #120: 7590 mit Labor Bootet nicht mehr

MasterRoCcO, 2020-11-26:
Ich habe mir ein Image für die 7590 mit der Labor gebaut und Box fährt nicht mehr hoch.

fda77, 2020-11-26:
Untrustedd entfernen oder watchdog deaktivieren

-> https://github.com/Freetz-NG/freetz-ng/issues/28

MasterRoCcO, 2020-11-26:
Es geht hier um die 7.24 FW und es ist egal ob watchdog an oder aus ist sie Bootet nicht mehr. Sie blinkt nur und nichts geht mehr

fda77, 2020-11-26:
Wenn es keinerlei Logs gibt ist es schwer zu sagen wo es hängt.
7490 läuft jedenfalls mit Labot

MasterRoCcO, 2020-11-26:
Aber die Box fähr nun mal nicht hoch wie soll ich da an logs kommen.

MasterRoCcO, 2020-11-26:
So weit wie ich das weiß hast ja selber auch eine 7590 teste es doch selber mal

PrinzVonBillAir, 2020-11-27:
Ich habe es gerade mal bei mir getestet.
Hier meine config, welche bei 07.21 funktioniert (siehe Bild), bei 07.24 Labor jedoch nicht. Als Pakete ausgewählt: dnsmasq und WOL, abgewählt, dropbear und authkeys.
Wie beim Threadersteller auch, bleibt die Box während dem Booten hängen und ist nicht erreichbar und nur mittels FS-Wechsel wieder gangbar zu machen.

[image]

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/5605379/config.txt)

MasterRoCcO, 2020-11-27:
Danke @PrinzVonBillAir fürs testen, aber wie sagt fda es ist ein duplicate. Seit der FW 7.19 kann man die 7490 nicht mehr mit der 7590 vergleichen. Musl sei dank

fda77, 2020-11-27:
7490:
Wer die nutzt sollte bei AVM um aktuelle Sourcen per eMail betteln! Da kommt vermutlich irgendwas wie "ach, warten sie doch bis zum Release" blah etc zurück. Dies dankendablehnen! Wenn der AVM keine Lust hat aktuellen Quellcode zu veröffentlichen, sollen er seine Labors auch behalten
```
unresolved symbol avm_pa_dev_register in file /lib/modules/3.10.107/kernel/drivers/dsld/kdsldmod.ko
WARNING: Unresolved symbols detected, not all AVM-features may work.
No current sources by AVM? Error in kernel's .config?
```

7590:
Am besten lötet jemand ne serielle console an seine Box, dann sieht man genau was das Problem ist
 - ich würde disable-watchdog auf jeden Fall aktivieren
 - dann einmal mit und einmal ohne remove-untrustedd testen
 - keine dnsmasq im image, dies belegt port 53 (DNS) was Probleme mit dem multid verusachen kann
Wenn ein Image nicht bootet, kann man danach über den Bootloader die alte Firmware wieder aktivieren oder über den Bootloader ein gutes Image flashen. ABER KEIN RECOVERY. Mit Glück findet man dann in den crash/debug Logs noch alte Meldungen

MasterRoCcO, 2020-11-27:
Du hast doch seber eine und bringst Freetz-ng raus
also löte du es doch ein. Und du hast es schon in deiner 7490. Nicht jeder user kann löten so wie du

Das sagte schon @PrinzVonBillAir

berndy2001, 2020-11-27:
Ist mir auf er 7590 auch letztens aufgefallen.
"Disable AVM watchdog" war gewählt; "remove-untrustedd" war nicht gewählt. Kein Dnsmasq.

MasterRoCcO, 2020-11-27:
Aber das 7.25 wird kommen und 100000 schreihen. Weil die 7590 nicht Hochfährt

fda77, 2020-11-27:
> Du hast doch seber eine und bringst Freetz-ng raus
> also löte du es doch ein.

Bin ich deine Mama?

> Aber das 7.25 wird kommen und 100000 schreien. Weil die 7590 nicht Hochfährt

Die Labor für die 7590 kann auch einfach wieder entfernt werden

MasterRoCcO, 2020-11-27:
Ich bin nicht dein PaPa

mach es mal lieber.

fda77, 2020-11-27:
Du langweilst mich und verschwendest meine Zeit. Entweder du lässt es oder ich muss nachhelfen

MasterRoCcO, 2020-11-27:
Das war klar das das kommt. Wie immer.

Kümmern dich um deine Fehler in freetz-ng und schieb es nicht auf andere

⁣BlueMail for Android herunterladen ​

fda77, 2020-11-27:
Ich hab wirklich keine Lust mich hier von dir nerven zu lassen

fda77, 2020-11-27:
Eine Idee von @PeterPawn: Evtl verbraucht Freetz zuviel "Zufälligkeit" weshalb der untrustedd nicht richtig läuft. Zum überprüfen:
 - 7590 mit recovery zurücksetzen
 - kein Backup zurückspielen!!
 - Grundlegende Konfiguration
 - Freetz mit Labor über Bootloader einspielen, untrustedd nicht entfernt, watchdog kann drin bleiben

cnbr, 2020-12-21:
Da meine 7590 heute beim Starten mit der Labor 7.24 hing, habe ich mal über die serielle Console ein Log gezogen und anschließend wieder die vorher funktionsfähige Partition aktiviert.
Aufgefallen ist, die Log stoppt  ca 7 Sekunden nach Boot im Bereich `random: init: uninitialized urandom read (4 bytes read)`
Eine Meldung `random: crng init done` kommt nach ca 32 Sekunden, bei der funktionierenden 7.21  kommt die schon ca. 0,5 Sekunden später.  Nach  Eingabe von Enter an der Console wird ab und an  `Waiting for entropy` ausgegeben. Nach über 2 Mit Wartezeit ohne weitere Ereignisse habe ich dann abgebrochen und das funktionsfähige Image wieder aktiviert.
Logs von der 7.24 und der funktionsfähigen 7.21 sind mit dabei, vielleicht hilft es ja bei der Diagnose.
[boot_7590_07.24.all-rev84707_labor_freetz-ng-17597.log](https://github.com/Freetz-NG/freetz-ng/files/5725969/boot_7590_07.24.all-rev84707_labor_freetz-ng-17597.log)
[boot_7590_07.21.all_freetz-ng-17273.log](https://github.com/Freetz-NG/freetz-ng/files/5725970/boot_7590_07.21.all_freetz-ng-17273.log)

fda77, 2020-12-21:
Danke!!
Auszüge aus deinen Logs:

```
[    0.000000] [fw-info] Version 07.24-84707

[    6.536963] [announce_root] filesystem (/dev/mtdblock6) will be used as root device
[    6.554939] VFS: Mounted root (squashfs filesystem) readonly on device 31:6.
[    6.566991] devtmpfs: mounted
[    6.569711] Freeing unused kernel: 372k freed
[    6.572907] This architecture does not have kernel memory protection.
[    7.014812] random: init: uninitialized urandom read (4 bytes read)
[    7.304165] random: 1: uninitialized urandom read (4 bytes read)
[    7.498423] random: head: uninitialized urandom read (4 bytes read)

[   10.345035] [0]system-load  49% loadavg 0.8 0.2 0.1 - pgstat: sum=217155 free=102303 slab=1837 alloc=770/s fault=105/s (sleep 1)
[   32.588305] 	
[   32.590312] random: 7 urandom warning(s) missed due to ratelimiting
[   50.688115] start slab_allocator-trace now  (use cat /proc/slab_allocators)

[32m[INFO][0m Waiting for entropy
[32m[INFO][0m Waiting for entropy
```

```
[    0.000000] [fw-info] Version 07.21

[    6.419887] [announce_root] filesystem (/dev/mtdblock5) will be used as root device
[    6.440305] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
[    6.453108] devtmpfs: mounted
[    6.455746] Freeing unused kernel: 388k freed
[    6.459032] This architecture does not have kernel memory protection.
[    6.899343] random: init: uninitialized urandom read (4 bytes read)
[    7.036616] random: 1: uninitialized urandom read (4 bytes read)
[    7.228091] random: head: uninitialized urandom read (4 bytes read)

[    7.465082] eip123_hw: loading out-of-tree module taints kernel.
[    7.469743] eip123_hw: module license 'Proprietary' taints kernel.
[    7.475929] Disabling lock debugging due to kernel taint
[    7.481658] [module-mem] give 0x4000 bytes at 884a8000 to module 'eip123_hw' (0x98a000 bytes left)
[    7.492344] [module-mem] give 0x2000 bytes at 884ac000 to module 'eip123' (0x988000 bytes left)
[    7.500970] eip123 1e000000.eip123: Detected hardware version 2.2.2
[    7.506069] eip123 1e000000.eip123: my_id=3 master_id=0 num_mailboxes=4 mailbox_size=256
[    7.514125] eip123 1e000000.eip123: Linked mailbox number 1
[    7.544405] eip123 1e000000.eip123: Sysinfo: FWVersion=2.4.2 HWVersion=2.2.2 MemorySize=32768 HostId=3 Identity=0 NVMAnomaly=0 NVMAnomalyLocation=0
[    7.776456] eip123 1e000000.eip123: eip123_hw init successful
[    7.809166] random: crng init done
[    7.811109] random: 7 urandom warning(s) missed due to ratelimiting
[    7.836475] eip123 1e000000.eip123: Device responded with error_src=0x0 error=0x1 opcode=0x1c
[    7.843608] eip123 1e000000.eip123: Could not find rootkey in OTP
```

Die eip*ko gibt es in der Labor auch, "modinfo" von diesen Dateien 
```
filename:       lib/modules/4.9.198/extra/eip123_hw.ko
description:    Hardware access functions for the eip123 driver.
author:         Alexander Theißen <a.theissen@avm.de>
license:        Proprietary
depends:
vermagic:       4.9.198 SMP mod_unload MIPS32_R2 32BIT

filename:       lib/modules/4.9.198/extra/eip123.ko
description:    Driver for the eip123 crypto chip.
author:         Alexander Theißen <a.theissen@avm.de>
license:        GPL v2
alias:          of:N*T*Cverimatrix,eip123C*
alias:          of:N*T*Cverimatrix,eip123
depends:
vermagic:       4.9.198 SMP mod_unload MIPS32_R2 32BIT

filename:       lib/modules/4.9.198/extra/eip97.ko
description:    Cryptographic accelerator driver for EIP97
author:         Robert Hering <r.hering@avm.de>
license:        GPL
alias:          of:N*T*Clantiq,crypto-xrx500C*
alias:          of:N*T*Clantiq,crypto-xrx500
depends:
vermagic:       4.9.198 SMP mod_unload MIPS32_R2 32BIT
parm:           eip97_desc_num:uint
```

Es könnte also so sein wie PeterPawn schrieb, dass zuwenig Zufälligkeit vorhanden ist. So früh dürften Daemons von Freetz damit noch nichts zu tun haben. Könnte auch ein Fehler in der busybox-config sein. Oder was mit dem "tty", ob das für das hw-crypto überhaupt genutzt wird weiss ich aber nicht..
Kannst du noch ein Log von der Labor-ohne-Freetz machen? PeterPawn vermutet dass so nötige Dateien generiert werden können und danach evtl die Labor auch mit Freetz starten kann

cnbr, 2020-12-23:
Log von der 07.24-84940 ist anbei, der Bereich um random sieht nun etwas anders aus, ` random: init:` fehlt z.B. vermutlich wird der Zufallsgenerator nun anders initialisiert, 

```
[    6.515359] devtmpfs: mounted
[    6.518067] Freeing unused kernel: 372k freed
[    6.521281] This architecture does not have kernel memory protection.
[    7.809150] eip123_hw: loading out-of-tree module taints kernel.
[    7.813925] eip123_hw: module license 'Proprietary' taints kernel.
[    7.819921] Disabling lock debugging due to kernel taint
[    7.825731] [module-mem] give 0x4000 bytes at 884a8000 to module 'eip123_hw' (0x9ac000 bytes left)
[    7.836712] [module-mem] give 0x2000 bytes at 884ac000 to module 'eip123' (0x9aa000 bytes left)
[    7.845266] eip123 1e000000.eip123: Detected hardware version 2.2.2
[    7.850278] eip123 1e000000.eip123: my_id=3 master_id=0 num_mailboxes=4 mailbox_size=256
[    7.858335] eip123 1e000000.eip123: Linked mailbox number 1
[    7.888905] eip123 1e000000.eip123: Sysinfo: FWVersion=2.4.2 HWVersion=2.2.2 MemorySize=32768 HostId=3 Identity=0 NVMAnomaly=0 NVMAnomalyLocation=0
[    7.924883] eip123 1e000000.eip123: Device responded with error_src=0x2 error=0x2 opcode=0x11
[    8.232884] eip123 1e000000.eip123: eip123_hw init successful
[    8.265309] random: crng init done
[    8.292899] eip123 1e000000.eip123: Device responded with error_src=0x0 error=0x1 opcode=0x1c
[    8.299988] eip123 1e000000.eip123: Could not find rootkey in OTP
[    8.581626] [module-mem] give 0x33000 bytes at 884ae000 to module 'led_module' (0x977000 bytes left)
```

[boot_FRITZ.Box_7590-07.24-84940-Labor.log](https://github.com/Freetz-NG/freetz-ng/files/5734748/boot_FRITZ.Box_7590-07.24-84940-Labor.log)

fda77, 2020-12-24:
Danke. Die Busybox .config der 7590 7.21 von AVM hat noch ein paar mehr Optionen aktiviert als Freetz.
```
+       select FREETZ_BUSYBOX_STACK_OPTIMIZATION_386
+       select FREETZ_BUSYBOX_FEATURE_EDITING_WINCH
+       select FREETZ_BUSYBOX_FEATURE_UNZIP_CDF
+       select FREETZ_BUSYBOX_FACTOR
+       select FREETZ_BUSYBOX_FEATURE_XARGS_SUPPORT_ARGS_FILE
+       select FREETZ_BUSYBOX_FEATURE_WAIT_FOR_INIT
+       select FREETZ_BUSYBOX_FEATURE_INIT_QUIET
+       select FREETZ_BUSYBOX_FEATURE_INIT_MODIFY_CMDLINE
+       select FREETZ_BUSYBOX_FEATURE_NOLOGIN
+       select FREETZ_BUSYBOX_FEATURE_SECURETTY
+       select FREETZ_BUSYBOX_FEATURE_HEXDUMP_REVERSE
+       select FREETZ_BUSYBOX_XXD
+       select FREETZ_BUSYBOX_FEATURE_VOLUMEID_EXFAT
+       select FREETZ_BUSYBOX_FEATURE_VOLUMEID_F2FS
+       select FREETZ_BUSYBOX_FEATURE_WGET_HTTPS
+       select FREETZ_BUSYBOX_ASH_BASH_NOT_FOUND_HOOK
+       select FREETZ_BUSYBOX_FEATURE_SH_READ_FRAC
+       select FREETZ_BUSYBOX_FEATURE_SH_HISTFILESIZE
```

fda77, 2020-12-27:
@cnbr https://github.com/Freetz-NG/freetz-ng/commit/9762e59b6e2ae31963206babff5b9d7ffceff276 könnte helfen

cnbr, 2020-12-30:
Leider nicht wirklich, keine Veränderung feststellbar, hängt leider weiterhin.
Auszug von der  serielle Console, komplette Log im Anhang.

`[    6.473972] [announce_root] filesystem (/dev/mtdblock6) will be used as root device
[    6.491883] VFS: Mounted root (squashfs filesystem) readonly on device 31:6.
[    6.503890] devtmpfs: mounted
[    6.506586] Freeing unused kernel: 372k freed
[    6.509810] This architecture does not have kernel memory protection.
[    6.966097] random: init: uninitialized urandom read (4 bytes read)
[    7.253298] random: 1: uninitialized urandom read (4 bytes read)
[    7.446095] random: head: uninitialized urandom read (4 bytes read)
[    9.865868] [0]system-load  52% loadavg 0.8 0.2 0.1 - pgstat: sum=217254 free=102369 slab=1791 alloc=782/s fault=106/s (sleep 0)
[   32.564816] random: crng init done
[   32.566816] random: 7 urandom warning(s) missed due to ratelimiting
[   50.676629] start slab_allocator-trace now  (use cat /proc/slab_allocators)

[32m[INFO][0m Waiting for entropy
[32m[INFO][0m Waiting for entropy
[32m[INFO][0m Waiting for entropy`

[boot_7590_07.24.all-rev84707_labor_freetz-ng-17605.log](https://github.com/Freetz-NG/freetz-ng/files/5755544/boot_7590_07.24.all-rev84707_labor_freetz-ng-17605.log)

fda77, 2020-12-30:
Mist ... und danke fürs testen!

fda77, 2020-12-30:
Mist .. und danke fürs testen
@PeterPawn Hast du noch Ideen?

cnbr, 2021-01-02:
Ich habe so den Verdacht, das es irgendwie mit /dev/random und /dev/urandom zusammenhängt, wie und von wem die Devices angelegt werden und wer dann was für den Zufall beisteuert.
Im bin der busybox habe ich dann mal nach rand gesucht, dabei gibt es einige Unterschiede. Im Orginal gibt es z.B. tls_get_random, welches in der von freetz-ng nicht auftaucht. 

freetz@freetz-linux:~/freetz-ng/build$ ls -l ./original/filesystem/bin/busybox ./modified/filesystem/bin/busybox
-rwxr-xr-x 1 freetz freetz 653884 Jan  2 21:15 ./modified/filesystem/bin/busybox
-rwxrwxrwx 1 freetz freetz 619188 Dec 11 16:33 ./original/filesystem/bin/busybox

```
freetz@freetz-linux:~/freetz-ng/build$ strings ./modified/filesystem/bin/busybox|grep rand
random
srand
/dev/urandom
%s: unknown operand
rand
srand

freetz@freetz-linux:~/freetz-ng/build$  strings ./original/filesystem/bin/busybox |grep rand
tls_get_random
/dev/urandom
%s: unknown operand

freetz@freetz-linux:~/freetz-ng/build$ 
```

Ich kennen nun nicht die Zusammenhänge im Kernel, aber vielleicht gibt es ja da Abhängigkeiten.
Evtl. mal für /sbin/init auf die originale Busybox verweisen.
Danke für Deine Mühen.

fda77, 2021-01-02:
Dann fehlt vielleicht eine Option in der von Freetz. 
tls_get_random gibt es im Quellcode jedenfalls, das ist keine Erweiterung von AVM
```
 grep tls_get_random  source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/ -r    l
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/busybox_unstripped.map
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/networking/tls.c
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/networking/tls.o
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/networking/lib.a
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/networking/tls.h
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/networking/tls_rsa.o
source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.0/busybox_unstripped
```

fda77, 2021-01-04:
Ich hab verschieden Optionen in der Busybox aktiviert, abes es nicht hinbekommen dass der String "tls_get_random" enthalten ist.
Man könnte testen ob es überhaupt an der busybox liegt, indem man diese nicht ersetzt sondern die von AVM verwendet. Dazu gibt es aber keine Option in Freetz. Man müsst sich zb was mit der ./fwmod_custom basteln.
Mit der avm-busybox wird freetz aber nicht starten, da Applets fehlen werden. Aber Freetz wird eh erst viel später gestartet

cnbr, 2021-01-05:
Habe mal /sbin/init mit der original busybox ersetzt, Image erstellt und gebootet, es war aber leider keine Veränderung feststellbar, Log sah so aus, wie bisher und es hing an der gleichen Stelle. 
Vielleicht ist das mit tls was anderes oder nicht alles und das Problem sitzt tiefer.

fda77, 2021-01-05:
Mit `find build/ -name busybox -exec ls -al {} ';'` waren es dann die gleichen binaries? Das würde schonmal busybox komplett ausschliessen.
AVM hat in der 7.24 labor im Gegensatz zu 7.2x stable was in /dev geändert

labor
```
$ l build/original/filesystem/dev/
insgesamt 4
lrwxrwxrwx. 1 x x   19  5. Jan 21:36 kdsld_misc -> /var/dev/kdsld_misc
-rw-rw-rw-. 1 x x    0 11. Dez 17:34 led
lrwxrwxrwx. 1 x x   10  5. Jan 21:36 log -> ../tmp/log
lrwxrwxrwx. 1 x x    9  5. Jan 21:36 new_led -> /dev/null
drwxr-xr-x. 2 x x 4096 11. Dez 17:33 pts
```

stable
```
 l build/original/filesystem/dev/
insgesamt 4
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace128
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace129
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace130
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace131
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace132
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace133
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace134
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace135
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace136
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace137
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace138
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace139
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace140
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace141
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace142
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace143
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace160
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace161
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace162
-rw-rw-rw-. 1 x x    0 20. Feb 2014  avm_net_trace163
-rw-rw-rw-. 1 x x    0 20. Feb 2014  capi20
-rw-rw-rw-. 1 x x    0 20. Feb 2014  console
lrwxrwxrwx. 1 x x   19  5. Jan 14:04 kdsld_misc -> /var/dev/kdsld_misc
-rw-rw-rw-. 1 x x    0 20. Feb 2014  kmem
-rw-rw-rw-. 1 x x    0 20. Feb 2014  led
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop1
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop2
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop3
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop4
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop5
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop6
-rw-rw-rw-. 1 x x    0 20. Feb 2014  loop7
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mem
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd1
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd10
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd11
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd12
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd13
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd14
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd15
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd2
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd3
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd4
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd5
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd6
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd7
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd8
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtd9
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock1
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock10
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock11
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock12
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock13
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock14
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock15
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock2
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock3
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock4
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock5
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock6
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock7
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock8
-rw-rw-rw-. 1 x x    0 20. Feb 2014  mtdblock9
lrwxrwxrwx. 1 x x    9  5. Jan 14:04 new_led -> /dev/null
-rw-rw-rw-. 1 x x    0 20. Feb 2014  null
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ptmx
drwxr-x---. 2 x x 4096 20. Feb 2014  pts
-rw-rw-rw-. 1 x x    0 20. Feb 2014  random
-rw-r--r--. 1 x x    0 20. Feb 2014  tiatm
-rw-rw-rw-. 1 x x    0 20. Feb 2014  tty
-rw-rw-rw-. 1 x x    0 20. Feb 2014  tty0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ttyp0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ttyp1
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ttyp2
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ttyS0
-rw-rw-rw-. 1 x x    0 20. Feb 2014  ttyS1
-rw-rw-rw-. 1 x x    0 20. Feb 2014  urandom
-rw-rw-rw-. 1 x x    0 20. Feb 2014  zero

```

/dev/random, /dev/urandom und die anderen müssen also irgendwo erstellt werden

cnbr, 2021-01-05:
War auf die schnelle gemacht, 

in der ` ./packages/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox/busybox.links`  
das `/sbin/init` rausgenommen, 
in der `fwmod` ein `cp -pv ~/freetz-ng/build/original/filesystem/sbin/init ~/freetz-ng/build/modified/filesystem/sbin/init`
vor dem `# Install new busybox links` eingebaut

Ergebnis war dann 

```
freetz@freetz-linux:~/freetz-ng$ find build/ -name busybox -exec ls -al {} ';' 
-rwxrwxrwx 1 freetz freetz 619188 Dec 11 16:33 build/original/filesystem/bin/busybox
-rwxr-xr-x 1 freetz freetz 653884 Jan  2 21:15 build/modified/filesystem/bin/busybox
```

```
freetz@freetz-linux:~/freetz-ng$ find build/ -name init -exec ls -al {} ';' 
lrwxrwxrwx 1 freetz freetz 14 Dec 21 10:26 build/original/filesystem/sbin/init -> ../bin/busybox
-rwxrwxrwx 1 freetz freetz 619188 Dec 11 16:33 build/modified/filesystem/sbin/init
freetz@freetz-linux:~/freetz-ng$ 
```
Ich denke, damit wurde dann die original Busybox als /sbin/init genommen.

fda77, 2021-01-05:
Ne, in original und modified müssen dann die gleichen Dateien sein, also Datum und Größe gleich. In modified ist drin was später als .image zusammengepackt wird. Das init brauchst du nicht zu kopieren, da es nur ein link ist.
So hattest du nur den Befehl "init" auf die AVM-Version zurückgesetzt.

`
cp -pv ~/freetz-ng/build/original/filesystem/bin/busybox ~/freetz-ng/build/modified/filesystem/bin/busybox
`
Dann  `make` und noch mal find find die busybox verlgeichen

cnbr, 2021-01-08:
Bei dem Versuch mit dem init hatte ich die leichte Hoffnung, das die originale busybox den init bereitstellt, damit  die Box bootet und für alles andere dann die modifizierte verwendet wird.  War nun nicht so. 

Nun bin ich noch den Weg „kompletter Austausch der busybox“ gegangen, aber leider hatte das auch keinen Erfolg, hing wieder, die Logzeilen sahen in den Bereich etwas anderes aus. 
```

[    6.511505] ubi0: available PEBs: 0, total reserved PEBs: 3928, PEBs reserved for bad PEB handling: 80
[    6.520821] ubi0: background thread "ubi_bgt0d" started, PID 629
[    6.530012] [announce_root] filesystem (/dev/mtdblock6) will be used as root device
[    6.548396] VFS: Mounted root (squashfs filesystem) readonly on device 31:6.
[    6.560783] devtmpfs: mounted
[    6.563419] Freeing unused kernel: 372k freed
[    6.566693] This architecture does not have kernel memory protection.
[    9.933031] [0]system-load  39% loadavg 0.8 0.2 0.1 - pgstat: sum=217500 free=102639 slab=1808 alloc=710/s fault=79/s (sleep 2)
[   33.530397] random: crng init done

#[32m[INFO]#[0m Waiting for entropy
#[32m[INFO]#[0m Waiting for entropy
#[32m[INFO]#[0m Waiting for entropy
#[32m[INFO]#[0m Waiting for entropy
#[32m[INFO]#[0m Waiting for entropy
[   50.679949] start slab_allocator-trace now  (use cat /proc/slab_allocators)
#[32m[INFO]#[0m Waiting for entropy

```
Der `find` zur busybox nach dem `make`
```
freetz@freetz-linux:~/freetz-ng$ find build/ -name busybox -exec ls -al {} ';' 
-rwxrwxrwx 1 freetz freetz 619188 Dec 11 16:33 build/original/filesystem/bin/busybox
-rwxrwxrwx 1 freetz freetz 619188 Dec 11 16:33 build/modified/filesystem/bin/busybox
freetz@freetz-linux:~/freetz-ng$ 
```

Somit wird die busybox wohl nicht die alleinige Ursache für das Hängen beim booten sein, wenn sie es überhaupt ist. 

[7590_07.24.all-rev84707_labor_freetz-ng-17657M.log](https://github.com/Freetz-NG/freetz-ng/files/5788837/7590_07.24.all-rev84707_labor_freetz-ng-17657M.log)

fda77, 2021-01-08:
Mit der AVM-Busybox fehlen damit die Zeilen
```
[    6.966097] random: init: uninitialized urandom read (4 bytes read)
[    7.253298] random: 1: uninitialized urandom read (4 bytes read)
[    7.446095] random: head: uninitialized urandom read (4 bytes read)
```
Genaueres wird man dann wissen wenn AVM die Quellen veröffentlicht

PeterPawn, 2021-01-08:
Ich kann mir fast nicht vorstellen, daß AVM am "init"-Applet irgendwie herumgeschraubt hat. 

Aber ich blicke bei den Tests auch nicht mehr durch ... ich sehe vier relevante Fälle, die zu unterscheiden/untersuchen wären:

- originales System mit AVM-BusyBox -> geht
- Freetz-NG mit Freetz-BusyBox -> hängt
- originales System mit Freetz-BusyBox -> ???
- Freetz-NG mit AVM-BusyBox -> ???

Die Fragezeichen könnte/müßte ich mir hier mit viel Glück zusammenklauben können ... aber eine (systematische) Zusammenfassung des Stands durch jemanden, der das auch (ebenso systematisch) getestet hat, macht ja irgendwo mehr Sinn.

Für die letzte Kombination in der Liste könnte/müßte man (falls Freetz-NG dann nicht richtig startet wg. fehlender Applets, was aber eigentlich auch nicht stören würde) mit "ds_mod=off" (https://github.com/Freetz-NG/freetz-ng/issues/113) den Rest auch abschalten, so daß man tatsächlich sicher sein kann, daß es an den "statischen" Modifikationen (das wäre die andere BusyBox und ggf. das veränderte Laden von LKM, falls das System überhaupt bis dahin kommt) seitens Freetz-NG liegt und nicht an irgendwelchen anderen Paketen, die später erst gestartet würden.

Eigentlich sollte/dürfte es keine Abweichungen zwischen AVM-Firmware und Freetz-NG geben bis zu dem Zeitpunkt, wo in der "rc.,hwcrypto" (oder so ähnlich, kann gerade nicht nachsehen) das EIP123-LKM geladen wird und das Warten auf ausreichende Entropie beginnt. Wenn ich mich recht erinnere, ist das ja alles noch vor dem Start des "supervisor" ... auch wieder nur aus dem Kopf und noch nicht einmal konkret für 07.24-Labor überprüft. Wenn AVM da geändert haben sollte ggü. der 07.2x-Release-Version, dann müßte man sich das noch einmal genauer ansehen.

fda77, 2021-01-08:
> Freetz-NG mit AVM-BusyBox -> ???

Ist doch das worauf sich die 2 Posts über deinem bezogen

Was ich noch weiter oben über den Inhalt von /dev schrieb kann ich nicht mehr nachvollziehen

PeterPawn, 2021-01-09:
Steht da aber nicht ... wer sich mit den Pfaden "im Inneren" von Freetz(-NG) nicht auskennt, versteht anhand des Textes (meiner Meinung nach) nicht, daß es sich dabei um den Fall 4 handeln soll und außerdem kann man auch nach den Änderungen nicht sehen, was von "init" angeworfen wurde und wie weit das jeweils vorankommt.

Ein "set -x" und ein "exec 2>/dev/console" im ersten Skript (das wäre dann /etc/boot.d/1) und man kann auf der Seriellen schön beobachten, was das "init" da macht.

Oder man geht mal hin und packt in die "rc.hwcrypto" noch die Anzeige des Wertes, der aus "/proc/sys/kernel/random/entropy_avail" ausgelesen wird (bisher wird der ja nur verglichen) ... wenn der nicht wachsen sollte beim Warten, stimmt wohl was mit dem "eip123.ko" nicht, das davor geladen werden sollte (und muß, damit da die Entropie überhaupt wachsen kann) - auch wenn das bisher aus den Kernel-Messages eigentlich nicht ersichtlich war.

Und wenn's um die statischen Änderungen in Freetz VOR "rc.hwcrypto" geht, dann kommt da ja nicht allzu viel in Frage ... am ehesten noch der Inhalt der "/var.tar", denn die wird ja zuvor noch entpackt nach "/var".

Wenn man da hineinsieht, fällt aber auch nichts direkt auf ... ggf. könnte noch die Änderung des Home-Verzeichnisses für den "root"-User ein Problem darstellen, weil das "init"-Applet wohl unter diesem Konto läuft und dann das Home-Verzeichnis natürlich ein anderes ist - damit zeigt auch das ".profile" im Home-Verzeichnis in den Wald, bis irgendwann mal die Freetz-Settings nach "/mod" entpackt werden. Das MUSS auch alles nichts heißen, aber das sind eben schon Unterschiede zwischen AVM-Firmware und Freetz-Image, die jenseits der anderen BusyBox liegen. Wobei mir dann auch einfällt, daß man bei "Freetz-BusyBox" ja noch zwischen "letzte Version" und "gleiche Version (der BusyBox) wie bei AVM" unterscheiden sollte bzw. das testen müßte.

Was ich überhaupt nicht verstehe ... warum wird eigentlich in der "/var/post_install" in Freetz-NG das Entladen der DECT-LKMs beim Herunterfahren übergangen?

```
vidar:/tmp/ng-check/build/modified/filesystem # diff -u /home/FritzBox/FB7590/154.07.24-84707/post_install /tmp/ng-check/build/modified/filesystem/post_install
--- FB7590/154.07.24-84707/post_install  2020-12-11 17:33:40.000000000 +0100
+++ build/modified/filesystem/post_install        2021-01-09 00:40:29.000000000 +0100
@@ -132,8 +132,6 @@
 lsmod
 echo "stopping modules using ubik2/pcmlink .."
 if lsmod | grep -q ulpcmlink ; then rmmod ulpcmlink ; fi
-if lsmod | grep -q dect_io ; then rmmod dect_io ; fi
-if lsmod | grep -q avm_dect ; then rmmod avm_dect ; fi
 if lsmod | grep -q pcmlink ; then rmmod pcmlink ; fi
 set +x
 echo "still running:"
```

Gibt's dafür irgendeinen plausiblen Grund? Selbst wenn die sich ggf. (zur Zeit) nicht entladen lassen, weil die entsprechenden Funktionen nicht implementiert sind, muß man die Kommandos ja nicht entfernen ... ändert AVM das wieder, denkt da niemand mehr wirklich daran, daß da noch Modifikationen rückgängig zu machen wären. Das gehört zwar auch nicht 100% zu diesem Problem, aber immerhin noch zum Komplex "Unterschiede AVM <-> Freetz-NG, die auch ohne "Freetz-Mod" zutage treten".

fda77, 2021-01-10:
Im Text noch ein paar Beiträge weiter oben steht was es mit den Verzeichnissen auf sich hat ;-)
Um ` /etc/boot.d/1` schnell mal zu bearbeiten würde ich das direkt in build/original/filesystem//etc/boot.d/1 machen. original wird nur neu erstellt falls man die Box im menuconfig ändert oder es noch nicht vorhanden ist. Um das ganze zurückzusetzen einfach das komplette build/ löschen, "make" legt es wieder an. Auf diesem Weg kann man auch eigene zusätzliche Dateien ins Image bringen. Was nicht geht: Dateien ändern die Freetz anlegt

Das HOME muss ich mir mal anschauen. Ist das schon länger so oder was neues von AVM?

DECT: Bei einem Gerät (evtl 7390?) hat sich das Entladen nicht beendet. Dazu gibt es bestimmt noch ein Trac-Ticket. Dh "reboot" war nicht erfolgreich und es war auch kein Flashen über beide Webif möglich. Für aktuelle Boxen sollte das nicht mehr nötig sein...

PeterPawn, 2021-01-10:
> ein paar Beiträge weiter oben steht was

Eben ... es gibt keine Zusammenfassung(en) für einen einfachen Überblick, sondern man muß sich Kommentar für Kommentar durcharbeiten (und die dann auch noch jeweils richtig verstehen, was häufig auch nicht einfach ist), wenn man den "aktuellen Stand" irgendwie erkennen will. Das kostet unnötige Zeit, provoziert Mißverständnisse (durch unterschiedliche Interpretation irgendwelcher Kommentare oder beschriebener Ergebnisse) und schreckt andere ab, das überhaupt verstehen zu wollen.

---

> in der AVM Busybox der Labor der "string" "tls_get_random" enthalten ist und bei Freetz nicht

Das "tls_get_random" als Funktion sollte auch bei Freetz  in der BusyBox enthalten sein, wenn man die richtigen Optionen bei der Übersetzung wählt - insbesondere kein `CONFIG_FREATURE_WGET_OPENSSL` und trotzdem `CONFIG_FEATURE_WGET_HTTPS` (letzteres sollte `CONFIG_TLS=y` bewirken). Ob es am Ende in der Liste der exportierten Symbole auftaucht, liegt nur marginal an der Existenz der `networking/tls.o` (die dann natürlich mit gelinkt werden sollte) und eher an den Linker-Optionen bzw. am `strip`. Letztlich braucht es diesen Namen ja auch nicht als externes Symbol, solange keiner "von außen" darauf zugreifen soll - die interne Verwendung klappt auch, wenn die Namen "verschleiert" sind. In der "unstripped version" der BusyBox solltest Du das Symbol in jedem Falle auch noch finden können (auch als externes Symbol, also bei einem `nm -g ...`).

An kompliziertere Vorgänge würde ich erst dann denken, wenn es keine anderen logischen Erklärungen mehr gibt. Techniken wie das Überladen dieser Funktion in der BusyBox (beim Laden in den Speicher durch den Kernel), für die man dann die betreffende Adresse natürlich auch "kennen" muß, sind schon sehr exotisch (und eher "Hacking") und ich glaube nicht, daß AVM dazu greifen würde, weil es sicherere Optionen gäbe. 

Außerdem erklärt das dann auch noch nicht, warum Freetz-NG mit der originalen BusyBox ebenfalls nicht klappt ... wobei ich immer noch skeptisch bin, ob die Randbedingungen alle so eingehalten wurden, daß da tatsächlich am Ende auch im laufenden System die AVM-BusyBox zum Einsatz kam (so ein `ls` wie oben prüft dann halt auch nur vor dem Einpacken und nicht wirklich das Ergebnis - das ist also nicht wirklich "überprüft", sondern nur "geschlußfolgert"). Aber das kann ein mutiges `/bin/busybox --help >/dev/console` am Beginn der Initialisierung ja auch schnell abklären.

---

> Für aktuelle Boxen sollte das nicht mehr nötig sein...

Das war m.W. nie wirklich notwendig, ist aber letztlich auch egal ... denn genau das meinte ich, als ich davon schrieb, daß am Ende kein Aas mehr weiß, warum solche Änderungen ausgeführt wurden und für welche Modelle sie sinnvoll sein könnten. Davor bewahrt einen eben die Aufteilung der Patches auf entsprechende "funktionelle Änderungen" in gesonderten Dateien, wo man gleich am Namen erkennt, was gehauen und gestochen ist (und die man ggf. auch in modell- und/oder versionsspezifische Ordner packen kann in Freetz) - und wenn es dann dreist 20 Files gibt, die an derselben Datei herumpatchen.

fda77, 2021-01-12:
Der aktuelle Stand ist dass halt gesucht wird wo das Problem liegt. Man kann ja nicht nach jeden neuen Post eine neue Zusammenfassung machen.

tls_get_random: Okay, wenn das sich nur auf wget bezieht spielt es da ganz am Anfang keine Rolle. Avm nutzt ja eh den eigenen downloader

> ein ls wie oben prüft dann halt auch nur vor dem Einpacken und nicht wirklich das Ergebnis

Doch, was make nach "modified" ablegt ist das was durch das gleiche make nachher im Image landet. Mit fällt da keine Möglichkeit ein wie man das falsch machen könnte.
Ausserdem verschwanden die 3 Zeilen im Bootlog damit

DECT: Doch, ist notwendig, ich hatte auch das Problem mit der 7390. Was diese Zeile macht stand doch im Script direkt in der Zeile darüber

PeterPawn, 2021-01-12:
Was sucht ein Patch für die `var/post_install` in einer Datei, die dem Namen nach die `var/install` ändert?

Das findet niemand, der danach sucht, wenn er sich nicht extrem gut auskennt mit dem Aufbau von Freetz und sich mit einem `grep` o.ä. selbst auf die Suche macht, in welcher Datei diese Stelle denn nun zu finden sein könnte - zumindest findet man es nicht in angemessener Zeit.

DAS meinte ich mit der Systematik - außerdem gehört so ein Patch (nach meiner Ansicht zumindest) dann eben nach `patches/devices` (wofür gibt es diese zusätzliche Struktur sonst überhaupt?) und nicht - mit zusätzlicher Abfrage von Bedingungen - nach `patches/scripts` ... selbst wenn im Kommentar vor dem Aufruf von `modsed` irgendwo die 7390 erwähnt war.

Da kann nämlich auch niemand sehen, ob der Kommentar nicht noch aus der Zeit stammt(e), wo es gar nichts Neueres als die 7390 gab und der Patch damit für die Nachfolgemodelle auch noch notwendig sein könnte oder nicht - ein Grund, warum man üblicherweise solche Kommentare auch noch mit Datum/Uhrzeit/Autor versieht, wie Du Dir in den AVM-Quellen mehrfach (in C halt) ansehen kannst - zumindest dann, wenn da mehr als ein Einzelner involviert ist und das ist bei Freetz ja auch so, selbst wenn das bei Freetz-NG wieder keine Frage mehr sein mag. Aber man kann so auch nur schwer zwischen "alten Änderungen" (die sich schon in Freetz finden) und neuen von Dir unterscheiden - nicht jeder geht erst einmal hin und macht ein `diff`, ob sich eine Datei zwischen Freetz und Freetz-NG geändert hat.

Ich weiß auch nicht, durch welche anderen Änderungen die Notwendigkeit dieses Patches für die DECT-LKM bei Dir entstand ... meine 7390 (mit angepaßter Original-Firmware 06.86) waren jedenfalls auch ohne das Entfernen dieser `rmmod`-Aufrufe jederzeit in der Lage, neu zu starten, wenn man sie dazu aufforderte. Das mag also wieder ein Phänomen gewesen sein, was mit anderen Änderungen in Freetz-NG zusammenhing und vielleicht hat auch hier wieder das Pflaster/der Flicken den Blick auf die eigentliche Ursache des Problems verstellt. 

Ich kann mich auch nicht an Freetz-Issues für die 7390 erinnern (zumindest an keine, die eine entsprechende Anzahl Betroffener hatte), wo ein Hängenbleiben beim Neustart ein Thema gewesen wäre - aber das muß tatsächlich nichts heißen.

fda77, 2021-01-12:
Die anderen Zeilen in der Datei verändern doch auch indirekt die post_install
Das ganze war lange vor ng, ich hab jetzt kein Ticket gefunden, vielleicht war das auch im Forum.
Ich glaube AVM hatte da gewisse Dect-Events ins Syslog geschrieben, und mit was anderem ausgelesen. Evtl ein spezieller printk-Dingens-Modus. Und durch Freetz/ gab es da irgend welche Problem. Aber so genau weiss ich das alles nicht mehr

PeterPawn, 2021-01-12:
> Die anderen Zeilen in der Datei verändern doch auch indirekt die post_install

Wo genau?`

Die ändern die `var/install` und erst die schreibt bei ihrer Ausführung im Rahmen eines Updates dann weitere Daten in die `var/post_install` ... aber auch da nur in die bereits aus der `var.tar` extrahierte Datei und nicht in das "template" (das es mittlerweile sogar noch zusätzlich gibt bei AVM in der `var.tar` - hatten wir letztens irgendwo anders schon) im Tarball.

fda77, 2021-01-12:
Ist doch so okay, hat ja so bislang funktioniert. Für die 7390 gab es eh kein Fritzos 7+ mehr. 
Normalerweis entlädt man für einen Reboot eh keine Module...

PeterPawn, 2021-01-12:
Das ist vermutlich der Unterschied zwischen "So ist es richtig." und "Pascht scho' irgendwie." - einen plausiblen Grund, warum man das nicht in einer gesonderten Datei patchen sollte, habe ich jedenfalls nicht gefunden/gelesen. 

Aber wie so oft ... lassen wir das Thema einfach bleiben, denn schließlich macht AVM das ja schon falsch, wenn sie sich dazu versteigen, die DECT-LKMs zu entladen - wobei vermutlich auch nur dort die Leute zu finden sind, die genau sagen können, was beim Entladen dieser LKMs eigentlich passieren sollte. 

Das kann ja bis hin zur Signalisierung an die DECT-Geräte, daß die Basis ab jetzt nicht mehr verfügbar ist, gehen - ich verstehe halt nicht, warum man bei Änderungen immer so weit wie möglich von der originalen Firmware "wegkommen" wollte, anstatt so dicht wie möglich dranzubleiben. 

Je geringer die Unterschiede (weil man nur das ändert, was tatsächlich erforderlich ist, um ein Ziel zu erreichen und nicht einfach "auf Verdacht" noch alles mögliche andere), um so geringer auch die Wahrscheinlichkeit für zusätzliche Probleme, die sich daraus ergeben können - und "unerklärliche Phänomene", von denen nur Freetz (oder sogar nur Freetz-NG) betroffen ist (auch wenn Freetz sicherlich nur ältere Versionen unterstützt, gibt es ja noch andere Optionen der Modifikation und von denen hört man zumindest nicht, daß die ähnliche Probleme machen), gibt es ja nun genug. 

Wenn man dann bei der Fehlersuche und der dabei erfolgenden Suche nach möglichen Unterschieden weniger Punkte abzuklappern hat, ist das sicherlich auch von Vorteil.

fda77, 2021-01-13:
Ist ja jetzt nur noch für die alten Geräte bei denen es auftrat, dass sollte auf jeden fall geändert werden.
Ansonsten benutze ich für die Verzeichnisse config/ und patches/ tatsächlich meist grep. Die Sortierung dort erschliesst sich mir nicht so ganz, aber wenn man was ändert und die Sortierung durcheinander wirft passt nachher irgendwas bestimmt nicht mehr

Busybox: Du meintest vor kurzem es wäre ganz gut genau die Version von AVM zu nehmen. Also man kann problemlos viel Versionen der Busybox aufnehmen (in ng). Das gibt aber eine unmenge an Symbolen im menuconfig, sieh make/busybox/. Ich habe die Befürchtung dass es irgend wann für kconfig zu viele werden.
Ausserdem werden vermutlich viele bei Problemmeldungen die genutzte Version nicht angeben was die Fehlersuche erschwert

PeterPawn, 2021-01-13:
Hinsichtlich der BusyBox hast Du mich vermutlich auch falsch verstanden - gegen deren Ersetzen ist wenig einzuwenden, solange sie minimal dieselben Funktionen bietet, wie die von AVM und sich im Verhalten nur so weit unterscheidet, daß es nicht zum Problem wird (Stichwort LKM-Kommandos). Deine Änderungen an dieser Stelle, wenn es um die Frage geht, welche Optionen denn bei der Freetz-BusyBox minimal zu aktivieren sind, sind ja vollkommen in Ordnung in meinen Augen. Ich weiß zwar nicht mehr, ob Du dafür auch das "generate.sh" für die BB angepaßt hattest, aber das Vorgehen war mir schon immer suspekt und für reibungslosen Austausch brauchte es schon immer eine "handgemachte" BusyBox-Konfiguration bei mir. 

_Die war ja auch der wesentliche Punkt (weil sie viel Zeit in Anspruch nimmt), warum __ich__ bei mir eine Möglichkeit brauchte, die `.config` für verschiedene Modelle und Versionen nachnutzen zu können, ohne jedesmal BusyBox und (wenige) Pakete neu konfigurieren zu müssen. Aber zurück zur BusyBox..._

Speziell die Optimierungen (Applets bevorzugen, NOFORK, etc.) können aber zu abweichendem Verhalten und ggf. zum Aufruf eines falschen Kommandos führen, wobei auch bei AVM mittlerweile an vielen Stellen immer öfter der Aufruf mit absoluten Pfaden zu finden ist, was zwar zum Problem wird, wenn man da etwas ersetzen/einschmuggeln will, aber die Sicherheit, daß man tatsächlich das gewünschte Kommando findet, deutlich erhöht - zumindest solange das Dateisystem unverändert ist. Ein Austausch nur über das Ausnutzen der Suchreihenfolge (explizit in PATH oder implizit) ist damit verhindert.

Irgendwann muß man auch einsehen, daß die AVM-Firmware deutlich komplexer geworden ist im Laufe der Jahre und daß eine Änderung durch Austausch/Löschen jedes beliebigen Teils/Kommandos immer öfter zu unerwarteten Seiteneffekten führt - hier will ich mal an die XFRM-Funktionen von "iptables" erinnern, die ja bei den GRX-Boxen (oder sogar bei allen Modellen) zum nächsten Problem führten, als AVM die VPN-Implementierung überarbeitete. 

Man muß daher m.E. bei jeder wirklich neuen Version (und in den Labor-Versionen ändert sich das eben noch ständig bzw. es besteht die Chance dafür) erst einmal ergründen, was sich gegenüber dem Vorgänger geändert hat und welche Auswirkungen das auf das eigene Projekt haben könnte. Einfach nur die Versionsnummer der Vorlage zu erhöhen und dabei dann darauf zu vertrauen, daß das schon weiterhin alles passen würde, führt mich wieder zu der (bisher nicht wirklich beantworteten) Frage, wo genau Du Dein Freetz-NG denn nun sehen willst - die Signale sind da (zumindest für mich) immer noch widersprüchlich, auch wenn ich Dein erstes Tag tatsächlich gesehen habe. 

Aber glaubst Du tatsächlich selbst, daß irgendjemand ein altes Tag (mit zusätzlichen Kommandos/Optionen, die es dafür braucht) auschecken und benutzen würde? Und was wäre Deine Reaktion auf einen Fehlerbericht, der sich nur auf die alte Version am Tag (täck, nicht taag) bezieht?

Das ist eben wieder ein gewaltiger Unterschied zwischen Tags und Branches ... ich kann einen Commit zur Fehlerkorrektur problemlos auch auf zwei Branches anwenden (solange sie sich nicht gerade an dieser Stelle auch noch unterscheiden) und habe das Problem damit in beiden Zweigen gelöst. Insofern hast Du (für mich) bisher nur den halben Weg zurückgelegt beim Versuch, eine "nutzerfreundliche Version" anzubieten, bei welcher der potentielle Interessent davon ausgehen kann, daß die tatsächlich auch (auf einer hinreichenden Anzahl von Installationen) getestet ist.

Und anders als Du (gerade wieder bei der 6591 zu sehen, iirc) bin ich nicht der Ansicht, daß bereits ein einzelner erfolgreicher Anwender ausreichend sein sollte, um irgendwo ein "funktioniert"-Schild draufzupappen. Das FRITZ!OS ist mittlerweile ein so komplexes Gebilde (vergleichen mit den allermeisten anderen OS für SOHO-Router, wobei es inzwischen auch andere Geräte gibt, die beim Funktionsumfang - auch mit mehr OSS-Benutzung - deutlich aufholen), daß ein Einzelner gar nicht alle Teile davon benutzen KANN - manches schließt sich ja auch gegenseitig aus, wie IP-Client und VPN/DynDNS/etc. 

Ich versuche mir gerade vorzustellen, wie das wäre, wenn AVM nach dem ersten Kunden, der kein Problem in der neuen Labor-Version hatte, das Zeug gleich als Release unter die Leute bringen würde. Und dort kapriziert man sich immer nur auf eine (nämlich die letzte aktuelle) Version und hat gar nicht den Ehrgeiz, auch noch die Geräte aus dem 30-jährigen Krieg zu unterstützen ... was das Ganze für AVM noch einmal deutlich leichter macht. 

Da sieht also schon AVM mit einem ganzen Programmierer-/Tester-Team sich gezwungen, gewisse Grenzen zu ziehen ... und das alles jetzt als Einzelner (mit ein paar "Freiwilligen", die auch nicht immer ganz freiwillig als Tester herhalten sollen/müssen) weiterhin im Griff zu behalten (während bei AVM die Entwicklung immer weiter fortschreitet), ist halt schon sehr ambitioniert und man KANN unmöglich auf allen Hochzeiten gleichzeitig tanzen. Da muß man sich mal für eine Veranstaltung entscheiden - sonst bleibt das (in meinen Augen zumindest) immer nur Stückwerk.

Und ich bin auch tatsächlich der Ansicht, daß Du ohne eine solche grundsätzliche Entscheidung und "offizielle" Positionierung für Freetz-NG nur sehr wenige (ernsthafte) Mitstreiter finden wirst - dann bleibt es bei einer (durchaus wachsenden) Zahl an "Benutzern", die dann aber erwarten (werden), daß Du Dich um ihre Probleme mit Deine [... cut ...] nteressiert dann die "Marketing-Abteilung" gar nicht mehr - das erinnert ein wenig an AVM und das "DNS over TLS", was zwar auch "implementiert" ist, aber noch so viele Probleme bereitet, daß viele Benutzer (zumindest diejenigen, die die Ursache verstehen und sich nicht einfach durch einen Neustart behelfen, weil sie gar nichts anderes können) das lieber wieder abschalten.

Aber wir sollten das tatsächlich eher nach "Discussions" verlagern - ich mache mir mal vorsichtshalber eine Kopie der Seite.

fda77, 2021-01-13:
Mit der Busybox ist das so eine Sache wie man an https://github.com/Freetz-NG/freetz-ng/commit/f2f5e8430e88db4fed98a43790403adc9ba9d2e0 sieht. Plötzlich fehlt das letzte ' das sich hinter der Klammer der Variable befindet. Ich hab die 1.32.0 zwar 6 Monate bei mir laufen gehabt, das fiel mir aber nicht auf. Evtl kam es erst durch 1.32.1.
Die von AVM genutzten Optionen werden noch nicht ausgewertet (die ganzen neuen Dateien in make/busybox/avm/), das ist nur manuell durch selects oder bei strings in der generate.sh.
Der Tag war halt vor größeren Änderungen an busybox und uclibc. Als etwas in richtung stable und der branch mach neues was noch Problem machen kann (war bei busybox ja so)

Die 2 Issues heute hab ich nur gelöscht weil durch die Umwandlung in eine Diskussion diese kopiert wurden, und der Status automatisch auf "locked" gesetzt wurde. Das half der Übersicht in den Issues nicht wirklich.
Eigen Diskussionen wären schon gut. Man kann hier Posts als off-topic markieren, die sind dann eingeklappt und werden erst nach einem klick angezeigt. Das wäre für dieses Issue ganz gut :)

Die Toolchain ist doch einfach, weil nämlich nur uclib verwendet wird! Was AVM verwendet ist egal bei den neueren FOS! Also so wie du vorgeschlagen hast. Dazu auch nur noch die neueren single-lib version (Für Nutzer von vorkompilierten Binaries ist das halt etwas stressig...)

PeterPawn, 2021-01-13:
> Also so wie du vorgeschlagen hast.

Eben nicht ganz so ... während `musl` ja genau dafür auch (mit-)entwickelt wurde, statisch gelinkt zu werden (https://musl.libc.org/about.html): 

> Due to lack of heavy internal coupling between libc components, static linking with musl pulls in very little code that the application isn't actually using. Minimal static-linked binaries can be under 10 kB of code, even with threads, and even useful programs can be under 50 kB.

ist das bei der `uClibc` nicht der Fall und die ganzen Optionen zur Konfiguration machen die Benutzung der `uClibc[-ng]` auch nicht gerade einfacher und konsistenter.

Verglichen mit der `uClibc[-ng]` (und __die__ wurde schon für "embedded devices" entwickelt), ist `musl` dann eben noch einmal effektiver - hier: https://elinux.org/images/e/eb/Transitioning_From_uclibc_to_musl_for_Embedded_Development.pdf findest Du ein Paper, was `uClibc` und `musl` miteinander vergleicht.

Es gab ja auch gute Gründe, warum OpenWRT vor mehr als fünf Jahren zu `musl` als Standard gewechselt hat: https://lists.openwrt.org/pipermail/openwrt-devel/2015-June/007680.html - letztlich hat AVM das jetzt (nach einem Ausflug zur `glibc`) auch nur nachvollzogen und ich hoffe mal (bzw. ich wage die These), daß das für künftige Systeme der präferierte Weg sein wird. 

Die `glibc` ist einfach deutlich umfangreicher und für Geräte wie die FRITZ!Boxen praktisch Overkill und die `uClibc` ist im Original ziemlich alt, während die `uClibc-ng` im Embedded-Umfeld (sicherlich auch, weil es da `musl` schon gab und die Vorteile - siehe das oben verlinkte Paper - auf der Hand liegen) nie so richtig in die Puschen kam.

Wenn man dann noch die Idee hinter dem Flatpak-Format (https://flatpak.org/) berücksichtigt (das ja auch "beim Containern" :-) zunehmend eine Rolle spielt), dann kommt man noch eher zur Notwendigkeit von Komponenten (oder eben speziell der C-Library), die ein effektives statisches Linken unterstützen, damit die erzeugten "Flat-Paks" (auch wenn eine FRITZ!Box sicherlich kein Desktop-System ist) auf die notwendige Größe beschränkt werden können. 

Alles das, was in Freetz in den Abhängigkeiten der Pakete von anderen bzw. von bestimmten Libraries steckt, hat sich automatisch überlebt, wenn man tatsächlich zu solchen "containerisierten Apps" (für einen DNS-/DHCP-Server, für einen SSH-Server, für ein VPN, etc. - die OS für die NAS von QNAP oder Synology machen vor, wie das geht) käme und das wäre dann (wenn es selbst alles mitbringt, was man dafür braucht) auch noch unabhängig von der Version des FRITZ!OS, auf der es laufen soll (solange die Architektur stimmt und es sich irgendwie in das FRITZ!OS integrieren kann).

Das ist sicherlich auch nichts, was man von heute auf morgen auf die Beine stellen kann ... aber es ist nun mal die Zukunft (mal abgesehen davon, daß man das dann tatsächlich auch wieder viel mehr (potentiellen) Benutzern zugänglich machen kann) und nur, weil es diese Architekturideen und die dafür notwendigen Werkzeuge vor 15 Jahren noch nicht gab, als Freetz aus der Taufe gehoben wurde, kann das Rezept für die heutige Weiterentwicklung ja nicht "Vorwärts in die Vergangenheit" heißen - auch ein Windows 95/98 (oder XP, wenn wir zeitlich besser liegen wollen zur "Geburt" von Freetz) mußte irgendwann mal etwas Neuerem weichen.

Entweder man sieht Freetz als die "gute alte Zeit" an, die man nicht missen möchte und die man mit (sehr) viel Aufwand am Leben erhält oder man sieht sie nur als Übergangslösung, bis etwas Moderneres zur Verfügung steht. Und damit komme ich schon wieder auf die "Positionierung" von Freetz-NG zu sprechen ... irgendwann kommt man einfach nicht mehr umhin, sich ein paar Gedanken darüber zu machen, wo man sich (in Relation zu anderen Optionen) in der Software-Landschaft sehen will und welche Perspektiven man (noch) sieht.

Einer "Übergangslösung" mag man noch so manche unsaubere Problemlösung nachsehen, weil es ja eine Perspektive für deren Ablösung gibt - von einer "Dauerlösung" werden die meisten aber etwas mehr erwarten. Das schließt dann auch wieder den Bogen zu Reboots und Watchdogs und AVM-Diensten mit bisher unbekanntem Zweck, die man einfach abschaltet und dann so tut, als gäbe es jetzt gar kein Problem mehr, obwohl man noch gar nicht weiß, was mit dieser Abschaltung (bzw. es ist ja im Moment wohl sogar das komplette Entfernen der dazu notwendigen Dateien) alles an Seiteneffekten verbunden ist. Bei AVM ist man ja nun auch nicht vollkommen blöd und entwickelt irgendwelche Services, die heute und in Zukunft niemand braucht - da sollte dann etwas mehr "Untersuchung", was der überhaupt machen soll, schon drin sein.

Und ein Watchdog hat ja genau die Aufgabe zu verhindern, daß das System oder ein einzelner Service auf diesem System einfach "stehenbleibt", ohne daß das jemand bemerkt und entsprechende Aktionen einleitet. Es muß ja auch irgendeinen Grund geben, warum der Watchdog hier tatsächlich auslöst - da jetzt einfach alle Watchdogs abzuschalten, hat irgendwie Ähnlichkeit mit dem Überbrücken einer (elektrischen) Sicherung, nur weil die immer wieder durchbrennt, anstatt einfach mal nach der Ursache dafür zu suchen.

fda77, 2021-01-15:
Du kannst dich gern um musl kümmern. Ich brauche es nicht, wogegen gelink wird ist egal solang es funktioniert. Für modfs klappt es mit dem statischen linken so wie es aktuell ist halt nicht so gut. Vergiss nicht dass auch dl-toolchains für alle erlaubten Kombinationen nötig sind

cnbr, 2021-02-26:
Mit der nun verfügbaren 7.25 habe ich es versucht, leider keine Änderung gegenüber der Labor. Hängt wieder in dem Teil um random: init.  Log vom Booten ist in Anhang.
[7590_07.25.all_freetz-ng-17993M_boot.log](https://github.com/Freetz-NG/freetz-ng/files/6051045/7590_07.25.all_freetz-ng-17993M_boot.log)

Source der 7.25 hatte ich nachgefragt, wurde schon bereitgestellt.

fda77, 2021-02-26:
Schade, ich hab gehofft mit der Release läuft es wieder.
Die Änderungen an kernel und busybox sind nicht so spektakulär

busybox
```
--- make/busybox/avm/07.21-7590--busybox.config.grx5	2020-12-30 05:54:39.053657734 +0100
+++ make/busybox/avm/07.25-7590--busybox.config.grx5	2021-02-22 14:14:27.000000000 +0100
-CONFIG_INSMOD=y
-CONFIG_LSMOD=y
-CONFIG_FEATURE_LSMOD_PRETTY_2_6_OUTPUT=y
-CONFIG_MODPROBE=y
-CONFIG_FEATURE_MODPROBE_BLACKLIST=y
-CONFIG_RMMOD=y
-CONFIG_FEATURE_CMDLINE_MODULE_OPTIONS=y
-CONFIG_FEATURE_CHECK_TAINTED_MODULE=y
-CONFIG_FEATURE_MODUTILS_ALIAS=y
-CONFIG_FEATURE_MODUTILS_SYMBOLS=y
-CONFIG_DEFAULT_MODULES_DIR="/lib/modules"
+CONFIG_DEFAULT_MODULES_DIR=""
-CONFIG_DEFAULT_DEPMOD_FILE="modules.dep"
+CONFIG_DEFAULT_DEPMOD_FILE=""
```

kernel
```
--- make/linux/configs/avm/config-grx5-7590_07.20	2020-11-01 06:20:37.129906111 +0100
+++ make/linux/configs/avm/config-grx5-7590_07.25	2021-02-16 18:58:32.000000000 +0100
+CONFIG_AVM_PA_SCH_TACK=m
+CONFIG_BLK_DEV_LOOP=m
+CONFIG_BLK_DEV_LOOP_MIN_COUNT=8
+CONFIG_AVM_WATCHDOG_SHIM=y
+CONFIG_WATCHDOG_CORE=y
+CONFIG_GRX500_IAP_WDT=y
+CONFIG_AVM_GRX500_IAP_WDT_NMI=y
+CONFIG_AVM_WATCHDOG_SHIM_SUPPORT=y
+CONFIG_NFS_FS=m
+CONFIG_NFS_V2=m
+CONFIG_NFS_V3=m
+CONFIG_GRACE_PERIOD=m
+CONFIG_LOCKD=m
+CONFIG_LOCKD_V4=y
+CONFIG_NFS_COMMON=y
+CONFIG_SUNRPC=m
```
Komplett:
[busybox.diff.txt](https://github.com/Freetz-NG/freetz-ng/files/6052933/busybox.diff.txt)
[kernel.diff.txt](https://github.com/Freetz-NG/freetz-ng/files/6052934/kernel.diff.txt)

fda77, 2021-02-27:
@PeterPawn liest du noch mit?
Ich hab mir das mit dem HOME mal angeschaut. 
Das Verzeichnis /mod/root wird hier gesetzt: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1098
mod -> var/mod ist ein Link im Dateisystem
/var/mod wird durch das entpacken der var.tar erstellt: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1085
Daher sollte es das Verzeichnis auch immer geben

@cnbr Du kannst mal versuchen in https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1098 das "/mod/root" durch "/" zu ersetzen
Und wenn es nicht hilft den ganzen Bereich rauswerfen
von https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1084
bis https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1122
(ich find es sehr suspekt dass die 4 .dummy Dateien von AVM gelöscht werden)

Dieser eip137 hat übrigens auch eine "root of trust" Funktion: https://www.chipestimate.com/log.php?from=%2FINSIDE-Secure-Corporation%2FEIP-123%2Fdatasheet%2Fip%2F30754&logerr=1
Vielleicht hilft es auch mal eins oder alle Kernelmodule zu löschen
 - build/original/filesystem/lib/modules/4.9.198/extra/eip123_hw.ko
 - build/original/filesystem/lib/modules/4.9.198/extra/eip123.ko
 - build/original/filesystem/lib/modules/4.9.198/extra/eip97.ko

Die Header/Sourcen dazu gibts in der hwcrypto-Datei im Sourcenpaket von avm

PeterPawn, 2021-02-28:
Ja, ab und an lese ich noch mit.

Aber ich habe den Überblick verloren - trifft das nun für alle 7590 mit 07.25 (bzw. der Labor davor) zu oder auch wieder nur für einige?

Ich hätte Ideen, wie man das Problem weiter einkreisen könnte und wo vielleicht die Ursachen für die (unterstellte) unterschiedliche Benutzung von Entropie liegen könnten - aber das ist eigentlich eine längere Geschichte (wenn, will ich sie halt auch so erzählen, daß sie Sinn macht) und mir fehlt im Moment die Zeit dafür. 

Ich habe das aber weiterhin auf dem Zettel ... wobei ich noch einmal daran erinnern will, daß die ursprüngliche Annahme, es könnte mit der Crypto-Hardware und Freetz-NG ein Problem geben, gar nicht an dieser Stelle auftauchte (hier ist es wegen der Meldungen in der SerCon ja deutlich), sondern das noch im Kontext des Abschaltens aller AVM-Watchdogs bzw. der Deaktivierung des `untrustedd` (als weniger drastischer Workaround) als "Gemeinsamkeit" der Boxen mit solchen Problemen als These in den Raum gestellt wurde (iirc war das auch in einem IPPF-Thread).

Ein weiterer Fakt ist es meines Wissens auch, daß dieses Problem offenbar durch Änderungen hervorgerufen wird (wenn es sich nicht doch auf nur wenige Einzelfälle beschränkt, womit wir wieder bei der "Systematik" und dem verlorenen Überblick wären), die für Freetz-NG spezifisch sind (ggf. auch für Freetz wären, aber das wissen wir nicht, weil Freetz 07.2x nicht unterstützt) und es bei anderen Modifikationen (z.B. hier: https://www.ip-phone-forum.de/threads/busybox-mit-telnet-in-fritz-os-7-2x.307385/post-2400543) bisher wohl noch keine Probleme gab.

Daher würde ich hier tatsächlich erst einmal den umgekehrten Weg nehmen (weil es einfacher zu realisieren ist) und Stück für Stück Komponenten aus Freetz in ein "von Hand" modifiziertes Image einbauen, bis das Problem auftritt. Beginnen würde ich mit einer (statisch gelinkten) BusyBox - damit könnte man feststellen, ob die abweichende Version der BusyBox oder die C-Library der BusyBox (aus Freetz-NG) generell daran schuld ist (dann tritt das Problem auch mit dieser BusyBox auf) oder ob es am Ende der dynamische Loader der `uClibc-ng` ist (AVM nutzt ja `musl` - ein weiterer Unterschied und m.W. nutzen dann auch die AVM-Bestandteile die BusyBox von Freetz[-NG] und damit die andere C-Library zumindest an einigen Stellen), der da zuviel Entropie "verbraucht" (das könnte z.B. dann der Fall sein, wenn ASLR verwendet werden sollte - ich weiß aber nicht sicher, ob das der Fall ist und müßte mich erst durch die Quellen/Beschreibungen wühlen).

Jedenfalls erscheint es mir (auch angesichts der Stelle, wo das Problem auftritt, denn das ist schon sehr, sehr früh in der Initialisierung des Systems) sinnvoller, nach und nach Änderungen dazuzunehmen, bis das Problem auftaucht, als Stück für Stück Freetz-Änderungen zu entfernen, bis es irgendwann verschwunden ist.

Der Vorschlag, die Routine zum "Warten auf Entropie" mal anzupassen und genauere Infos auszugeben, bleibt davon unberührt ... es wäre nämlich weiterhin interessant zu wissen, ob es (a) überhaupt einen Zuwachs an Entropie gibt oder ob der Hardware-RNG gar nicht arbeitet oder (b) erzeugte Entropie gleich wieder von irgendeinem Prozess "verbraucht" wird, weil da jemand permanent (und mit falscher Implementierung) an `/dev/[u]random` nuckelt und es daher nie bis zu den erwarteten 112 Byte (bzw. 896 Bit) als "Entropie-Vorrat" kommt, die in der `rc.hwcrypto` erwartet werden.

fda77, 2021-02-28:
Ich hab von keiner 7590 gelesen die erfolgreich bootete, daher gehe ich von allen 7590 mit 7.25 aus. Mein 7490 läuft problemlos mit der neuesten Labor
Stück für Stück Freetz deaktivieren hab ich auch schon überlegt und mir das Verzeichnis https://github.com/Freetz-NG/freetz-ng/tree/master/patches/scripts angeschaut. Da sind schon ein paar suspekte Dinge drin, zb
https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/108-patch_multid-wait.sh
Ursprünglicher commit von 2007: https://trac.boxmatrix.info/freetz-ng/browser/freetz-ng/trunk/patches/110-multid-wait.patch?rev=1415 Vielleicht braucht man den gar nicht mehr ^^ 

Die AVM Busybox half nicht wie hier zu sehen: https://github.com/Freetz-NG/freetz-ng/issues/120#issuecomment-756938962

@cnbr Kannst du dir die Entropie ausgeben lassen wie in https://github.com/Freetz-NG/freetz-ng/issues/120#issuecomment-757059543

cnbr, 2021-02-28:
Probiere ich die Tage aus, https://github.com/Freetz-NG/freetz-ng/issues/120#issuecomment-787012329 und auch das mit Entropie und die weitern Tipps von PeterPawn. Welche Module geladen sind könnte auch interessant  sein.
Da dann während der Versuche ohne Internet und Telefon der WAF im Haus schlecht wäre, muss ich noch eine Ersatzbox besorgen.

Ob es relevant ist, kann ich nicht beurteilen, bei den Sourcen von AVM ist mir aufgefallen, in der 7.25 kommt Kernel 4.9.198 zum Einsatz, im tar.gz ist linux-headers-4.9.160.

fda77, 2021-02-28:
> linux-headers-4.9.160.

die sind mir auch aufgefalle, aber egal da
 - man für  grx5 / 7590 eh keinen replace-kernel auswählen kann
 - im Minimalimage keine Kernelmodule eingebaut und geladen werden

fda77, 2021-03-01:
Es war doch was mit Busybox. Wenn man sich das Diff oben anschaut sieht man dass modprobe usw nicht mehr enthalten sind. Wenn man ins Image schaut sieht man dass dort nun ein gewisses kmod dies übernimmt (ka was das besser kann). Und wenn man sich das make Log anschaut sieht man dass dort 3 modules*bin Dateien gelöscht werden. kmod mag diese aber!
Also einfach die Schleifn https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1442 löschen

Was nicht geht: modprobe von freetz-modulen. die sind nur in der .deb und nicht in der .deb.bin. Vorerst entwerder modprobe-busybox nehmen oder mit insmod den kompletten Pfad (in der richtigen Reihenfolge) aufrufen
Ob man man die .bin erzeugen kann weiss ich nicht, da die avm Module lokal nie gebaut werden. Die .deb wird von Freetz mit https://github.com/Freetz-NG/freetz-ng/blob/master/tools/depmod.pl erstellt

freetz-ng-mod, 2021-03-01:
So meine 7590 lauft nun mit fw 7.25 ohne reboot. Was mich aber nun sehr wundert ist das ich eine Mail bekommen habe von der Box

Änderungsnotiz
Diese E-Mail wurde Ihnen von Ihrer FRITZ!Box 7590 (UI) [7590-client]  gesendet. Sie enthält Informationen über Änderungen in Ihrer FRITZ!Box  oder Ereignisse in Ihrem Heimnetz. | Diese E-Mail wurde Ihnen von Ihrer FRITZ!Box 7590 (UI) [7590-client]  gesendet. Sie enthält Informationen über Änderungen in Ihrer FRITZ!Box  oder Ereignisse in Ihrem Heimnetz.
Diese E-Mail wurde Ihnen von Ihrer FRITZ!Box 7590 (UI) [7590-client]  gesendet. Sie enthält Informationen über Änderungen in Ihrer FRITZ!Box  oder Ereignisse in Ihrem Heimnetz.

Der FRITZ!Box-Benutzer "fritz5432" wurde nach dem Update automatisch angelegt. Dieser FRITZ!Box-Benutzer nutzt das FRITZ!Box-Kennwort. Sie können sich weiterhin wie gewohnt mit Ihrem FRITZ!Box-Kennwort an Ihrer FRITZ!Box anmelden.

fda77, 2021-03-02:
Meine läuft jetzt seit 1 Tag ohne Probleme.
Also ausser denen die AVM mal wieder verursacht hat, wie sich dass sich hinter einem Hybrid Router keine Telekom VOIP Nummern mehr registrieren lassen. Andere Anbieter gehen nach wie vor. Das wurde auch schon an anderer Stelle berichtet

[2 more replies]

closed 2021-03-02 by fda77