Freetz-NG/freetz-ng #121: uclibc: bump version to `1.0.31`

JM1, 2020-11-30:
AVM's FRITZ!Box 7490 firmware `7.21` features uclibc [`1.0.31`][9], but freetz(-ng) has [`1.0.15`][10] only. Upgrading to `1.0.31` isn't trivial because [since `1.0.18`+ uclibc does not ship libpthread* and others as separate libraries anymore][1] and [uclibc's RPC implementation has been dropped upstream][2]. I did [upload a draft of how that could work][3] but I had no time to test it yet and it still requires a lot of work to make packages compatible to uclibc `1.0.31`. My patch for uclibc `1.0.31` is `HIGHLY EXPERIMENTAL and UNTESTED`!

But as you've found my commit ahead of time and already applied parts of it in commits [768bc9d][4], [e268cc1][5], [b6bcd48][6], [ca829af][7] and [894c433][8] (thanks for mentioning 😊), I do open this issue now to warn about its draft status and to track progress.

[1]: https://github.com/wbx-github/uclibc-ng/commit/29ff9055c80efe77a7130767a9fcb3ab8c67e8ce
[2]: https://gogs.waldemar-brodkorb.de/oss/uclibc-ng/commit/a1a8064169aeda79e3266a2db9cce25e361a86dc
[3]: https://github.com/JM1/freetz-ng/commit/65719505a594110df2b32e70e9a21ec85bef3d48
[4]: https://github.com/Freetz-NG/freetz-ng/commit/768bc9dde459f59574f64c9557b5960949bd24a6
[5]: https://github.com/Freetz-NG/freetz-ng/commit/e268cc1343bcef3628f121e62a8f4e672ea93f73
[6]: https://github.com/Freetz-NG/freetz-ng/commit/b6bcd4864834a6a2d4139ce5f5706c6569e44a8b
[7]: https://github.com/Freetz-NG/freetz-ng/commit/ca829af628275af305a4f539839d8d6d71eeee89
[8]: https://github.com/Freetz-NG/freetz-ng/commit/894c43389dd35f7c924a8c666f9973873760cfc6
[9]: https://osp.avm.de/fritzbox/fritzbox-7490/source-files-FRITZ.Box_7490-07.21.tar.gz
[10]: https://github.com/Freetz-NG/freetz-ng/tree/master/toolchain/make/target/uclibc

fda77, 2020-11-30:
Hallo, ich schreibe auf deutsch da dies vermutlich für uns beide einfacher ist. Hoffe das ist okay.
Nur bis uClibc 1.0.14 wird die Version von AVM ersetzt und wird von AVM und Freetz genutzt. Neuere werden ausschliesslich von Freetz genutzt. Daher hab ich auch gleich noch die 1.0.36 dazugemacht, für die .31 gibt es eigentlich keinen Grund und könnte man auch wieder löschen. Die .configs von AVM sind eh nicht aktuell.
Ich hab das ganze auf einer 7490 getestet, es scheint grundlegend zu funktionieren. Manche Packages müssen noch angepasst werden, wie libtirpc für autofs.
Um die neuen Versionen in https://github.com/Freetz-NG/freetz-ng/commit/9f503b5d54f02e3da3d4c9e86a1e8acabcfb27af auszuwählen muss man "Developer" im menuconfig ausgewählt haben (https://github.com/Freetz-NG/freetz-ng/commit/f03a3db7ca35533cf0aa0eff59835ca10b5765f3).
Falls du mit "make uclibc-menuconfig" was änderst muss unbedingt build-toolchain (https://github.com/Freetz-NG/freetz-ng/commit/337c19ab1baddf6ab68f4acb80bd33144d725683) ausgewählt werden.
Normalerweise gebe ich die Quelle/Author an ... ausser ich vergesse es ;-)

fda77, 2020-12-04:
@JM: Hast du was dagegen die 1.0.31 wieder zu löschen und nur noch die 1.0.36 zu verwenden?

JM1, 2020-12-04:
Ich hatte versucht die `uclibc` Versionen von freetz-ng auf die Version von AVM zu heben, aber nicht höher, um "Kompatibilitätsprobleme" zu vermeiden. Wenn freetz-ng beim Bauen des images die `uclibc` von AVM strikt separiert von der `uclibc` der von freetz-ng gebauten Pakete, dann spricht eigentlich nichts gegen ein Upgrade auf die aktuellste Version.

fda77, 2020-12-04:
Bei älteren Versionen wurde die uclibc ersetzt, da diese Geräte oft nur 4, 8 oder 16 MB Flash für alles haben. Aktuelle haben 44-48 fürs Filesystem. Ausserdem haben gar nicht mehr alle Geräte eine uclibc, sonder musl-libc (zB 7590) oder glibc. Dazu sind es bereits verschieden Versionen diese 3 Libs. Das wäre viel Aufwand. Und es kann immer zu Inkompatibilitäten kommen.
So wie es aktuell mit der separierten ist, hat es nur den Nachteil dass 2 libc im Flash sind. Ich lösche die .31 dann wieder

fda77, 2020-12-06:
Die Packages die ich verwendet funktionieren. Falls es mit anderen Probleme gibt wird man es bemerken

closed 2020-12-06 by fda77