Freetz-NG/freetz-ng #123: WLAN im IP-Client-Modus

bmaehr, 2020-12-05:
Hallo,

Ich bin schon seit Stunden auf der Suche nach einer Möglichkeit die Fritzbox als WLAN Router mit einem beliebigen WLAN-Netzwerk (also nicht als Repeater mit einem AVM Access Point sondern dem WLAN einer "Fremdfirma" als Verbindung zum Internet) zu betreiben. 
Da das über die AVM Oberfläche von AVM deaktiviert wurde, Netzwerke die nicht von AVM sind zu repeaten, hab ich gehofft, dass es zumindest ein freetz package gibt, das diese Möglichkeit eröffnet, auch wenn die Konfiguration evt nicht schön über die Oberfläche möglich ist.

Ich hab jedoch nichts gefunden und bin erstaunt, da es meiner Ansicht nach doch eine häufige Anforderung ist und nach meinem Verständnis mit Linux auch nicht allzuschwer zu implementieren sein sollte. Oder übersehe ich was?

fda77, 2020-12-06:
Ja, mehreres:
 - Dies ist keine Forum sondern ein Bugtracker
 - Die Option selbst im AVM Webinterface

closed 2020-12-06 by fda77

bmaehr, 2020-12-06:
Es war als Feature Request gedacht.

Nach vielen Stunden Suche in unzähligen Diskussionsforen, in denen ich Hinweise auf diese Funktion gefunden habe, konnte ich das Problem beheben: Durch die Installation der ungefreezten AVM Version. Es ist also schlussendlich ein Bug in freetz.

berndy2001, 2020-12-06:
Fehlt ein Eintrag in der Webconfig oder wie kann man sich das vorstellen?

fda77, 2020-12-06:
Nö, Anwenderfehler. Bei mir gehts nämlich, auch mit Freetz

bmaehr, 2020-12-07:
So
[image]

berndy2001, 2020-12-07:
Hab bei mir nochmals geschaut, hier sieht alles wie gewohnt aus.
Hast du schon probiert einen frischen Checkout mit Minimal-Image zu erstellen?

Conan179, 2020-12-07:
@bmaehr ferräst du auch mal WELCHE fritzbox du hast?

bmaehr, 2020-12-07:
"ferräst" ... Augenkrebs
Fritzbox 7390 (Ja, als mir klar geworden ist, dass mir da ein Feature fehlt, hätte ich daran denken sollen es dazu zu schreiben)

Conan179, 2020-12-07:
> ... Augenkrebs
Nennt sich LRS, ich bin raus, dumm anmachen lasse ich mich nicht.

bmaehr, 2020-12-07:
Es war nicht böse gemeint, nur schüttelt es mich, wenn ich sowas sehe ....