Freetz-NG/freetz-ng #124: Patch to support new default cipher AES-GCM in OpenVPN 2.5

stefanpuch, 2020-12-07:
With the release of OpenVPN 2.5 the default cipher where changed from BF-CBC to AES-GCM (see https://openvpn.net/community-downloads-2/).
In order to take support the new cipher within the OpenVPN GUI (v1 - old full featured) I extended the dropdown with the attached patch (works for me).

[openvpn-cgi-new-cipher-v25.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/5654631/openvpn-cgi-new-cipher-v25.patch.txt)

fda77, 2020-12-07:
Thanks. Does OpenVPN 2.4.9 support "GCM" ? If not it should be conditional.
The old version cant be removed for old devices: https://github.com/Freetz-NG/freetz-ng/blob/master/make/openvpn/Config.in#L29

stefanpuch, 2020-12-07:
As far as I'm aware of, YES: "OpenVPN 2.4 and newer support cipher negotiation. If both peers (client and server) support cipher negotiation, OpenVPN will default to using AES-GCM."

Source
https://community.openvpn.net/openvpn/wiki/SWEET32?__cf_chl_jschl_tk__=35f8d89038085ff90c8c8f7ca2fe6e3866c9860e-1607372914-0-Aa79s-kfnddueN5TCQ0P96DmhRAlmM8jhb-wD5YAfopRjLjU0TK6GEHuKC8mKxuwxZiNf-WrOIlTK7J1pCno-tkMXXC4sg-pX-TfatSL8AF_y6IV72bXs-ajhKTpxMry-A-AJb4DUwkF9kY9hpHS3o0uHqjKgJ6gG9CPUJSVBLoiBShbdPATjBGxlOPoliwHeWtMnfvs-0GMvglEbDgma_IzkK28al6BHzERcrdXtv0V15XSejC2vWCPri24tZANjxoNXYGCvfF8Zn99vOHDzxyBnId1DBaL8PPXE2iLnFwLanAvuBEWugPkI1-vNRpVdK-UAuT_MF62hjAZtHRIl1Y

https://openvpn.net/community-resources/reference-manual-for-openvpn-2-4/

closed 2020-12-07 by fda77

fda77, 2020-12-07:
You should try wireguard: https://freetz-ng.github.io/freetz-ng/make/wireguard
Its faster than openvpn. Also a simple server configuration, client config by qr-code

stefanpuch, 2020-12-07:
> You should try wireguard: https://freetz-ng.github.io/freetz-ng/make/wireguard

I had a short look into it but so far not enough time to test all the details and the windows client before 0.3 missed some important features like starting and stopping a tunnel without administration rights
https://lists.zx2c4.com/pipermail/wireguard/2020-November/006075.html