Freetz-NG/freetz-ng #125: Patch to read password for private keyfile to sign image from command line

stefanpuch, 2020-12-07:
In order to get freetz on my 7490 on firmware 07.21 I switched from old freetz to freetz_ng (thanks for the awsome work so far!) Unfortunately the mechanism for signing self build images is slightly different. While (old) freetz always reads the the password from the private keyfile from command line, freetz-ng expects the password to be set within the .config file. Combined with the option to include the .config file into the build image the password will be available in PLAIN TEXT on the Fritz!Box.
The following patch adds an opportunity to the user to provide the password by command line like the old freetz did before. Thus the user can decide to set one within the .config file or - when left empty - pass it on the command line during build process. The patch is inspired by the original source of PeterPawn in his "check_signed_image" script, strengthen the rsa cipher from des3 to aes128 and updates the help description accordingly.
Maybe it would be good to add a hint or warning that a password provided through .config file might be available in plain text on the build firmware if .config file is included in the image?

[fwmod_ask_private_key_passwd.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/5654735/fwmod_ask_private_key_passwd.patch.txt)

fda77, 2020-12-07:
Ask on the terminal sounds great. Shouldnt be the "read" be in a  ```while [ ${#FREETZ_FWMOD_SIGN_PASSWORD} -lt 6 ]``` loop? Openssl will fail if there is no/a short password.
I never used other signing then the "tar" version in ng. PeterPawns is slighty different than in Freetz/!. 

> Combined with the option to include the .config file into the build image the password will be available in PLAIN TEXT on the Fritz!Box.

Nether PLAIN nor ENCRYPTED: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L2028

stefanpuch, 2020-12-07:
> Ask on the terminal sounds great. Shouldnt be the "read" be in a `while [ ${#FREETZ_FWMOD_SIGN_PASSWORD} -lt 6 ]` loop? Openssl will fail if there is no/a short password.
> I never used other signing then the "tar" version in ng. PeterPawns is slighty different than in Freetz/!.

Of cause this could be an improvement. Since the fwmod script originally also fails when the user specifies to short password i didn't touch this
https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1821

> 
> > Combined with the option to include the .config file into the build image the password will be available in PLAIN TEXT on the Fritz!Box.
> 
> Nether PLAIN nor ENCRYPTED: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L2028

OK, sorry my fault, I'd just look into the local .config file in build folder and assumed it would directly put into the image

fda77, 2020-12-07:
What about this:
```
+       if [ -z "${FREETZ_FWMOD_SIGN_PASSWORD}" ]; then
+               while true; do
+                       read -sp "Enter the password for the signing key (hidden input): " FREETZ_FWMOD_SIGN_PASSWORD
+                       echo
+                       if [ -e "${SIGNATURE_SUBDIR}/prv" ]; then
+                               openssl rsa -in "${SIGNATURE_SUBDIR}/prv" -passin "pass:$FREETZ_FWMOD_SIGN_PASSWORD" -noout &>/dev/null && break || echo "Wrong password!";
+                       else
+                               [ ${#FREETZ_FWMOD_SIGN_PASSWORD} -ge 6 ] && break
+                       fi
+               done
+       fi
```

If the private key exists: ask until the password is correct,
or has at least 6 chars

stefanpuch, 2020-12-07:
I'm no expert regarding the different redirections of 
'echo' vs. 'echo "" 1>&2' 
and
'-noout &>/dev/null' vs '-noout >/dev/null 2>&1' (used in line 1837)
but I think (without testing) that should work.

fda77, 2020-12-07:
1>&2 redirects stdout to stderr. But you just want a newline after the pasword
&>/dev/null just redirect 1+2 to nowhere

closed 2020-12-08 by fda77