Freetz-NG/freetz-ng #127: [libusb] build with libusb 1.24 failed

harryboo, 2020-12-11:
Since latest change the build process hangs at libusb compilation.

Error message: configure: error: compiler with C11 support is required to build libusb

[image]

harryboo, 2020-12-11:
Will try gcc > 4.8.5, coz 4.8.x doesnt support c11

fda77, 2020-12-11:
libusb1 compiles for 7490. Maybe you are using a (unentioned...) old FOS version or which the gcc version cant bechanged easily. 
1.0.23 was okay? Then something like https://github.com/Freetz-NG/freetz-ng/commit/b3f8bdf5a99a4ea241d9957be873b0e2313ed2dc could be used

harryboo, 2020-12-11:
target toolchain options changed to 4.9.4 or higher will work with libusb1.0.24
And yes libusb1.0.23 was okay with 4.8.x

closed 2020-12-11 by fda77

fda77, 2020-12-11:
Thanks! Its not optimal, as there is no FREETZ_TARGET_GCC_4_9 MIN/MAX.
But nobody using a very old uclibc from 2012 needs a libusb from 2020 ^^

harryboo, 2020-12-12:
Ich schreibs mal in deutsch.
Nein ist nicht optimal, da die 1.0.24 nicht genommen wird obwohl ich in der target toolchain die version 5.5 unter 'target compiler' ausgewählt habe.
Da war es vorher besser da man es selbst in der Hand hatte.

fda77, 2020-12-12:
Deutsch ist schon einfacher :)
Ja, weil es von der von AVM verwendeten Version abhängt. Dein Beispiel würde die alte Version für die 7590 die gcc8 verwendet nehmen... Und wenn irgend wann noch eine neuere gcc Version von AVM verwendet wird müsste man diese Package auch wieder anpassen.
Aber da dein 7490 eh schon 7.2 hat könntest du auch einfach die verwenden. Mit neuerem gcc und damit neuerer libusb

harryboo, 2021-02-08:
Nach der Einführung von GCC_MIN und GCC_MAX könnte man auch die Bedingung von 1.0.24 (FREETZ_TARGET_GCC_5_MIN) vereinfachen?

fda77, 2021-02-08:
Ah, hier war das. Ich wusste doch dass das vor kurzem schon mal fehlte...