Freetz-NG/freetz-ng #128: [OPENSSL 1.1.1 + CURL] compilation failed

harryboo, 2020-12-14:
Compilation of curl will fail with new 'openssl 1.1.1i' library selected.

Log:
  CC       ../lib/curl-dynbuf.o
  CCLD     curl
/home/jens/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr//usr/lib/freetz/libssh2.so.1: undefined reference to `EVP_ripemd160'
collect2: error: ld returned 1 exit status

fda77, 2020-12-15:
Wenn du die Unteroption wieder deaktivierst wird zumindest curl gebaut

harryboo, 2020-12-15:
Will bei mir trotzdem nicht (7490 / FW6.93).

fda77, 2020-12-16:
Komisch

closed 2020-12-18 by fda77

harryboo, 2020-12-19:
Works...  Thanx...

fda77, 2020-12-19:
Mit der neuen openssl Version können noch einige Packages Probleme bekommen. Das wird man dann sehen :)
Deshalb ist sie wohl noch länger hinter "developer" versteckt