Freetz-NG/freetz-ng #129: [OPENSSL 1.1.1 & ShellInABox] compilation failed

harryboo, 2020-12-19:
Problem wit compilation of ShellInABox with the new OpenSSL 1.1.1:

libtool: compile:  /home/jens/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -g -std=gnu99 -Wall -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -ffunction-sections -fdata-sections -MT ssl.lo -MD -MP -MF .deps/ssl.Tpo -c libhttp/ssl.c  -fPIC -DPIC -o .libs/ssl.o
libhttp/ssl.c: In function 'sslFreeHndl':
libhttp/ssl.c:953:18: error: dereferencing pointer to incomplete type 'BIO {aka struct bio_st}'
       if (readBIO->next_bio == writeBIO) {
                  ^
make[2]: *** [Makefile:585: ssl.lo] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
libtool: compile:  /home/jens/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -g -std=gnu99 -Wall -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -ffunction-sections -fdata-sections -MT httpconnection.lo -MD -MP -MF .deps/httpconnection.Tpo -c libhttp/httpconnection.c -o httpconnection.o >/dev/null 2>&1
mv -f .deps/httpconnection.Tpo .deps/httpconnection.Plo
make[2]: Verzeichnis „/home/jens/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/shellinabox-2.14“ wird verlassen
make[1]: *** [Makefile:399: all] Fehler 2
make[1]: Verzeichnis „/home/jens/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-0.9.33.2-nptl_kernel-3.10/shellinabox-2.14“ wird verlassen

ERROR: Build failed.

Solution #1: Change to the github fork (https://github.com/shellinabox/shellinabox)

Solution #2: Patch (https://github.com/shellinabox/shellinabox/commit/d0d8c588826b0b1c570229a95d867fcccb17f32c)

fda77, 2020-12-19:
Codeblöcke zB mit ``` in einer Zeile beginnen und beenden
Am besten nimmt man wohl das aktuelle git von vor 2 Jahren, da ist auch noch eine DOS-fix drin

fda77, 2021-01-05:
This should be the patch to fix it: https://github.com/shellinabox/shellinabox/pull/477
shellinabox is a big mess. There are 5 year old unmerged pull-requests. I' not sure if it should be used

fda77, 2021-01-05:
Probier mal aus ob's läuft

closed 2021-01-05 by fda77

harryboo, 2021-01-06:
Hi, compiliert durch. Jedoch bootet die Box mit dem Image (7490 / 6,93 / OPENSSL1.1.1) nicht mehr.
Ich muss jetzt erstmal das alte System wieder lauffähig bekommen (linux_fs_start).
Ich werde erstmal mit alter OPENSSL lib probieren und dann schauen ob ich das Problem mit der neuen eingrenzen kann.

Was mir aufgefallen ist: Die Compiler-Optimierungen werden nicht in allen Paketen übernommen. da diese schon fest verdrahtet sind (zB make\busybox\avm\06.93-7490--busybox.config.vr9 : CONFIG_EXTRA_CFLAGS="-march=34kc -Wa,-march=34kc -mtune=34kc -mips16 -minterlink-mips16 -fno-reorder-blocks -O2). Kann man das mal geradeziehen?
Danke

fda77, 2021-01-06:
Compiliert hab ich's auch :]

In dem Verzeichnis liegen die Dateien die AVM benutzt! Darin gibt es auch eine README (ganz unten) https://github.com/Freetz-NG/freetz-ng/tree/master/make/busybox/avm

Wen das Gerät wegen openssl crasht, kann es nur an den Pakages liegen die du openssl nutzen und die du installiert hast. Vermutlich dann eines das automatisch beim Booten gestartet wird

harryboo, 2021-01-07:
So ShellInABox funktioniert ((7490 / 6,93 / OPENSSL1.0.2).
Test was mit 1.1.1 nicht funktioniert hat steht noch aus. Malde mich.

Auch in anderen Paketen ist '-O2' festverdrahtet bzw. wird gepatcht.

fda77, 2021-01-07:
Mit 1.1.1 hab ich es starten können. Wie man es konfiguriert um sich einloggen zu können habe ich nicht zusammengesucht, leider fehlt noch https://freetz-ng.github.io/freetz-ng/make/#shellinabox
In avm/ liegen nur unveränderte Dateien wie AVM sie nutzt, freetz hat eigene