Freetz-NG/freetz-ng #130: [compile failed]

loeffelpan, 2020-12-20:
I'm trying to compile freetz-ng for my 7490.
Without any changes on my minimal config compiling now fails with this error.
Seems to fail at the same point wether which config I'm using.
Any help appreciated.

[config.txt](https://github.com/Freetz-NG/freetz-ng/files/5721493/config.txt)

```
make[1]: Entering directory '/home/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2'

** Configuration summary for NCURSES 6.2 20200212:

       extended funcs: yes
       xterm terminfo: xterm-new

        bin directory: /usr/bin
        lib directory: /usr/lib
    include directory: /usr/include
        man directory: /usr/share/man
   terminfo directory: /usr/share/terminfo
 pkg-config directory: /usr/lib/pkgconfig

make[1]: Leaving directory '/home/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2'
cmd() { PATH="/home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };  if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc \
        DESTDIR="/home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc" \
        all install.data
building ... make[1]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc“ wird betreten
WHICH_XTERM=xterm-new \
XTERM_KBS=BS \
datadir=/usr/share \
/bin/sh ./gen_edit.sh >run_tic.sed
mkdir -p /home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/tabset
echo '** adjusting tabset paths'
** adjusting tabset paths
sed -f run_tic.sed ../misc/terminfo.src >terminfo.tmp
/bin/sh ./gen-pkgconfig
DESTDIR=/home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc \
prefix=/usr \
exec_prefix=/usr \
bindir=/usr/bin \
top_srcdir=.. \
srcdir=. \
datadir=/usr/share \
ticdir=/usr/share/terminfo \
source=terminfo.tmp \
cross_compiling=yes \
/bin/sh ./run_tic.sh
** Building terminfo database, please wait...
Running /home/freetz/freetz-ng/tools/tic to install /home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/terminfo ...

        You may see messages regarding extended capabilities, e.g., AX.
        These are extended terminal capabilities which are compiled
        using
                tic -x
        If you have ncurses 4.2 applications, you should read the INSTALL
        document, and install the terminfo without the -x option.

./run_tic.sh: 149: /home/freetz/freetz-ng/tools/tic: not found
? tic could not build /home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/terminfo
make[1]: *** [Makefile:93: install.data] Fehler 1
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
** creating ncurses.pc
** creating panel.pc
** creating menu.pc
** creating form.pc
touch pc-files
make[1]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/ncurses/ncurses.mk:83: /home/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/tabset/std] Fehler 1
```

kamasudo, 2020-12-20:
Du weisst doch laut DEB woran es liegt
```
Ist sehr dubios. Wenn ich die custom und das Oscam nicht hinzufüge, läufts durch.
```

fda77, 2020-12-20:
Du weisst doch laut DEB woran es liegt
```
Ist sehr dubios. Wenn ich die custom und das Oscam nicht hinzufüge, läufts durch.
```

closed 2020-12-20 by fda77