Freetz-NG/freetz-ng #131: 7590 7.21: Telefonie über Mobilfunk wird in Webconfig nicht angezeigt

forenuser, 2020-12-22:
Könnte vielleicht ein ähnliches Problem sein, wie bei tr069...

Bisher nutze ich einen Huawei E1750 mit aktivierter GSM/UMTS Telefonie. Dabei ist die Konfiguration **Telefonie über Mobilfunk** in der AVM Webkonfiguration unter Internet ->Mobilfunk sichtbar und kann entsprechend konfiguriert werden.

Ich habe den Stick nun durch einen 4G Systems XSStick W1208 ersetzt, welcher ebenfalls Telefonie über das Mobilfunknetz zusammen mit der Fritz!Box unterstützt. Leider ist die Konfiguration **Telefonie über Mobilfunk** in der Webkonfiguration nicht sichtbar und kann nicht konfiguriert werden. Könnte dies eine ähnliche Ursache, wie bei der tr069 Konfiguration sein?

fda77, 2020-12-23:
Ich glaube nicht dass Freetz damit was zu tun hat. Die 7430 hat einen ganz anderen Kernel, damit andere module (Treiber), evtl andere udev-regeln etc
Am besten installierst du kurz die AVM Firmware, nach dem Test kannst du über den Bootloader wieder zurückwechseln.
Es gibt auch einen Patch in Freetz der Telefonie für alle Sticks freischaltet

forenuser, 2020-12-23:
Mit Original-AVM-Firmware funktioniert die Telefonie mit dem Stick mit der 7590.

fda77, 2020-12-23:
Dann deaktivier nacheinandere alle (remove)patches bis es funktioniert. Evtl entfernt einer davon einfach die Elemente in der Webseite

forenuser, 2020-12-23:
Ich hatte nur WEBDAV und dtrace als Removal Patches. Hilft beides nicht.

berndy2001, 2020-12-23:
Schon einen frischen Checkout und nur den Box-Typ anpassen probiert?

forenuser, 2020-12-23:
Neuer Checkout, nur Box-Typ und Expertenstatus eingestellt:

- Mit Festplatte und LTE-Stick: Ins Mobilnetz eingebucht, Mobiltelefoniekonfiguration nicht sichtbar, danach Reset
- Festplatte und Stick entfernt: Noch bevor VDSL syschron ist resetet die Box

Gehe nun auf ältere funktionierende FW zurück und teste nun mit 7430 oder 7490 weiter...

Conan179, 2020-12-23:
> * Festplatte und Stick entfernt: Noch bevor VDSL syschron ist resetet die Box
Da oder nicht da?

forenuser, 2020-12-23:
Was meinst Du mit da oder nicht da? Ich musste die Box über den Bootloader wiederbeleben, da ich nicht mehr drauf gekommen bin.

forenuser, 2020-12-23:
7490 7.21 aktuelle freetz-ng Version in Minimalkonfiguration: Mobiltelefonie mit LTE Stick kann nicht konfiguriert werden.

fda77, 2020-12-23:
Die 7590 aus der Überschrift hat einer 4er Kernel
> 7490 7.21 aktuelle freetz-ng Version in Minimalkonfiguration: Mobiltelefonie mit LTE Stick kann nicht konfiguriert werden.

Einen 3er

Also geht der Stick nur mit 7430, aber nicht mit 7490 und 7590?
Ganz ehrlich, ich komm nicht mehr mit!

forenuser, 2020-12-23:
Mit der 7.21er Original AVM FW funktioniert die Telefonie mit LTE Stick bei allen 3 Boxen.
Mit einer 7.21er Freetz FW funktioniert die Telefonie mit LTE Stick bei der 7490 und der 7590 nicht und kann nicht konfiguriert werden. Die 7430 habe ich nicht mit Freetz FW geflasht und deshalb nicht damit getestet.

fda77, 2020-12-24:
Wenn ein Minimalimage auf der 7590 resettet, wie hast du dann erkannt dass Mobiltelefonie nicht geht?
Eine 7430 nutzt den gleichen Kernel (und Sourcen in Freetz) wie die 7490. Das muss sich dann ja gleich verhalten

forenuser, 2020-12-24:
Nach dem ersten Hochfahren wurde die Box erst resettet, nachdem der Stick bereits im LTE Netz eingebucht war. Ich hatte vor dem Reset noch die Möglichkeit, die Seite zu aktualisieren -> keine Konfiguration für Telefonie. Deshalb habe ich den Stick und die angeschlossene Festplatte abgesteckt. Ergebnis: die 7590 wurde schneller resettet.

fda77, 2020-12-24:
Googel mal bitte nach der Bedeutung von "reset" und benutz das Wort nur wenn es zutrifft! Wenn du mehr Strom verbrauchen willst wie der Host liefern kann, musst du halt einen Hub dazwischenstecken.
Anfängerfragen bitte ins Forum...

closed 2020-12-24 by fda77

forenuser, 2020-12-24:
Nun liegst Du komplett falsch!