Freetz-NG/freetz-ng #135: Kernel panic on Fritz!Box 7412

pascaltippelt, 2021-01-01:
Hi!

Hardware: Fritz!Box 7412 with 7412.06.30 stock image

I tried to flash a unpached, clean freetz-image without any additional packets, based on 06.30 or 06.86. Both freetz images were compiled with the freetz-linux-vm and the latest freetz-ng. Both images run in the same kernel panic, then beeing stuck in a bootloop.

`[   78.840000] kthreads 44 childs 7
[   85.340000] ar7wdt_hw_reboot: wake up task upnpd (pid= 1641):
[   90.350000] Kernel panic - not syncing: ar7wdt_hw_reboot: delayed watchdog expired
[   90.350000]
[   90.350000] [tffs_open_panic] Called. panic_handle_use: 0
[   90.350000] [TFFS3_Panic_Lock] Called.
[   90.350000] CPU: 1 PID: 1641 Comm: upnpd Tainted: P           O 3.10.73 #1
[   90.350000] Stack : 00000006 8007adc8 00000000 00000000 00000000 00000000 808abc5a 0000003e
          00000000 00000400 00000669 00000001 00000060 8747fd28 806654fc 807208a7
          00000669 00000001 808ab3f0 8747fd28 808c0000 00000001 80730ff0 8004c638
          00000006 80048850 00000000 00000000 80669124 8587b29c 8587b29c 806654fc
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 8587b210
          ...
[   90.350000] Call Trace:
[   90.350000] [<800228e0>] show_stack+0x80/0xa0
[   90.350000] [<8034bcb0>] TFFS3_Panic_Lock+0x30/0xe0
[   90.350000] [<8034bee8>] TFFS3_Open+0x188/0x2e0
[   90.350000] [<8034a340>] tffs_open_panic+0x80/0x240
[   90.350000] [<803594b0>] tffs_panic_log_printkbuf+0x290/0x2a0
[   90.350000]
[   90.350000] [panic_reinit] Called for mtd ifx_hsnand
[   90.350000] [tffs_open_panic] z_length: 0x8010
`

(Read from the serial interface from the pcb).

Please tell how I can help you to find the issue.

Best regards

Pascal from Germany

pascaltippelt, 2021-01-01:
Additional Log: https://pastebin.com/F5bR3KUm

fda77, 2021-01-02:
I searched for "TFFS3_Panic_Lock" which shows 3 matches. 
Maybe its a hardware failure: ip-phone-forum.de/threads/287117/post-2175384
And maybe it helps to use remove-patches to make the device stable with less features

pascaltippelt, 2021-01-02:
Thanks for the fast reply. I tried to create a system with freetz (without ng) for my 7412, wich works like a charm, altough the .8x version is still under devellopment. I'll read those those theads, but I think I can exclude a hardware failure.

fda77, 2021-01-02:
You are sure you are NOT using replace kernel?

pascaltippelt, 2021-01-02:
Hope this helps:

[image]

```
#
# Automatically generated file; DO NOT EDIT.
# Freetz Configuration
#
FREETZ_HAVE_DOT_CONFIG=y

#
# Freetz revision  -----------------------------------------
#

#
#  freetz-ng 17613-b9bbd0c5c master 2020-12-31
#

#
# User competence ------------------------------------------
#
FREETZ_USER_LEVEL_BEGINNER=y
# FREETZ_USER_LEVEL_ADVANCED is not set
# FREETZ_USER_LEVEL_EXPERT is not set
# FREETZ_USER_LEVEL_DEVELOPER is not set

#
# Hardware/Firmware ----------------------------------------
#
FREETZ_SERIES_ALL_CONFIG=y
FREETZ_SERIES_ALL=y

#
# T-Com hardware
#
# FREETZ_TYPE_300IP is not set
# FREETZ_TYPE_W501V is not set

#
# LAN
#
# FREETZ_TYPE_2170 is not set

#
# WLAN
#
# FREETZ_TYPE_3020 is not set
# FREETZ_TYPE_3030 is not set
# FREETZ_TYPE_3130 is not set
# FREETZ_TYPE_3131 is not set
# FREETZ_TYPE_3170 is not set
# FREETZ_TYPE_3270_V2 is not set
# FREETZ_TYPE_3270_V3 is not set
# FREETZ_TYPE_3272 is not set
# FREETZ_TYPE_3370 is not set
# FREETZ_TYPE_3390 is not set
# FREETZ_TYPE_3490 is not set

#
# WAN
#
# FREETZ_TYPE_4020 is not set
# FREETZ_TYPE_4040 is not set

#
# Fon
#
# FREETZ_TYPE_FON is not set
# FREETZ_TYPE_5010 is not set
[... 2668 lines cut ...]
FREETZ_KERNEL_VERSION_2_6_39_MIN=y
FREETZ_KERNEL_VERSION_3_MIN=y
FREETZ_KERNEL_VERSION_3_10_MIN=y
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_VERSION_3_12_MAX=y
FREETZ_KERNEL_VERSION_3_MAX=y
FREETZ_KERNEL_VERSION_4_1_MAX=y
FREETZ_KERNEL_VERSION_4_4_MAX=y
FREETZ_KERNEL_VERSION_4_9_MAX=y
FREETZ_KERNEL_VERSION_3=y
FREETZ_AVM_KERNEL_CONFIG_AREA_SIZE=64
FREETZ_AVM_UCLIBC_0_9_33=y
FREETZ_AVM_UCLIBC_NPTL_ENABLED=y
FREETZ_AVM_UCLIBC_XLOCALE_ENABLED=y
FREETZ_AVM_SOURCE_7490_06_83=y
FREETZ_AVM_SOURCE_ID="7490_06.83"
FREETZ_AVM_SOURCE_FOR_SYSTEM_TYPE_VR9=y
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="7412"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="06_8X"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y

```

berndy2001, 2021-01-02:
Can you provide the .config.compressed, please?

fda77, 2021-01-02:
As the reboot is caused by the avm-watchdog, you could use disable-watchdog-patch (developer-competence only atm) to prevent the reboot and have more time to check the system

fda77, 2021-01-02:
You wrote in OP
> I tried to flash a unpached, clean freetz-image without any additional packets

but the config contains `FREETZ_REMOVE_UPNP=y`
And the last message before watchdog
```
[   85.340000] ar7wdt_hw_reboot: wake up task upnpd (pid= 1641):
[   90.350000] Kernel panic - not syncing: ar7wdt_hw_reboot: delayed watchdog expired
```

pascaltippelt, 2021-01-02:
@berndy2001 .config.compressed contains only 1 line:
```
FREETZ_TYPE_7412=y
```

@fda77 (1st message) I will try to deactivate the watchdog tomorrow and report.

@fda77 (2nd message) this was a copy-paste-error. I tried to remove UPNP becouse I thougt it could be the reason for the crash. So I accidentially copied the .config with this first try-and-error workaround.

fda77, 2021-01-02:
After you removed upnp, is there still the same error mesage?

pascaltippelt, 2021-01-02:
@fda77 correct, with or without upnp the error remains.

pascaltippelt, 2021-01-03:
Well, it did not work. I disabled the AVM Watchdog and this was the result:

```bash
[  150.060000] Call Trace:
[  150.060000] [<800228e0>] show_stack+0x80/0xa0
[  150.060000] [<80086a74>] dequeue_task_idle+0x54/0x80
[  150.070000] [<8000cd40>] __schedule+0x4a0/0x8c0
[  150.070000] [<803b7950>] nand_get_device+0xf0/0x1e0
[  150.080000] [<803b914c>] nand_write_oob+0xac/0x160
[  150.080000] [<803532e8>] do_write+0x3c8/0x6c0
[  150.090000] [<80355700>] TFFS3_NAND_Write+0x200/0x3e0
[  150.090000] [<8034c368>] TFFS3_Write+0x88/0xe0
[  150.100000] [<80349f80>] write_segment+0x40/0xa0
[  150.100000] [<8034aa48>] tffs_release_kern+0x2a8/0x2c0
[  150.110000] [<8035941c>] tffs_panic_log_printkbuf+0x1fc/0x2a0
[  150.110000]
[  150.110000] bad: scheduling from the idle thread!
[  150.120000] CPU: 1 PID: 0 Comm: swapper/1 Tainted: P           O 3.10.73 #1
[  150.130000] Stack : 00000006 0000044c 00000000 00000000 00000000 00000000 808abc5a 0000003f
          00000000 00000000 00000000 00000001 0000000c 87c5ef78 806654fc 807208a7
          00000000 00000001 808ab3f0 87c5ef78 00000000 81cde5c0 87c5eda8 8004c638
          00000006 80048850 00000000 00000000 80669124 87c775dc 87c775dc 806654fc
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 87c77550
          ...
```

This is outputted in the console over and over again until I pull the powerplug...

Config: only "Disable AVM watchdog", happens also if I addidtionally disable UPNP.

fda77, 2021-01-04:
I've no idea why it crashes. If freetz without ng still works, you could try to find sicne when it does not work anymore. 2 years ago ng was forked, so you check out a revision 1 year ago. Depending if it works or not you taks as next a commit 6 month earlier or later. And so on

PeterPawn, 2021-01-05:
@pascaltippelt 
You should provide a listing of `/etc/init.d` directory content (file names and sizes are of interest) and the (file) content of `/etc/init.d/rc.tail.sh` from your `modified/filesystem` folder (build system).

The watchdog has to be disabled by a line from the mentioned `rc.tail.sh` file - at the very beginning of it. Usually Freetz initialization would start AFTER this file (due to its name - `E99-zzz-rcmod` - and the assumption, that globbing results are sorted alphabetically) ... there are only little chances, that very early differences in the `init` process (between Freetz and Freetz-NG) could affect the `rc.tail.sh` file processing - but it may happen.

A "binary search" for the commit, which introduced the occurred problem, should not be required - try to find any reason for `init` to finish its processing before `S99-tail` was processed (if this file exists AND it calls `/etc/init.d/rc.tail.sh` AND this file on the other hand contains the command to disable the `init` watchdog).

One possible reason are shell `exit` commands from a "sourced" include file in all `init` files starting with letter `S` in their names. Uppercase letters (E and S) are for "startup" files, files (file names) starting with `S` will be sourced with a `.` (dot) command and `E` files are run with their own shell instance - where an `exit` command doesn't harm the `init` process). Another possible reason (to give up/abort `init` processing) are syntax errors from script files modified by Freetz-NG - there were no regression tests (at least no published ones) after the changes to support 07.19+ versions (with a completely changed initialization) have been applied/published.

pascaltippelt, 2021-01-05:
@PeterPawn 
Thanks for your reply. I copied the requested listing:

ls -hl of /etc/init.d

```bash
insgesamt 348K
-rwxrwxrwx 1 freetz freetz   11 Aug 21  2019 E05-basemodules
-rwxrwxrwx 1 freetz freetz  800 Aug 21  2019 E15-device
-rwxrwxrwx 1 freetz freetz  228 Aug 21  2019 E40-avmipc
-rwxrwx--- 1 freetz freetz 2,3K Aug 21  2019 e40-dsl
-rwxrwx--- 1 freetz freetz 6,8K Aug 21  2019 E40-dsl
-rwxrwxrwx 1 freetz freetz   37 Aug 21  2019 e46-net
-rwxrwxrwx 1 freetz freetz 3,2K Aug 21  2019 E46-net
-rwxrwxrwx 1 freetz freetz  246 Aug 21  2019 E47-voip
-rwxr-xr-x 1 freetz freetz   90 Jan  3 09:53 E99-zzz-rcmod
-rw-r--r-- 1 freetz freetz  553 Jan  1 19:05 loglibrc
-rw-r--r-- 1 freetz freetz 2,9K Jan  1 19:05 modlibinetd
-rw-r--r-- 1 freetz freetz  12K Jan  1 19:05 modlibrc
-rwxrwxrwx 1 freetz freetz  698 Aug 21  2019 rc.audiod.sh
-rwxr-xr-x 1 freetz freetz  847 Jan  1 19:05 rc.authorized-keys
-rwxrwxrwx 1 freetz freetz  296 Aug 21  2019 rc.chrony
-rwxrwxrwx 1 freetz freetz  19K Jan  3 09:53 rc.conf
-rwxr-xr-x 1 freetz freetz  825 Jan  1 19:05 rc.crond
-rwxrwxrwx 1 freetz freetz  700 Aug 21  2019 rc.down
-rwxr-xr-x 1 freetz freetz 1,7K Jan  1 19:05 rc.dropbear
-rwxr-xr-x 1 freetz freetz 2,3K Jan  1 19:05 rc.dsld
-rwxrwxrwx 1 freetz freetz 6,3K Aug 21  2019 rc.dsl.sh
-rwxr-xr-x 1 freetz freetz 1,4K Jan  3 09:53 rc.inetd
-rwxr-xr-x 1 freetz freetz 6,8K Jan  3 09:53 rc.mod
-rwxr-xr-x 1 freetz freetz 3,9K Jan  1 19:05 rc.multid
-rwxrwxrwx 1 freetz freetz  12K Jan  3 09:53 rc.net
-rwxrwxrwx 1 freetz freetz  353 Aug 21  2019 rc.pictured.sh
-rwxrwxrwx 1 freetz freetz  272 Aug 21  2019 rc.preaudio.sh
-rwxrwxrwx 1 freetz freetz 2,1K Jan  3 09:53 rc.S
-rwxr-xr-x 1 freetz freetz 2,6K Jan  3 09:53 rc.syslogd
-rwxrwxrwx 1 freetz freetz 2,6K Jan  3 09:53 rc.tail.sh
-rwxrwxrwx 1 freetz freetz 1,2K Aug 21  2019 rc.voip
-rwxr-xr-x 1 freetz freetz 1012 Jan  1 19:05 rc.webcfg
-rwxrwxrwx 1 freetz freetz 2,5K Aug 21  2019 rc.wlan
-rwxrwxrwx 1 freetz freetz 8,0K Aug 21  2019 rc.wlan_lifetest
-rwxrwxrwx 1 freetz freetz  797 Aug 21  2019 S00-signal
-rwxrwxrwx 1 freetz freetz 2,8K Jan  3 09:53 S01-head
-rwxrwxrwx 1 freetz freetz   31 Aug 21  2019 S02-led
-rwxrwxrwx 1 freetz freetz  158 Aug 21  2019 S05-watchdog
-rw-r--r-- 1 freetz freetz  438 Jan  3 09:53 S06-logging
-rwxrwxrwx 1 freetz freetz  12K Jan  3 09:53 S08-tffs
-rwxrwxrwx 1 freetz freetz   89 Aug 21  2019 S09-config
-rwxr-xr-x 1 freetz freetz  130 Aug 21  2019 S09-rtc
-rwxrwxrwx 1 freetz freetz  420 Aug 21  2019 S10-html
-rwxrwxrwx 1 freetz freetz 1,5K Aug 21  2019 S10-vm
-rwxrwxrwx 1 freetz freetz  37K Aug 21  2019 S11-config_def
-rwxr-xr-x 1 freetz freetz  230 Aug 21  2019 S11-piglet
-rwxrwxrwx 1 freetz freetz 5,6K Aug 21  2019 S12-default
-rwxrwxrwx 1 freetz freetz 4,1K Jan  3 09:53 S15-filesys
-rwxr-xr-x 1 freetz freetz  241 Aug 21  2019 S17-capi
-rwxrwxrwx 1 freetz freetz 2,1K Aug 21  2019 S17-tam
-rwxrwxrwx 1 freetz freetz   40 Aug 21  2019 S19-debugfs
-rwxr-xr-x 1 freetz freetz   43 Jan  3 09:53 S20-modload
-rwxrwxrwx 1 freetz freetz  300 Aug 21  2019 S21-udev
-rwxrwxrwx 1 freetz freetz   11 Aug 21  2019 S30-ledcfg
-rwxrwxrwx 1 freetz freetz  554 Aug 21  2019 S44-hostname
-rwxrwxrwx 1 freetz freetz   89 Aug 21  2019 S46-usb
-rwxrwxrwx 1 freetz freetz  257 Aug 21  2019 S48-configd
-rwxrwxrwx 1 freetz freetz  684 Aug 21  2019 S73-capitcp
-rwxrwxrwx 1 freetz freetz  304 Aug 21  2019 S76-media
-rwxrwxrwx 1 freetz freetz   69 Aug 21  2019 S79-boxnotify
-rwxrwxrwx 1 freetz freetz   10 Aug 21  2019 S85-apps
-rwxrwxrwx 1 freetz freetz  101 Aug 21  2019 S99-tail
```
[... 28 lines cut ...]
if test -x /usr/bin/ethnator ; then
/usr/bin/ethnator -d /etc/init.d/linkdown.sh -u etc/init.d/linkup.sh
fi
#########################################################################
## cleanup - if running, stop debug (0 normal, 1 flush buffer)
#########################################################################
if `ps | grep -v grep | grep -q "cat /dev/debug"` ; then
echo Info: have to stop 'cat /dev/debug'.
echo AVMDBG_EOF 1 >/dev/debug
fi;
#########################################################################
## PTEST: warten, bis der laufenende WLAN-Lifetest beendet ist
#########################################################################
if [ -n "$PTEST_WAIT_PID" ] ; then
wait $PTEST_WAIT_PID
fi
#########################################################################
## modulemem: mit 'fork' <set_m_sleep> Minuten warten, bis alle module gestartet sind.
#########################################################################
if [ -x "/bin/set_modulemem" ] ; then
set_m_sleep=$((10*60))
nohup sh -c "echo \"\$0[\$\$]: ++++fork set_modulemen, sleep ${set_m_sleep}++++\" > /dev/console ; sleep ${set_m_sleep}; echo \"\$0[\$\$]: ++++do set_modulemen++++\" > /dev/console; /bin/set_modulemem;" &
fi
#########################################################################
```

PeterPawn, 2021-01-05:
`rc.tail.sh` looks good - line 5 should disable the watchdog, the last line from original (containing a `exit 0` statement) was removed.

The file sizes from the directory listing are useless ... as mentioned, names and sizes matter (for comparision with original values to detect, whether they have been changed or not) and any rounded values (like `2,1K`) are too imprecise for such a comparision.

Up to this point, there's no obvious reason, why line 5 was not executed (or had not the expected effect).

You could add a line `set -x` to `/etc/init.d/rc.S` to debug executed shell commands/scripts during initialization to detect, how far the progress goes on. 

Another option would be the (additional) redirection of file descriptor 2 (STDERR) of `init` process to some persistent storage for later inspection ... but this is only possible after `/var/media/ftp` was mounted by `/etc/init.d/S15-filesys` - there's no other persistent (writable) storage on a 7412 device.

In some (rather rare) cases, errors on a 7412 device are even triggered by a connected serial line. My theory (after watching the problem from time to time) is, that there a stint draft of circuit was used, where the serial console and the serial port of the FPGA from Xilinx are connected together (there's only one place with required drill holes on PCB and I would bet, you've used this, too) and a firmware upload to FPGA is done via this path. If a terminal program is listening/active at the same time and this tends to react on transmitted data (e.g. if they form valid control sequences for RS232 connections), the firmware upload may fail ... with some later problems, if a daemon/driver can not connect (correctly) to the FPGA.

At this time, I would not believe, the connected terminal is the real reason ... but you should give it a try, if no other idea is remaining. Start the box without connected serial line and attach it later, after the WLAN led starts to blink (or even later, but within the `init` watchdog timeout) - at this point the `piglet` driver should have finished its firmware upload.

edeso, 2021-06-09:
@pascaltippelt maybe wanna try my patch in https://github.com/Freetz-NG/freetz-ng/issues/325 . current checkout with it applied comes up for me.

closed 2022-05-02 by fda77