Freetz-NG/freetz-ng #136: Support for eventhandler in package smstools3

cybermaus, 2021-01-02:
As also mentioned [here in IP-Phone](https://www.ip-phone-forum.de/threads/freetz-ng-smstools3-proposed-improvement-for-sms-eventhandler.308931/), user @opto suggested I opened a case with a patch

This patch allows for an smstools3 eventhandler script to be configured in Freetz, through the web page.
It is working well for me on a 7560 and a 7362SL. I also tested that if you leave the field empty, smstools3 works as before.

Patch creation:
```
git diff > smstools3_config_eventhandler.patch
```

Patch content:
```
diff --git a/make/smstools3/files/root/etc/default.smstools3/smstools3.cfg b/make/smstools3/files/root/etc/default.smstools3/smstools3.cfg
index 588af4d22..ad6c66059 100644
--- a/make/smstools3/files/root/etc/default.smstools3/smstools3.cfg
+++ b/make/smstools3/files/root/etc/default.smstools3/smstools3.cfg
@@ -3,3 +3,4 @@ export SMSTOOLS3_DIR=''
 export SMSTOOLS3_DEVICE='/dev/ttyUSB0'
 export SMSTOOLS3_PIN=''
 export SMSTOOLS3_LOGLEVEL='5'
+export SMSTOOLS3_EVENTS=''
diff --git a/make/smstools3/files/root/etc/default.smstools3/smstools3_conf b/make/smstools3/files/root/etc/default.smstools3/smstools3_conf
index 5a43ac08b..24450374c 100755
--- a/make/smstools3/files/root/etc/default.smstools3/smstools3_conf
+++ b/make/smstools3/files/root/etc/default.smstools3/smstools3_conf
@@ -4,6 +4,7 @@ cat << EOF
 devices = GSM
 logfile = /var/log/smsd.log
 loglevel = $SMSTOOLS3_LOGLEVEL
+eventhandler = $SMSTOOLS3_EVENTS
 
 [GSM]
 device = $SMSTOOLS3_DEVICE
diff --git a/make/smstools3/files/root/usr/lib/cgi-bin/smstools3.cgi b/make/smstools3/files/root/usr/lib/cgi-bin/smstools3.cgi
index dbbd92448..a19120e10 100755
--- a/make/smstools3/files/root/usr/lib/cgi-bin/smstools3.cgi
+++ b/make/smstools3/files/root/usr/lib/cgi-bin/smstools3.cgi
@@ -18,10 +18,12 @@ sec_end
 sec_begin '$(lang de:"Konfiguration" en:"Configuration")'
 cgi_print_textline_p "dir"      "$SMSTOOLS3_DIR"      50/255 \
   "$(lang de:"Datenverzeichnis" en:"data directory"): "
+cgi_print_textline_p "events"   "$SMSTOOLS3_EVENTS"   50/255 \
+  "$(lang de:"Event-Handler" en:"event handler"): "
 cgi_print_textline_p "device"   "$SMSTOOLS3_DEVICE"   15/255 \
   "$(lang de:"UMTS-Ger&auml;t" en:"UMTS device"): "
 cgi_print_textline_p "pin"      "$SMSTOOLS3_PIN"      8      \
-  "PIN: " " ($(lang de:"leer lassen wenn deaktiviert" en:"leafe empty if deactivated"))"
+  "PIN: " " ($(lang de:"leer lassen wenn deaktiviert" en:"leave empty if deactivated"))"
 cgi_print_textline_p "loglevel" "$SMSTOOLS3_LOGLEVEL" 1      \
   "$(lang de:"Loglevel" en:"Log level"): "
 sec_end
```

closed 2021-01-02 by fda77

fda77, 2021-01-02:
Thanks!
If you want you could add an example and a reminder to yourself to https://freetz-ng.github.io/freetz-ng/make/smstools3
This is linked by the "help"  in the freetz-webif. There is an "Improve" on the bottom, and no german/english restriction

As my ippf account is locked again... if you want to add a vid to a module you could patch the kernel in source/kernel.... after "make kernel-dirclean" + "make kernel-unpacked" and later put the patch to make/linux/patches/...
If another device contains the vids, the kernel may be newer! Only the version used by avm could be used to load the closed-source drivers for devt, wlan etc. Freetz has a patch to add vids for ftdi-device, which could help you.
Replace kernel should not be needed to just change a module. You checked if the module already exists by avm? May it is enough to load the module with some parameters to detect unknown vids

cybermaus, 2021-01-02:
Hmm, I had not realised I could recompile the AVM module, without replacing kernel... I will play with it just to see if I understand what you say, and if it builds. But I cannot test anymore: the devices are now "deployed" (it is actually family who lives on the house boat) so unless if something fails, I will leave them alone.

I will add my forward handler as example.

fda77, 2021-01-02:
To build a module the kernel needs to be compiled which builds all modules. But you copy only to used the desired files. In most cases the kernel does not need to be replaced. You could even build a module und copy it manualy to the device and load it to test
But the kernelversion has to be always the same!

cybermaus, 2021-01-02:
No sources are available for the 7560 with V7.1x, Freetz is using 7583 V7.1x as reference. But those would likely be the same kernel version and thus may work. It means I would not even need to force a rebuild recompile, but instead build for a 7583 with kernel recompile and salvage and inject the module.ko from that?

fda77, 2021-01-02:
In Freetz some devices use same sources as the have the same cpu, defined in config/avm/hardware.in
As make/linux/patches/3.10.104/7583_07.10/000-7583.07.10-7590.07.12-delta.patch is applied, it even uses newer 7590 sources
But select always your device (7560), maybe there are device-specific patches
On "make" not everything is rebuild. If you put the kernel modul file to a remote device and "insmod" it by terminal, in the worst cast the power hast to be removed for a short time to reboot...