Freetz-NG/freetz-ng #137: nfs-utils und portmap kompilieren nicht mehr

e6e7e8, 2021-01-02:
Seit commit dd74b0a "remove uclibc 1.0.15" kann ich nfs-utils und portmap nicht mehr kompilieren. Die rpc-Header werden, obwohl installiert, nicht gefunden. Getestet auf Ubunto 20.04 LTS. Installation von tirpc hilft auch nicht weiter. Dies betrifft sowohl die 7490 als auch die 7590. Vorher gab's keine Probleme.

fda77, 2021-01-02:
Die rpc header sind installiert? Wie? Wohin?
Eigentlich hat die neue uclibc gar kein rpc support mehr, wodurch das wohl ausgelöst wird. libtirpc auswählen bringt nur etwas wenn die Packages die auch verwenden. Und falls ja muss man das den Packages auch sagen, wie bei autofs: a8abe5e22c6907f3ccd9c4c422dbfc472be46809)

e6e7e8, 2021-01-03:
Die rpc-header sind installiert. Bei Ubuntu 20.04 in /usr/lib/rpc, bei fedora f33 in /usr/include/tirpc/. Sie werden von uclibc-1.0.0.36 nur nicht gefunden. Auf der anderen Seite sind sie in uclibc-1.0.0.15 noch vorhanden und daher funktioniert das Kompilieren damit. Die Lösung wäre also entweder uclibc-1.0.0.15 wieder freizuschalten oder einen patch wie für autofs a8abe5e, da autofs ja auch auf die rpc-Header angewiesen ist. Letzteres ist leider jenseits meiner Expertise.

fda77, 2021-01-04:
Welche Libs auf deinem PC installiert sind ist egal. Diese können nur auf der CPU deines PCs laufen, die meist eine andere wie in der Fritzbox ist. Deshalb muss man die von Freetz compilierte nehmen.
portmap wird man wohl nicht mehr verwenden können. Es muss sich jemand finden der "rpcbind" als Package in Freetz einbaut.
Du kannst übrigens auch den "Tag" auschecken (siehe NEWS), der wurde vor dem Entfernen der uralten uclibc erstellt.

edeso, 2021-01-11:
@fda77 nach nem patch von make/portmap.mk ala trickle mit libtirpc compiliert, portmap weiter, stirbt aber später an

```
building ... make[1]: Entering directory '/home/user/freetz-ng-7530/source/target-arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/portmap-6.0'
/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc -MM -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc -DHOSTS_ACCESS -DFACILITY=LOG_DAEMON -DIGNORE_SIGCHLD *.c > .depend
/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc -DHOSTS_ACCESS -DFACILITY=LOG_DAEMON -DIGNORE_SIGCHLD -DHOSTS_ACCESS -DCHECK_PORT -DIGNORE_SIGCHLD	 -DFACILITY=LOG_DAEMON   -c -o portmap.o portmap.c
/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc -DHOSTS_ACCESS -DFACILITY=LOG_DAEMON -DIGNORE_SIGCHLD -DHOSTS_ACCESS -DCHECK_PORT -DIGNORE_SIGCHLD	 -DFACILITY=LOG_DAEMON   -c -o pmap_check.o pmap_check.c
In file included from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/svc.h:463:0,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/rpc.h:63,
                 from portmap.c:79:
portmap.c: In function 'reg_service':
/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/svc_soc.h:53:26: warning: passing argument 1 of 'check_default' from incompatible pointer type [-Wincompatible-pointer-types]
 #define svc_getcaller(x) (&(x)->xp_raddr)
                          ^
portmap.c:445:17: note: in expansion of macro 'svc_getcaller'
   check_default(svc_getcaller(xprt), rqstp->rq_proc, (u_long) 0);
                 ^
In file included from portmap.c:130:0:
pmap_check.h:5:12: note: expected 'struct sockaddr_in *' but argument is of type 'struct sockaddr_in6 *'
 extern int check_default(struct sockaddr_in *addr,
            ^
In file included from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/svc.h:463:0,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/rpc.h:63,
                 from portmap.c:79:
/home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/svc_soc.h:53:26: warning: passing argument 1 of 'check_setunset' from incompatible pointer type [-Wincompatible-pointer-types]
 #define svc_getcaller(x) (&(x)->xp_raddr)
                          ^
pmap_check.h:29:17: note: in expansion of macro 'svc_getcaller'
  check_setunset(svc_getcaller(xprt),proc,prog,port)
                 ^
portmap.c:460:9: note: in expansion of macro 'CHECK_SETUNSET'
    if (!CHECK_SETUNSET(xprt, ludpxprt, ltcpxprt,
         ^
In file included from portmap.c:130:0:
pmap_check.h:13:1: note: expected 'struct sockaddr_in *' but argument is of type 'struct sockaddr_in6 *'
 check_setunset(struct sockaddr_in *addr, u_long  proc,
 ^
In file included from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/bits/byteswap-common.h:34:0,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/bits/byteswap.h:1,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/byteswap.h:22,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/endian.h:59,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/sys/types.h:216,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/types.h:41,
                 from /home/user/freetz-ng-7530/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.4/arm-linux-uclibcgnueabi/include/tirpc/rpc/rpc.h:38,
                 from portmap.c:79:
portmap.c:488:32: error: 'struct sockaddr_in6' has no member named 'sin_port'
      (ntohs(svc_getcaller(xprt)->sin_port)
                                ^
[... cut ...]  [-Wincompatible-pointer-types]
 #define svc_getcaller(x) (&(x)->xp_raddr)
                          ^
portmap.c:750:20: note: in expansion of macro 'svc_getcaller'
  if (!check_callit(svc_getcaller(xprt), 
                    ^
In file included from portmap.c:130:0:
pmap_check.h:19:12: note: expected 'struct sockaddr_in *' but argument is of type 'struct sockaddr_in6 *'
 extern int check_callit(struct sockaddr_in *addr, u_long  proc,
            ^
make[1]: *** [<builtin>: portmap.o] Error 1
```

fda77, 2021-01-12:
Die neueste Portmap Version die auch in Freetz ist ist 14 Jahre alt
Daher denke ich nicht dass man da noch Zeit reinstecken sollte, sondern lieber in den Nachfolger rpcbind: http://www.linuxfromscratch.org/blfs/view/svn/basicnet/rpcbind.html

fda77, 2021-01-14:
Die NFS-Utils könnten auch mal aktualisiert werden, dafür wird rpcsvc-proto benötigt.

closed 2021-01-14 by fda77

e6e7e8, 2021-01-15:
nfs-utild und rpcbind kompilieren wunderbar. Vielen Dank. Leider funktioniert jetzt der nfs-Service nicht mehr (getestet auf 7490). Er started ok, aber "Mount-point not found" sagt mein Laptop. Beim restart von nfs gibt es folgende Meldungen in syslog:

`Jan 15 09:26:53 fb7490 daemon.err rpcbind[11791]: cannot get local address for udp: Servname not supported for ai_socktype
Jan 15 09:26:53 fb7490 daemon.err rpcbind[11791]: cannot get local address for tcp: Servname not supported for ai_socktype
Jan 15 09:26:53 fb7490 daemon.err rpcbind[11791]: cannot get local address for udp6: Servname not supported for ai_socktype
Jan 15 09:26:53 fb7490 daemon.err rpcbind[11791]: cannot get local address for tcp6: Servname not supported for ai_socktype
Jan 15 09:26:53 fb7490 daemon.notice mountd[11805]: Version 1.3.3 starting`

Mit rpcinfo bekomme ich

> program version netid     address              service    owner
100000    4    local     /var/run/rpcbind.sock      -          superuser
100000    3    local     /var/run/rpcbind.sock      -          superuser
100003    3    tcp       0.0.0.0.7.255                   -          superuser

Auf meinem Laptop würde unter "service" jeweils "portmapper" stehen.

fda77, 2021-01-15:
Versuch mal eine Revision vor https://github.com/Freetz-NG/freetz-ng/commit/44b96badcf3e0b0641786c8185cfeb1d501988a5 oder nenne die Datei von Hand wieder um. Zur Sicherheit noch ein `make rpcbind-dirclean` und dann wieder make+flashen.
Ich konnte /tmp freigeben, mounten und beschreiben

e6e7e8, 2021-01-15:
Danke, so geht's.

fda77, 2021-01-15:
So, hoffe jetzt geht alles