Freetz-NG/freetz-ng #140: Unknown checksum algorithm for a hash length of 1 für uClibc

rusagent, 2021-01-04:
Hey,

ich baue das freetz-ng genau nach Anleitung mit dem gestelltem Virtualbox freetz-ng Image für die FB Fon WLAN 7170.

Und es tritt am Ende immer dieser Fehler hier auf:

freetz@freetz-linux:~/freetz-ng$ make

```
--2021-01-04 21:59:26--  https://freetz-ng.dropfiles.net/mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.tar.lzma
Auflösen des Hostnamens freetz-ng.dropfiles.net (freetz-ng.dropfiles.net) … 85.13.146.62
Verbindungsaufbau zu freetz-ng.dropfiles.net (freetz-ng.dropfiles.net)|85.13.146.62|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 9048410 (8,6M) [application/x-tar]
Wird in »dl/mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.tar.lzma« gespeichert.

mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.t 100%[========================================================================================================================================>]   8,63M  2,60MB/s    in 3,3s

2021-01-04 21:59:30 (2,60 MB/s) - »dl/mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.tar.lzma« gespeichert [9048410/9048410]

Download succeeded - "https://freetz-ng.dropfiles.net/mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.tar.lzma"  ->  saved to folder "dl"
ERROR: Unknown checksum algorithm for a hash length of 1: SHA256:=055f3bbf15c0c9d95d7663f85e402077da0226d2e966f00b1b752cff1047bdef
make: *** [toolchain/make/download-toolchain.mk:27: dl/mipsel_gcc-4.6.4_uClibc-0.9.29-freetz-r17470-shared-glibc.tar.lzma] Fehler 1

```

Nen Plan woran das liegt und wie ich den beheben kann? Habe schon sicher 3h nach ner Lösung gesucht aber nix gefunden...

fda77, 2021-01-05:
Das ist ein Sicherheitsfeature :) Wenn keine Checksumme hinterlegt ist wird ein ungültiges X genommen. Die Fehlermeldung könnte man noch optimieren

closed 2021-01-05 by fda77

rusagent, 2021-01-05:
Alright - ich probiers gleich mal aus. Danke dir für die schnelle Hilfe!