Freetz-NG/freetz-ng #141: nach "make clean" braucht busybox eine neue config

ixs, 2021-01-05:
Mit git head führt ein make dazu, dass busybox scheinbar eine neue config braucht und diese interaktiv abfragt.

```
vagrant@freetz-ng:~/freetz-ng$ make
cmd() { PATH="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; 	if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1 \
	CC="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CROSS_COMPILE="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-" EXTRA_CFLAGS="-march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" ARCH="mips"
make[1]: Entering directory '/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1'
scripts/kconfig/conf -s Config.in
#
# using defaults found in .config
#
*
* Restart config...
*
*
* Settings
*
Enable compatibility for full-blown desktop systems (8kb) (DESKTOP) [N/y/?] n
Provide compatible behavior for rare corner cases (bigger code) (EXTRA_COMPAT) [N/y/?] (NEW) ^Cmake[3]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/scripts/kconfig/Makefile:40: silentoldconfig] Interrupt
make[2]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/Makefile:444: silentoldconfig] Interrupt
make[1]: *** [Makefile:522: include/autoconf.h] Interrupt
make: *** [make/busybox/busybox.mk:49: source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/busybox] Interrupt

vagrant@freetz-ng:~/freetz-ng$
```

Mit `git checkout c9b37939f75781f0d792fad09a9ebdbf9f9b6e19` (der commit vor dem busybox 1.32 upgrade) hingegen läuft make sauber durch und fragt nicht interaktiv eine neue config ab.
Das ganze ist hier durch ein `make clean` reproduzierbar.

fda77, 2021-01-05:
Eigentlich sollte die .config automatisch aktualisiert werden wenn man ins menuconfig geht und speichert.
Geschiegt das auch nach einem `make olddefconfig`?

ixs, 2021-01-05:
Hier das Verhalten nach `make menuconfig`(save und exit).

```
vagrant@freetz-ng:~/freetz-ng$ make
ERROR: The .config file needs to be updated, because
       you have either updated to a newer revision
       or altered one of the menuconfig files manually.
       Please run 'make olddefconfig'.
make: *** [Makefile:505: check-dot-config-uptodateness] Error 3
vagrant@freetz-ng:~/freetz-ng$ make menuconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

vagrant@freetz-ng:~/freetz-ng$ make
cmd() { PATH="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; 	if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1 \
	CC="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CROSS_COMPILE="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-" EXTRA_CFLAGS="-march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" ARCH="mips"
make[1]: Entering directory '/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1'
scripts/kconfig/conf -s Config.in
#
# using defaults found in .config
#
*
* Restart config...
*
*
* Settings
*
Enable compatibility for full-blown desktop systems (8kb) (DESKTOP) [N/y/?] n
Provide compatible behavior for rare corner cases (bigger code) (EXTRA_COMPAT) [N/y/?] (NEW) ^Cmake[3]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/scripts/kconfig/Makefile:40: silentoldconfig] Interrupt
make[2]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/Makefile:444: silentoldconfig] Interrupt
make[1]: *** [Makefile:522: include/autoconf.h] Interrupt
make: *** [make/busybox/busybox.mk:49: source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/busybox] Interrupt

vagrant@freetz-ng:~/freetz-ng$
```

Und das selbe nach `make olddefconfig`

```
vagrant@freetz-ng:~/freetz-ng$ make olddefconfig
#
# No change to .config
#
vagrant@freetz-ng:~/freetz-ng$ make
cmd() { PATH="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; 	if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1 \
	CC="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CROSS_COMPILE="/home/vagrant/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-" EXTRA_CFLAGS="-march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" ARCH="mips"
make[1]: Entering directory '/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1'
scripts/kconfig/conf -s Config.in
#
# using defaults found in .config
#
*
* Restart config...
*
*
* Settings
*
Enable compatibility for full-blown desktop systems (8kb) (DESKTOP) [N/y/?] n
Provide compatible behavior for rare corner cases (bigger code) (EXTRA_COMPAT) [N/y/?] (NEW) ^Cmake[3]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/scripts/kconfig/Makefile:40: silentoldconfig] Interrupt
make[2]: *** [/home/vagrant/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/Makefile:444: silentoldconfig] Interrupt
make[1]: *** [Makefile:522: include/autoconf.h] Interrupt
make: *** [make/busybox/busybox.mk:49: source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/busybox-1.32.1/busybox] Interrupt
```

Ich kann es auch mit einem frischen `git clone` replizieren: `git clone`, `make menuconfig` (save&exit), `make` kompiliert fröhlich drauf hin und bei busybox dann "restart config..."

ixs, 2021-01-05:
Ein Blick in das busybox makefile lässt vermuten, dass wir über folgenden Teil des Makefiles laufen:

```
# If .config is newer than include/autoconf.h, someone tinkered
# with it and forgot to run make oldconfig.
# If kconfig.d is missing then we are probarly in a cleaned tree so
# we execute the config step to be sure to catch updated Kconfig files
include/autoconf.h: .kconfig.d .config $(wildcard $(srctree)/*/*.c) $(wildcard $(srctree)/*/*/*.c) | gen_build_files
        $(Q)$(MAKE) -f $(srctree)/Makefile silentoldconfig
 ```

fda77, 2021-01-05:
Ich hatte diesen Fehler noch nie. Geht die Uhr an deinem Computer richtig? Die Busybox wurde vor kurzem nur von 1.32.0 auf 1.32.1 aktualisiert, das waren nur ein Bugfix. Da die .0 bei Busybox immer unstable ist war damit die 1.31.x nicht mehr nötig. Die 1.32 nutze ich schon seit einem halbe Jahr. Von diesem "Restart config" hat vor kurzem schonmal jemand berichtet, der hat den PC gewechselt und dort lief es, das war ein Ubuntu LTS von 2016.
Vielleicht fehlt bei der Busybox ein "touch" wie es verschieden andere Packages auch machen:
```
grep touch make/*/*mk|grep 2000
make/inotify-tools/inotify-tools.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 configure.ac aclocal.m4;
make/iptables/iptables.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.ac;
make/madplay/madplay.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.ac;
make/quagga/quagga.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.ac;
make/shellinabox/shellinabox.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.ac ./Makefile.am;
make/slurm/slurm.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.in;
make/tmux/tmux.mk:$(PKG)_PATCH_POST_CMDS += touch -t 200001010000.00 configure.ac;
make/tor/tor.mk:$(PKG)_PATCH_POST_CMDS += touch -t 200001010000.00 ./configure.ac;
make/trickle/trickle.mk:$(PKG)_CONFIGURE_PRE_CMDS += touch -t 200001010000.00 ./configure.in ./config.h.in ./acconfig.h;
```

ixs, 2021-01-05:
Das ist ne Vagrant VM... Die Uhrzeit ist okay...

Jetzt um 19:14 sagt sie `Tue 05 Jan 2021 06:14:40 PM GMT`. Das ist in Ordnung für die Zeitzone:
```
vagrant@freetz-ng:~/t/freetz-ng$ ls -all /etc/localtime
lrwxrwxrwx 1 root root 29 Jan  4 12:34 /etc/localtime -> /usr/share/zoneinfo/Etc/GMT+0
vagrant@freetz-ng:~/t/freetz-ng$
```

Ich habe das Problem gerade nochmal reproduziert:

```
mkdir test
cd test
git clone https://github.com/Freetz-NG/freetz-ng.git
cd freetz-ng
make olddefconfig
make
```

Endet wie gehabt:

```
---> package/busybox ... preparing ... mkdir -p source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1; tools/bunzip2 -c dl/busybox-1.32.1.tar.bz2 | tools/tar-gnu -x -C source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1 --transform='s|^./\+||' --strip-components=1
set -e; shopt -s nullglob; for i in make/busybox/patches/*.patch* make/busybox/patches/1.32/*.patch* make/busybox/patches/1.32/4.9/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1 $i; done;
    applying patch file make/busybox/patches/1.32/002-df_remove_no_more_necessary_f_frsize_workaround.freetz.patch
    patching file coreutils/df.c
    ----------------------------------------------------------------------
    applying patch file make/busybox/patches/1.32/010-fix_ldlibs_trylinking.freetz.patch
    patching file scripts/trylink
    ----------------------------------------------------------------------
    applying patch file make/busybox/patches/1.32/023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch
    patching file archival/unzip.c
    ----------------------------------------------------------------------
    applying patch file make/busybox/patches/1.32/030-busybox-1.31.0_lzip-0.patch
    patching file archival/Config.src
    patching file archival/bbunzip.c
    patching file archival/dpkg.c
    patching file archival/dpkg_deb.c
    patching file archival/libarchive/Kbuild.src
    patching file archival/libarchive/decompress_lunzip.c
    patching file archival/libarchive/filter_accept_list_reassign.c
    patching file archival/libarchive/get_header_tar_lz.c
    patching file archival/libarchive/lzip.h
    patching file archival/libarchive/open_transformer.c
    patching file archival/lzip.c
    patching file archival/tar.c
    patching file configs/TEST_nommu_defconfig
    patching file configs/TEST_noprintf_defconfig
    patching file configs/TEST_rh9_defconfig
    patching file configs/android2_defconfig
    patching file configs/android_defconfig
    patching file configs/android_ndk_defconfig
    patching file configs/cygwin_defconfig
    patching file configs/freebsd_defconfig
    patching file include/bb_archive.h
    patching file include/libbb.h
    patching file miscutils/man.c
    patching file scripts/fix_ws.sh
    patching file testsuite/bunzip2.tests
    patching file testsuite/bzcat.tests
    patching file testsuite/lunzip.tests
    patching file testsuite/lzip.tests
    ----------------------------------------------------------------------
    applying patch file make/busybox/patches/1.32/130-scripts_bash.patch
    patching file scripts/trylink
    patching file scripts/gen_build_files.sh
    ----------------------------------------------------------------------
    applying patch file make/busybox/patches/1.32/240-telnetd_intr.openwrt.patch
    patching file networking/telnetd.c
[... 167 lines cut ...]
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/kxgettext.o
  HOSTCC  scripts/kconfig/mconf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/lex.zconf.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf -s Config.in
#
# using defaults found in .config
#
*
* Restart config...
*
*
* Settings
*
Enable compatibility for full-blown desktop systems (8kb) (DESKTOP) [N/y/?] n
Provide compatible behavior for rare corner cases (bigger code) (EXTRA_COMPAT) [N/y/?] (NEW) ^Cmake[3]: *** [/home/vagrant/t/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1/scripts/kconfig/Makefile:40: silentoldconfig] Interrupt
make[2]: *** [/home/vagrant/t/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1/Makefile:444: silentoldconfig] Interrupt
make[1]: *** [Makefile:522: include/autoconf.h] Interrupt
make: *** [make/busybox/busybox.mk:49: source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/busybox-1.32.1/busybox] Interrupt

```

ixs, 2021-01-05:
Und zum OS: Das ist ein Stock Debian 10 plus halt die Pakete zum installieren. Wenn Du magst, kann ich Dir gerne mal das Vagrant setup hochladen. Dann kannst Du schauen, ob's bei dir auch funktioniert.

Das Interessante ist halt, dass es sauber tut, wenn ich c9b37939f75781f0d792fad09a9ebdbf9f9b6e19 vom 1. Januar nutze...

fda77, 2021-01-05:
Wenn du die alte Revision nimmst, und dann unter BusyBox die 1.32 auswählst - gehts dann?
Dass es am Linux liegt glaube ich nicht, da das andere ältere Ubuntu etwas ganz anderes ist
Die BusyBox wird übrigens 2x gebaut: Für die Fritzbox und für den PC selbst als host-tool. Diese host-tools werden vor den packages gabaut und macht scheinbar keine Probleme...
Nur für Experimente mit der target-busybox reicht übrigens
`make busybox-dirclean`
`make busybox-precompiled`

ixs, 2021-01-05:
Also. Nochmal ein bisschen eingegrenzt:

Kaputt:
```
commit df0ceec8f611634f62629db243721980a57ec939
Author: fda77 <fda77@users.noreply.github.com>
Date:   Fri Jan 1 23:41:12 2021 +0100

    busybox: remove 1.31
```

Funktioniert:
```
commit c35ab7557bfcfe03661e8b02d155dd3c275fb88c
Author: fda77 <fda77@users.noreply.github.com>
Date:   Fri Jan 1 23:37:20 2021 +0100

    busybox: update 1.32
```

Lustigerweise funktioniert c35ab7557bfcfe03661e8b02d155dd3c275fb88c sowohl mit 1.32.1 als auch mit 1.31.1 ausgewählt unter "Busybox Applets" -> "Busybox Version".

Nach `git checkout df0ceec8f611634f62629db243721980a57ec939; make clean; make olddefconfig; make` ist das Problem reproduzierbar da.

fda77, 2021-01-06:
Die default-config ist "beginner" und in dem Fall wurde die Version nicht angezeigt und somit war keines der Busybox-Symbole in der .config...

closed 2021-01-06 by fda77

ixs, 2021-01-06:
Neat...