Freetz-NG/freetz-ng #142: trickle does not compile, seems to need libtirpc

edeso, 2021-01-05:
similar to https://github.com/Freetz-NG/freetz-ng/issues/137

fda77, 2021-01-05:
Nicht jedes Programm unterstützt libtirpc. Wie es aussieht trickle scheinbar nicht. Es scheint aber eh nicht mehr supported zu sein, die letzten Änderungen war vor 8 Jahren: https://github.com/mariusae/trickle und die Webseite gibt es erst gar nicht mehr

edeso, 2021-01-05:
hey fda77,

"mein" github hat nen commit von 2019. und es compiled gegen libtirpc wenn ich den patch von nfs_utils verwende, allerdings scheiterts dann am linker. Makefile hacken ist jetzt aber auch keine Spezialität von mir.

siehe auch https://stackoverflow.com/questions/50590546/unable-to-build-rpc-project-with-libtirpc

edeso, 2021-01-05:
hab ihm nochmal ne chance eingeräumt und LDFLAGS hinzugefügt. jetzt röffelts. wollt ihr ein patchset?

fda77, 2021-01-05:
Sorry,  zu spät gesehn. Makefiles sind auch nicht so mein Fall!!!
Compiliert jetzt jedenfalls und ldd zeigt die lib auch an. Soll -ltirp besser zu LDFLAGS? Habs mir einfach gemacht und alles in 1 Variable gepackt.
Welches version von trickle besser ist weiss ich nicht - ich hab es noch nie genutzt. Bei Freetz sollte es auch noch auf den uralten Boxen laufen, die kein ipv6 kennen etc

closed 2021-01-05 by fda77

edeso, 2021-01-05:
On 05.01.2021 23:15, cuma wrote:
> Sorry, zu spät gesehn. Makefiles sind auch nicht so mein Fall!!!

kein thema

> Compiliert jetzt jedenfalls und ldd zeigt die lib auch an. Soll -ltirpc besser zu LDFLAGS? Habs mir einfach gemacht und alles in 1 Variable gepackt.

wenn's tut dann tut's

> Welches version von trickle besser ist weiss ich nicht - ich hab es noch nie genutzt. Bei Freetz sollte es auch noch auf den uralten Boxen laufen, die kein ipv6 kennen etc

benutze es auch nur sporadisch. aber ist ein nifty tool, wenn man grad mal eben shapen möchte.. ede

fda77, 2021-01-05:
Okay, lassen wir es. Probier am besten mal aus ob die Version mit libtirpc auf der Box auch das macht was sie soll. Nicht dass es erst auffällt wenn du es wirklich brauchst

fda77, 2021-01-06:
Mit ldflags ist besser,das wird so überall verwendet, zb https://gogs.hasufell.de/tgurr/hasufell-repository/commit/47e3a327f5b0125649d3069107b93ecaaa5a9366

edeso, 2021-01-06:
ach Semantics ;). tested und funzt

```
root@fritz:/var/mod/root# trickle -d 200 wget -O /dev/null --no-check-certific
ate https://cdimage.debian.org/cdimage/weekly-builds/amd64/iso-bd/debian-edu-t
esting-amd64-BD-1.iso
trickle: Could not reach trickled, working independently: No such file or directory
--1970-01-01 01:16:59--  https://cdimage.debian.org/cdimage/weekly-builds/amd64/iso-bd/debian-edu-testing-amd64-BD-1.iso
Resolving cdimage.debian.org... 2001:6b0:19::165, 2001:6b0:19::173, 194.71.11.173, ...
SNIP
HTTP request sent, awaiting response... 200 OK
Length: 5829304320 (5.4G) [application/x-iso9660-image]
Saving to: '/dev/null'

/dev/null             0%[                  ]   1.09M   203KB/s    eta 7h 18m 

vs.

root@fritz:/var/mod/root# wget -O /dev/null --no-check-certificate https://cdi
mage.debian.org/cdimage/weekly-builds/amd64/iso-bd/debian-edu-testing-amd64-BD
-1.iso
--1970-01-01 01:18:25--  https://cdimage.debian.org/cdimage/weekly-builds/amd64/iso-bd/debian-edu-testing-amd64-BD-1.iso
Resolving cdimage.debian.org... 2001:6b0:19::165, 2001:6b0:19::173, 194.71.11.173, ...
SNIP
HTTP request sent, awaiting response... 200 OK
Length: 5829304320 (5.4G) [application/x-iso9660-image]
Saving to: '/dev/null'

/dev/null             0%[                  ]   9.52M  1.85MB/s    eta 51m 12s^C
```

fda77, 2021-01-06:
Danke! Ach, die Benutzung ist so einfach. Das Tool muss ich mir merken
Wenn du ca-bundle installiert hast, brauchst du kein "no-check-certificate"

edeso, 2021-01-06:
> Danke! Ach, die Benutzung ist so einfach. Das Tool muss ich mir merken

japp klasse kleines stück werkzeug

> Wenn du ca-bundle installiert hast, brauchst du kein "no-check-certificate"

hab ich. die zeit war nur nicht aktuell und ich war zu faul zu aktualisieren ;)