Freetz-NG/freetz-ng #147: [External] Illegal instruction

harryboo, 2021-01-10:
Hi,

sometimes 'external' will be normal executed and sometimes i get this error message:

/var/log/external.log
Starting external (/var/media/ftp/uStor01/external):
Waiting for mod-startup: .... done.
Waiting for time-synchronisation: ............................. done.
Illegal instruction
Starting external finished.

/var/log/mod_mount.log
[INFO] /dev/sda1 - mounted as /var/media/ftp/FREETZ (ext4)

Can it be that the folder is not mounted at this point (this happens since async changes)?

fda77, 2021-01-12:
External starts after a devices is mounted, udevmount hooks into avm methods and starts it.
There is a very new logfile which is not accessible by webif. It contains all messages from boot with timestamps: `cat /var/log/freetz.log`
Which Packages should be started? Have you a `/tmp/flash/mod/rc.external` file?

With my 7490 i have also a problem with a very old usb-stick: Sometimes it is not detected on system start, a "reboot" fixes it

harryboo, 2021-01-19:
Hi, close this ticket. After a reboot everything works.

closed 2021-01-20 by fda77

fda77, 2021-01-27:
You are using the DEB oscam addon??

harryboo, 2021-01-28:
No, only freetz-ng...