Freetz-NG/freetz-ng #155: Buidl Umgebung für freetz-ng

janfreetz, 2021-01-12:
Hallo Developer,
wo finde ich ein Virtualbox VDI Datei für die Standard Build Umgebung?
Oder wie erstelle ich mir eine default Build Umgebung, damit ich / wir auf gleichem Stand sind?
VG
Jan

fda77, 2021-01-12:
Hallo Benutzer, jedes halbwegs aktuelle Linux sollte passen. Mindestens diese Pakete werden benötigt: https://freetz-ng.github.io/freetz-ng/wiki/10_Beginner/install.de.html#installation-der-ben%C3%B6tigten-pakete-ubuntu
Oder wie viele eine fertige VM von Gismotro benutzen, siehe readme unten auf https://github.com/Freetz-NG/freetz-ng

fda77, 2021-01-12:
Es gibt übrigens oben das "Discussions", "Issues" sind für Fehler

closed 2021-01-12 by fda77

janfreetz, 2021-01-12:
Ok, sorry. 
Wo sollte man allgemeine Fragen stellen?

fda77, 2021-01-12:
Wie gesagt, zu "Discussions", das ist eine neue Funktion von Github.
Q&A steht für Fragen&Antworten

janfreetz, 2021-01-12:
Viel Erfolg beim Neubeginn.

fda77, 2021-01-12:
Das ist ne Kategorie in den Diskussionen

PeterPawn, 2021-01-12:
Wenn das Problem (https://github.com/github/feedback/discussions/2952) gefixt wurde, kann man auch wieder Issues in Discussions umwandeln - das hält die "Fehlerliste" klein und übersichtlich (was auch bei "closed" das Ziel sein sollte, damit andere bei der Suche nicht über zu viele falsch zugeordnete Themen iterieren müssen).

fda77, 2021-01-12:
Prima dass das bald kommt. Ich hatte das hinter "Transfer" erwartet, aber das wird man wohl nicht so oft (nie) brauchen