Freetz-NG/freetz-ng #158: FB 7580 mit 7.21 - bootloop.

JohnPaden, 2021-01-13:
FB 7580 - neu installiert, so vanilla wie ich es konnte
- recovery auf 7.10 - neustart
- flashen einer 7.12 Freetz .image Datei via EVA-FTP
(wie kann man eine freetz.ng image Datei erzeugen? Das Feature zeigt make menuconfig nicht an - anders als in Freetz) - reboot
- Werksreset in AVM/Sicherung/Werkseinstellungen
- Umstellung als 'Internet via WAN'
- neuer build wie vom GIT - keine Änderungen ausser Einstellen der Box auf 7580
- flashen der 7580_07.21.ger_freetz-ng-17695-8a0f3ba1c_20210113-091439.image via Freetz/System/Systemupdate - INSTALL-SUCCESS_REBOOT - Neustart

... und trotzdem Reboot in Dauerschleife. Das Freetz Syslog sehe ich noch bis ca 3:00.

Geht Euch das auch so? Wie ziehe ich am besten ein Log jetzt in max 3 Minuten ab 'Strom an'?

berndy2001, 2021-01-13:
Siehe auch #28 .

Wenn du das Syslog auf einen persistenten Speicher schreibst, dann kanns du nach dem Reboot möglicherweise sehen was los war.

JohnPaden, 2021-01-13:
Ich habe den Watchdog Patch eingespielt. Die Box resetted trotzdem. 
Das ist der Rest vom Log vor dem Reset.

Jan 13 13:38:26 fritz syslog.info syslogd started: BusyBox v1.33.0
Jan 13 13:38:26 fritz kern.notice kernel: klogd started: BusyBox v1.33.0 (2021-01-13 09:12:26 UTC)
Jan 13 12:38:26 fritz daemon.info chronyd[4856]: chronyd version 3.5 starting (+CMDMON +NTP +REFCLOCK +RTC +PRIVDROP -SCFILTER -SIGND +ASYNCDNS -SECHASH +IPV6 -DEBUG)
Jan 13 12:38:26 fritz daemon.warn chronyd[4856]: commandkey directive is no longer supported
Jan 13 13:38:34 fritz kern.info kernel: [   91.814033][3]/proc/tffs: info request: success
Jan 13 13:38:39 fritz kern.warn kernel: [   97.247133][0]TCP: dsl: Driver has suspect GRO implementation, TCP performance may be compromised.

JohnPaden, 2021-01-13:
Und noch was merkwürdiges: 
nach dem reboot kommt man ohne erneute PW Eingabe einfach nach F5 (firefox page refresh) auf die noch geöffnete Seite wie fritz.box:81/cgi-bin/system.cgi

berndy2001, 2021-01-13:
> Und noch was merkwürdiges:
> nach dem reboot kommt man ohne erneute PW Eingabe einfach nach F5 (firefox page refresh) auf die noch geöffnete Seite wie fritz.box:81/cgi-bin/system.cgi

Das ist normales Verhalten bei http_auth

PeterPawn, 2021-01-13:
> wie kann man eine freetz.ng image Datei erzeugen?

https://www.ip-phone-forum.de/threads/wie-erstell-ich-aus-einem-image-ein-inmemory.307950/

Für eine effektive Fehlersuche in einem Fall, wie er hier beschrieben ist, installiert man sich in ein Partition-Set eine (nur mäßig modifizierte) originale Firmware (einen Shell-Zugang sollte die schon haben, deshalb ist "ganz original" zwar auch denkbar, aber schwieriger/langwieriger) und in das andere dann das nichtfunktionierende Freetz(-NG)-Image.

Stürzt jetzt das Freetz-Image mit einer "kernel panic" ab, werden Informationen zu diesem Absturz noch gesichert - und zwar von einem speziellen Zweig im TFFS-Treiber von AVM. Diese kann man dann - beim Start mit dem anderen System - auslesen lassen ... entweder (langwierig) über die Support-Datei oder direkt über eigene Shell-Kommandos. Das ist zumindest bei einer "kernel panic" (die dann zum Reboot führt) oder bei einem "crash" einer AVM-Komponente (dafür gibt es ein gesondertes Log) der einfachste Weg, zumindest die ungefähre Stelle und häufig auch die Ursache zu ermitteln.

Das persistente Speichern von Protokollen ist allerdings auf einer FRITZ!Box gar nicht sooo einfach ... der NAS-Flash wird erst relativ spät gemountet und USB-Sticks, die man angeschlossen hat, noch einmal deutlich später. Damit steht man beim Protokollieren auch immer vor der Aufgabe des "Flushens" der neuen Einträge, bevor das System dann irgendwann abstürzt und die Daten nicht mehr "tatsächlich schreibt" - wobei wieder zu häufiges Flushen von Puffern auch zu sehr vielen Schreib-Operationen auf dem Flash-Speicher führt und das wieder zu sehr vielen Löschoperationen für ebendiesen Flash-Speicher, was wieder dessen Lebensdauer deutlich verkürzt. Eine forcierte Schreiboperation für das Protokollieren eines Eintrags mit - sagen wir mal - 256 Byte ist bei einer "erase size" von 128 KByte(!) ja ersichtlich unpassend und jede Optimierung durch irgendwelche Treiber verzögert das "echte Schreiben" dann entsprechend.

Am einfachsten ist es hier tatsächlich immer noch, die serielle Console zu bestücken (mit einem RS232-USB-Adapter) und mit einem Terminal-Programm den Systemstart zu beobachten - das geht nur bei den DOCSIS-Modellen nicht, weil da die Serielle abgeschaltet wird im Bootloader und später noch einmal im Kernel - das müßte man dazu erst ändern, ggf. auch nur über Kernel-Parameter (mute). 

Ohne die Benutzung der seriellen Schnittstelle hat man in FRITZ!OS ein "log gap" (zumindest beim Absturz) und das kann man z.B. dadurch schließen, daß man eine sehr frühe Konfiguration der Netzwerk-Schnittstellen einbaut (wie das geht, findet man z.B. in der "E03-flash_update") und danach einen Syslog-Daemon benutzt (z.B. den aus der BusyBox), der die per `logger`-Applet der BusyBox protokollierten Einträge nicht nur lokal (in seinem Ringbuffer im Speicher) ablegt, sondern sie über das Netzwerk an einen weiteren Syslog-Server sendet. Das setzt die "Protokoll-Lücke" zwar immer noch nicht auf Null, aber es verringert sie deutlich.

Wenn man keinen solchen Server hat (obwohl es für alle OS welche gäbe), kann man das auch einfach mit einem `netcat` (als UDP-Listener) auf einem anderen System nachbilden (das geht notfalls sogar unter Windows) - das Syslog-Protokoll ist da nicht sehr kryptisch, das kann man auch in der `netcat`-Ausgabe einigermaßen lesen.

Jedenfalls ist es deutlich leichter, das Protokollieren auf ein externes System zu verlagern - bei einem `logger`-Applet ist das `flush` praktisch inklusive und in Netzwerk gibt es da auch fast kein Caching. Beim Schreiben in eine Datei auf einem Datenträger sind da noch alle möglichen Layer dazwischen, die ihrerseits (üblicherweise natürlich nicht absichtlich, sondern zur Optimierung der Zugriffe) für verzögerte Ausführung der tatsächlichen Schreiboperation sorgen können und damit fehlt dann gerne mal genau der letzte Teil der Protokollierung, was einen schnell an der falschen Stelle suchen läßt.

Aber all das sollte (nach meiner Ansicht) tatsächlich nur mit dem Wunsch nach einer Debug-Version eines Images aktiviert werden - weil man dann auch deutlich früher als im Rahmen der "allgemeinen" Freetz-Initialisierung dafür sorgen sollte, daß ein lokaler Syslog-Daemon gestartet wird, der auch noch zusätzlich ins Netzwerk loggt (dafür muß das schon mal konfiguriert sein und zwar auch vor dem Start des "dsld" von AVM, der das ansonsten übernehmen würde) und natürlich auch die Skripte (sowohl die eigenen als auch die von AVM) deutlich geschwätziger sein sollten (bis hin zum `set -x` und der Umleitung der Shell-Protokollierung auf den Syslog-Server), als das ohne den expliziten Zweck der Fehlersuche erforderlich wäre.

PeterPawn, 2021-01-13:
> Ansonsten mal den disable-watchdog Patch aktiveren, es mehren sich die Berichte das der bei einigen (mit 7590 7.2x) noch nötig ist

Und damit kriegt man dann raus, warum das System abstürzt? Wie geht das genau?

Daß es sich nicht um ein generelles Problem im FRITZ!OS handelt, sieht man daran, daß es mit der AVM-Firmware ja läuft - wäre es da nicht schlauer, man sucht und beseitigt die Ursache in Freetz-NG, anstatt da einfach nur ein Pflaster drüber zu kleben, während das darunter weiterhin eitert (wenn's schlecht läuft)?

Manchmal verstehe ich die Welt nicht mehr ... aber vielleicht erwarte ich da auch zuviel.

fda77, 2021-01-13:
> Das ist normales Verhalten bei http_auth wie freetz es verwendet.

Man kann im Webif auf "Neue Loginversion mit Session-ID " umstellen

fda77, 2021-01-25:
Gleiche CPU wie 7590, damit am besten hier weiter --> https://github.com/Freetz-NG/freetz-ng/issues/28

closed 2021-01-25 by fda77