Freetz-NG/freetz-ng #161: Cannot build when FREETZ_VERBOSITY_LEVEL=1

pkgzm, 2021-01-13:
make menuconfig

is ok,
but when I set FREETZ_VERBOSITY_LEVEL=1 I get:

xxxxxxx@yyyyy:~/src/sys/freetz-ng$ make 
tar -C /home/xxxxxxx/src/sys/freetz-ng/source/host-tools  -xf dl/tar-1.32.tar.xz
set -e; shopt -s nullglob; for i in tools/make/tar-host/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch /home/xxxxxxx/src/sys/freetz-ng/source/host-tools/tar-1.32 $i; done;
make: *** [tools/make/tar-host/tar-host.mk:16: /home/xxxxxxx/src/sys/freetz-ng/source/host-tools/tar-1.32/.unpacked] Error 1

fda77, 2021-01-14:
Maybe you downloaded file is corrupt? try `rm -f dl/tar-1.32.tar.xz` and then "make" again

pkgzm, 2021-01-14:
After removal, same result:

xxxxxxx@yyyyy:~/src/sys/freetz-ng$ rm ~/.freetz-dl/tar-1.32.tar.xz
xxxxxxx@yyyyy:~/src/sys/freetz-ng$ make
tools/freetz_download dl tar-1.32.tar.xz @GNU/tar 83e38700a80a26e30b2df054e69956e5

--2021-01-14 09:44:47--  http://ftpmirror.gnu.org/tar/tar-1.32.tar.xz
Risoluzione di ftpmirror.gnu.org (ftpmirror.gnu.org)... 209.51.188.200
Connessione a ftpmirror.gnu.org (ftpmirror.gnu.org)|209.51.188.200|:80... connesso.
Richiesta HTTP inviata, in attesa di risposta... 302 Moved Temporarily
Posizione: http://mirror.inode.at/gnu/tar/tar-1.32.tar.xz [segue]
--2021-01-14 09:44:48--  http://mirror.inode.at/gnu/tar/tar-1.32.tar.xz
Risoluzione di mirror.inode.at (mirror.inode.at)... 84.116.34.228
Connessione a mirror.inode.at (mirror.inode.at)|84.116.34.228|:80... connesso.
Richiesta HTTP inviata, in attesa di risposta... 200 OK
Lunghezza: 2103348 (2,0M) [application/x-xz]
Salvataggio in: "dl/tar-1.32.tar.xz"

tar-1.32.tar.xz                                100%[================================================================>]   2,01M  7,67MB/s    in 0,3s    

2021-01-14 09:44:49 (7,67 MB/s) - "dl/tar-1.32.tar.xz" salvato [2103348/2103348]

Download succeeded - "http://ftpmirror.gnu.org/tar/tar-1.32.tar.xz"  ->  saved to folder "dl"
Checksum verified for dl/tar-1.32.tar.xz: MD5:=83e38700a80a26e30b2df054e69956e5
tar -C /home/xxxxxxx/src/sys/freetz-ng/source/host-tools  -xf dl/tar-1.32.tar.xz
set -e; shopt -s nullglob; for i in tools/make/tar-host/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch /home/xxxxxxx/src/sys/freetz-ng/source/host-tools/tar-1.32 $i; done;
make: *** [tools/make/tar-host/tar-host.mk:16: /home/xxxxxxx/src/sys/freetz-ng/source/host-tools/tar-1.32/.unpacked] Error 1

fda77, 2021-01-14:
https://github.com/Freetz-NG/freetz-ng/issues/163 bedeutet das es nun geht?

pkgzm, 2021-01-14:
Unfortunately no. I had to force FREETZ_VERBOSITY_LEVEL=2 in order to achieve #163...

closed 2021-01-14 by fda77