Freetz-NG/freetz-ng #163: Compile options not recognized

pkgzm, 2021-01-14:
When compiling the toolchain with this options:

Kernel binutils (binutils-2.24)
Kernel compiler (gcc-4.8.5)

and

Target uClibc version (0.9.33.2)
Target binutils (binutils-2.24)
Target compiler (gcc-4.8.5)

I get the following messages:

xxxxxxxxxx@yyyyy:~/src/sys/freetz-ng$ make
...
...
/home/xxxxxxxxxx/src/sys/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -I.. -I../.. -I../modes -I../asn1 -I../evp -I../../include  -fPIC -DOPENSSL_PIC -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -DOPENSSL_SMALL_FOOTPRINT -DOPENSSL_NO_ERR -DB_ENDIAN -DTERMIO -march=24kc -mtune=24kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -ffunction-sections -fdata-sections -fomit-frame-pointer -Wall -DOPENSSL_BN_ASM_MONT -DSHA1_ASM -DSHA256_ASM -DAES_ASM -c  -o bn-mips.o bn-mips.s
mips-linux-uclibc-gcc: error: unrecognized command line option '-fdiagnostics-color'
make[3]: *** [<builtin>: bn-mips.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/xxxxxxxxxx/src/sys/freetz-ng/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/openssl-1.0.2u/crypto/bn'
make[2]: *** [Makefile:90: subdirs] Error 1
make[2]: uscita dalla directory "/home/xxxxxxxxxx/src/sys/freetz-ng/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/openssl-1.0.2u/crypto"
make[1]: *** [Makefile:287: build_crypto] Error 1
make[1]: uscita dalla directory "/home/xxxxxxxxxx/src/sys/freetz-ng/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/openssl-1.0.2u"

ERROR: Build failed.
make: *** [make/openssl/openssl.mk:86: source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/openssl-1.0.2u/libssl.so.1.0.0] Error 1

on the other hand, by selecting

Target compiler (gcc-5.5)

the compilation is successful.

fda77, 2021-01-14:
Please attach you .config

pkgzm, 2021-01-16:
Here it is.

[router.2021.01.13-20.37.00.config-ng.txt](https://github.com/Freetz-NG/freetz-ng/files/5824058/router.2021.01.13-20.37.00.config-ng.txt)

fda77, 2021-01-25:
`-fdiagnostics-color` was added in GCC 4.9
https://gcc.gnu.org/gcc-4.9/changes.html
```
Support for colorizing diagnostics emitted by GCC has been added. The -fdiagnostics-color.....
```

fda77, 2021-01-25:
I compiled with you provided config and it was okay. Then I changed your .config as you wrote in OP:
```
$ diff -Naur router.2021.01.13-20.37.00.config-ng.txt .config | grep -vE "^ |^@|^.#"|sort -r

-FREETZ_TARGET_GCC_MINOR_VERSION="5.0"
+FREETZ_TARGET_GCC_MINOR_VERSION="5"
-FREETZ_TARGET_GCC_MAJOR_VERSION="5"
+FREETZ_TARGET_GCC_MAJOR_VERSION="4.8"
-FREETZ_TARGET_GCC_5=y
+FREETZ_TARGET_GCC_4_8=y
-FREETZ_TARGET_BINUTILS_VERSION="2.26.1"
+FREETZ_TARGET_BINUTILS_VERSION="2.24"
-FREETZ_TARGET_BINUTILS_2_26=y
+FREETZ_TARGET_BINUTILS_2_24=y
-FREETZ_GNULIBSTDCXX_VERSION="6.0.21"
+FREETZ_GNULIBSTDCXX_VERSION="6.0.19"
-FREETZ_DL_TARGET_TOOLCHAIN_MD5="X"
+FREETZ_DL_TARGET_TOOLCHAIN_MD5="23347e3df542890426b439919b794d55"
```
And "make" worked also for me

closed 2021-01-28 by fda77