Freetz-NG/freetz-ng #169: freetz 7590 wlan not working any longer

jpsollie, 2021-01-25:
after installing freetz, my wlan did not work anymore,
there must be an error in the config which causes the wlan setup to fail.
resetting the default options (so minimal modifications) restored wlan functionality
[working_config.txt](https://github.com/Freetz-NG/freetz-ng/files/5864894/working_config.txt)
[no_wlan_config.txt](https://github.com/Freetz-NG/freetz-ng/files/5864896/no_wlan_config.txt)

non-working .config + working .config attached,
(image file available on request)
Could anyone point me to a *possible* strategy on how to bisect the issue?

berndy2001, 2021-01-25:
In any case, it would be helpful to see the config.compressed because it is more lightweight.

As far as I have seen, you have enabled some packages and removal patches. Try to deactivate this selection step by step (It can also be several steps at once) and test if wifi works again.

jpsollie, 2021-01-25:
any idea of *possible* targets I should focus on first?

berndy2001, 2021-01-25:
I would start as follows: As step 1 the removal patches and as step 2 Wireguard and Openvpn.

fda77, 2021-01-25:
Maybe the configuration of the VPN packages messed with the network-interfaces or ip-ranges/routing. You coudl try to disable them in webif and reboot

fda77, 2021-01-25:
You should not use
```
FREETZ_USER_LEVEL_DEVELOPER
```
which enables things which my not are working yet, like iptables:
```
FREETZ_MODULES_OWN="ah6 esp6 ip6_tunnel sit tunnel6 ip_tables iptable_filter xfrm6_mode_transport x_tables"
```
So it seems you've also altered the kernel-.config

closed 2021-01-25 by fda77