Freetz-NG/freetz-ng #172: Build failed for 3490 7.12 with freetz-ng 17773-017064b11

PhoenixG36, 2021-01-29:
since some days I can not build an actuel freetz image for the 3490.

freetz-ng 17773-017064b11 master 2021-01-28   

It stop with an error:

make[4]: Entering directory '/home/freetz/freetz-ng-3490/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1/lib'
  CC       getrandom.o
  CC       spawn_faction_init.o
  CC       spawnattr_destroy.o
  CC       spawnattr_init.o
  CC       spawnattr_setflags.o
In file included from ./sys/random.h:40:0,
                 from getrandom.c:22:
/home/freetz/freetz-ng-3490/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/mips-linux-uclibc/include/sys/random.h:27:35: error: unknown type name 'size_t'
 extern int getrandom(void *__buf, size_t count, unsigned int flags)
                                   ^
make[4]: *** [Makefile:2632: getrandom.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/home/freetz/freetz-ng-3490/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1/lib'
make[3]: *** [Makefile:2192: all] Error 2
make[3]: Leaving directory '/home/freetz/freetz-ng-3490/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1/lib'
make[2]: *** [Makefile:1707: all-recursive] Error 1
make[2]: Leaving directory '/home/freetz/freetz-ng-3490/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1'
make[1]: *** [Makefile:1659: all] Error 2
make[1]: Leaving directory '/home/freetz/freetz-ng-3490/source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1'

ERROR: Build failed.
make: *** [make/wget/wget.mk:68: source/target-mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10/wget-1.21.1/src/wget] Error 1

fda77, 2021-01-31:
The problem is the wget bump: https://github.com/Freetz-NG/freetz-ng/commit/08bbcd8898e90dc4dd26cf8e044343aa518b0525

From https://stackoverflow.com/questions/26410466/gcc-linaro-compiler-throws-error-unknown-type-name-size-t:
```
As per C99, §7.17, size_t is not a builtin type but defined in <stddef.h>.
Including the <stddef.h> header should fix your problem.
```

Meanwhile you could use a old revision: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md#checkout-old-revision

edeso, 2021-02-05:
just stumbled about it myself. wouldn't it make sense to unbump wget to the working version until this is fixed?

fda77, 2021-02-05:
Für aktuelles FritzOS funktioniere es ja! Man könnte nun
 - Warten bis es eine neue offizielle Version gibt in der es (hoffentlich) gefixt ist, das kann dauern
 - Wieder zu einer alten Version wechseln, blöd für die die neuere Geräte haben
 - Jemand korrigiert wget (make/wget/patches/). Das ist der beste und schnellste Weg, wenn sich jemand gefunden hat der sich dem annimmt

edeso, 2021-02-05:
@fda77 probier mal diesem patch. damit läufts bei mir mit nem image für 7362SL durch. 

make/wget/patches/030-size_t.patch
```
--- ./lib/getrandom.c.sav	2021-02-05 14:42:30.633615449 +0100
+++ ./lib/getrandom.c	2021-02-05 14:42:41.041684997 +0100
@@ -18,7 +18,7 @@
 /* Written by Paul Eggert.  */
 
 #include <config.h>
-
+#include <stddef.h>
 #include <sys/random.h>
 
 #include <errno.h>
```

warum geht patch anhängen hier nicht, damn it.

edeso, 2021-02-05:
und mir ist aufgefallen. beim manuellen setzen der version ohne hash aktualisierung in ``make/wget/wget.mk`` gab's keinen abbruch weil der hash nicht stimmte. muss das so?

fda77, 2021-02-05:
Cool, danke! Evtl muss man ".txt" für Github an die Datei anhängen
Oder magst du einen "Mailbox Format" Patch anhängen? https://freetz-ng.github.io/freetz-ng/SUPPORTDann wäre dein Name im Commit.
Funktioniert es noch wenn der Patch nicht nötig wäre?

Der Hash wird NACH dem DOWNLOADgeprüft (sieht man in grün). Da die Datei schon in dl/ war gibt es keine Fehlermeldung... Nur bei Firmware wird VOR dem ENTPACKEN geprüft (Signatur)

edeso, 2021-02-05:
name ist wurst. nimm's wie's ist :). stören sollte der zus. header nicht wenner nicht gebraucht wird. man könnte ne bedingung drum tun, aber hey. 

hab testweise mit patch nen image für die 7530 basierend auf 7.21 gebaut. lief durch.

closed 2021-02-05 by fda77