Freetz-NG/freetz-ng #173: build failed if a different dir (not freetz-bg) is used

gneiss15, 2021-01-29:
I made a copy of the dir freetz-ng, then the make failed when applying the patches from "tools/make/busybox-host/patches/".
I found that in this case all links point into the wrong (old) dir. Below are the dir lists of both cases. I assume that there is some entry inside a "config.in" still points to the old version 1.31 of busybox...

Short comparsion of only one link inside tools/make/busybox-host/patches/:
OK:  023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch -> ../../../../make/busybox/patches/1.32/023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch
NOK: 023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch -> ../../../../make/busybox/patches/1.31/023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch

Full dir list:
OK:
~/freetz-ng$ ll tools/make/busybox-host/patches/
insgesamt 52
-rw-r--r-- 1 gneiss gneiss  487 Jan 29 21:32 010-tar_preserve_directory_permissions.freetz.patch
lrwxrwxrwx 1 gneiss gneiss  101 Jan 29 21:32 023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch -> ../../../../make/busybox/patches/1.32/023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   69 Jan 29 21:32 030-busybox-1.31.0_lzip-0.patch -> ../../../../make/busybox/patches/1.32/030-busybox-1.31.0_lzip-0.patch
lrwxrwxrwx 1 gneiss gneiss   89 Jan 29 21:32 425-applet_aliases.MAX_APPLET_NAME_LEN.freetz.patch -> ../../../../make/busybox/patches/1.32/425-applet_aliases.MAX_APPLET_NAME_LEN.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   84 Jan 29 21:32 426-applet_aliases.busybox_suffix.freetz.patch -> ../../../../make/busybox/patches/1.32/426-applet_aliases.busybox_suffix.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   82 Jan 29 21:32 430-blkid_bogus_device_handling.freetz.patch -> ../../../../make/busybox/patches/1.32/430-blkid_bogus_device_handling.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   74 Jan 29 21:32 431-blkid_offset_option.freetz.patch -> ../../../../make/busybox/patches/1.32/431-blkid_offset_option.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   77 Jan 29 21:32 500-archival_Wmaybe-uninitialized.patch -> ../../../../make/busybox/patches/1.32/500-archival_Wmaybe-uninitialized.patch
lrwxrwxrwx 1 gneiss gneiss   63 Jan 29 21:32 901-crc32.PeterPawn.patch -> ../../../../make/busybox/patches/1.32/901-crc32.PeterPawn.patch
lrwxrwxrwx 1 gneiss gneiss   63 Jan 29 21:32 902-sum_pure.freetz.patch -> ../../../../make/busybox/patches/1.32/902-sum_pure.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   73 Jan 29 21:32 903-expr_double_dashes.freetz.patch -> ../../../../make/busybox/patches/1.32/903-expr_double_dashes.freetz.patch

NOK:
~/freetz/7490$ ll tools/make/busybox-host/patches/
insgesamt 52
-rw-r--r-- 1 gneiss gneiss  487 Jan 29 20:55 010-tar_preserve_directory_permissions.freetz.patch
lrwxrwxrwx 1 gneiss gneiss  101 Jan 29 20:55 023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch -> ../../../../make/busybox/patches/1.31/023-unzip_49d54afe4e196f3dd51188394a922259dd8ff939.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   69 Jan 29 20:55 030-busybox-1.31.0_lzip-0.patch -> ../../../../make/busybox/patches/1.31/030-busybox-1.31.0_lzip-0.patch
lrwxrwxrwx 1 gneiss gneiss   89 Jan 29 20:55 425-applet_aliases.MAX_APPLET_NAME_LEN.freetz.patch -> ../../../../make/busybox/patches/1.31/425-applet_aliases.MAX_APPLET_NAME_LEN.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   84 Jan 29 20:55 426-applet_aliases.busybox_suffix.freetz.patch -> ../../../../make/busybox/patches/1.31/426-applet_aliases.busybox_suffix.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   82 Jan 29 20:55 430-blkid_bogus_device_handling.freetz.patch -> ../../../../make/busybox/patches/1.31/430-blkid_bogus_device_handling.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   74 Jan 29 20:55 431-blkid_offset_option.freetz.patch -> ../../../../make/busybox/patches/1.31/431-blkid_offset_option.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   77 Jan 29 20:55 500-archival_Wmaybe-uninitialized.patch -> ../../../../make/busybox/patches/1.31/500-archival_Wmaybe-uninitialized.patch
lrwxrwxrwx 1 gneiss gneiss   63 Jan 29 20:55 901-crc32.PeterPawn.patch -> ../../../../make/busybox/patches/1.31/901-crc32.PeterPawn.patch
lrwxrwxrwx 1 gneiss gneiss   63 Jan 29 20:55 902-sum_pure.freetz.patch -> ../../../../make/busybox/patches/1.31/902-sum_pure.freetz.patch
lrwxrwxrwx 1 gneiss gneiss   73 Jan 29 20:55 903-expr_double_dashes.freetz.patch -> ../../../../make/busybox/patches/1.31/903-expr_double_dashes.freetz.patch

fda77, 2021-01-31:
It seems you have not updated the checkout correct/completely
With https://github.com/Freetz-NG/freetz-ng/commit/ddce20e905dc57f16c2f58527518ff3a6d46331a the links are changed from 1.31 to 1.32
To check: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md#show-git-states
To (maybe) fix: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md#delete-local-changes

closed 2021-02-02 by fda77

gneiss15, 2021-02-02:
Sorry, I had some problem with the I-Net (local cable damaged), that's why I could not answer in the meanwhile.
I am not shure what I should do. (I know very little about git).
git status shows (german)
---------------
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

nichts zu committen, Arbeitsverzeichnis unverändert
---------------
So I had assumed I had the latest version.

even a "git pull" does not change/correct the links.
After I execute the commands under "Delete local changes", the links are corrected...
So, problem is reals solved, thanks.

fda77, 2021-02-03:
Git ist etwas komplizierter als svn, man kann Dateien ändern, geänderte Dateien mit "add" hinzufügen, geänderte Dateien lokal commit.
Auch könnte man einen anderen Branch lokal erstellen, oder einen eigenen Fork ausgecheckt haben der nicht aktuell ist
Und vermutlich noch viele andere Dinge