Freetz-NG/freetz-ng #176: Minor Bug in tools/freetz-revision

gneiss15, 2021-02-03:
Today I checked out a frech copy of freetz onto a older PC (Ubuntu 14.04.6 LTS).
Inside "make menuconfig" was displayed:
--- Freetz revision  -----------------------------------------                                            
fatal: unknown date format format:%Y-%m-%d freetz-ng 17793-afcded6 master 

Because I always awoid having "warnings" (and a "fatal" makes me highly alerted), I do some research.
I found two problems with this script:
1. During testing
> bash -x freetz-revision
... from inside tools
The script doen't correctly initialise the var "DIR"
> DIR="$(readlink -f ${0%/*}/..)"
That only works if the script is not called from it's own dir.
So I changed this to:
> DIR="$(dirname $(readlink -f "$0") )/.."
Which should produce the correct result in any case.
But this is only a "debug-problem"

2. The real problem is the use of (line 25):
> [ "$1" ] && RB="${RB/master/}" || RD="$(git --git-dir=$GDB log --pretty='format:%cd' --date=format:'%Y-%m-%d' -1 @)"
Only newer git versions support the "format:..." for date.
Ubuntu 14.04.6 uses git version 1.9.1.
So the "fatal" shown above is related to the use of this.
I could't see any reason for using the "format" variant, with a format that is already known as "short" in older versions.
So after I changed this line to:
> [ "$1" ] && RB="${RB/master/}" || RD="$(git --git-dir=$GDB log --pretty='format:%cd' --date=short -1 @)"
Everything is working as expected.

As written in my prev. report, I am not that familar with git & how to supply commits (I believe that I am not authorisized to make commits, too).
That's the reason I wrote this report.
Maybe if someone could give me some hints how to prepare commits I will do so in the future.
I guess I will have some more bugfixes/advantages later ;-)

Another question:
I am a german. Is it allowed/preferd to write in german here ?

fda77, 2021-02-03:
Ich schreib/red' lieber deutsch mit denen die es können.
Das Problem ist bekannt bei Ubuntu aus dem Jahr 2014 (7 Jahre alt) und eigentlich sollte diese Meldung erscheinen (die hast du auskommentiert?):
```
Your Linux System is too old. Please upgrade it or use Freetz-Linux: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md
```
Es gibt später aber noch andere Probleme mit diesem uralten System, zb make ist zu alt

gneiss15, 2021-02-03:
Hallo,

 

ich hab da nichts auskommentiert.

war ja erst bei "make menuconfig". (Soweit hat alles funktioniert)

In der zwichenzeit ist das make gelaufen und meldet auch weitere Fehler.

Werd dann mal auf desem System die Versuche abbrechen (hab ja noch zwei deutlich aktuellere ;-)

 

Aber wie gesagt eine solche Meldung kam nicht.

Erst bei make..... (nach einiger Zeit):

> checking whether the C compiler works... no

 
 

Gesendet: Mittwoch, 03. Februar 2021 um 13:51 Uhr

fda77, 2021-02-03:
Schade, sollte aber: https://github.com/Freetz-NG/freetz-ng/blob/master/Makefile#L88
Wie ist denn bei die Ausgabe von `uname -r`?

gneiss15, 2021-02-05:
> Wie ist denn bei die Ausgabe von uname -r?

4.4.0-148-generic

 

... und da "ist der Hund begraben" :-)

Dort wird ja nur der kernel abgefragt und nur wenn der <= 3.x ist die Warnung ausgegeben.

Mein kernel ist deutlich neuer, was an Ubuntus HWE liegt.

 

Wenn man das "sicherer" machen wil, müßte da noch ein check der "tools" erfolgen.

So könnte man via 

 

ifeq ($(shell git version 1.9.1 | sed 's/[^0-9]*//;s/^[1]//'),)
  $(error Your Linux System seams to be too old. git version is below 2.x. Please upgrade it or use Freetz-Linux: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md ..)
endif

 

noch die git version abfragen und eine entspr. Fehlermeldung ausgeben.

 

Ist allerdings mühevoll alle möglichen Fehlerquellen abzufangen.

 

 

Gesendet: Mittwoch, 03. Februar 2021 um 14:44 Uhr

fda77, 2021-02-05:
Danke! Von "Ubuntus HWE" hab ich bis jetzt noch nie was gehört.
Den einfache Test auf Kernel v3 gitb es weil einige noch silent-tears' uraltes Freetz-Linux genutzt hatten und nicht wussten dass Gismotro es weiterführt.

closed 2021-02-05 by fda77