Freetz-NG/freetz-ng #177: build schlägt fehl cmac.ko und sha256_generic.ko scheinen zu fehlen.

Conan179, 2021-02-03:
Hallo erst mal.
Ich bräuchte mal eure hilfe.
Ich musste meinen pc kommplet neu aufsetzt und wollte heute für meine 6590 ein neues Image bauen.
ausgecheckt ist die freetz Version: "freetz-ng 17793M-afcded65c master 2021-02-01"
Ich hab die .config von meiner 6590 übernommen, wie sie im web interface angezegt wird, keine änderungen.
Jetzt bekomme ich beim bauen immer die fehlermeldung:
WARNING: File of kernel module not found: cmac.ko
WARNING: File of kernel module not found: sha256_generic.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
make: *** [Makefile:277: firmware-nocompile] Fehler 1

Die suche im make kernelconfig zegt mir folgendes an:
  │ Symbol: MACMACE [=n]                                                                                                                                      │
  │ Type  : boolean                                                                                                                                           │
  │ Prompt: Macintosh (AV) onboard MACE ethernet                                                                                                              │
  │   Defined at drivers/net/Kconfig:413                                                                                                                      │
  │   Depends on: NETDEVICES [=y] && NET_ETHERNET [=y] && MAC                                                                                                 │
  │   Location:                                                                                                                                               │
  │     -> Device Drivers                                                                                                                                     │
  │       -> Network device support (NETDEVICES [=y])                                                                                                         │
  │         -> Ethernet (10 or 100Mbit) (NET_ETHERNET [=y])                                                                                                   │
  │   Selects: CRC32 [=y]

In dem punkt, habe ich keinen "CRC32" punkt gefunden

und
  │ Symbol: CRYPTO_SHA256_S390 [=n]                                                                                                                           │
  │ Type  : tristate                                                                                                                                          │
  │ Prompt: SHA256 digest algorithm                                                                                                                           │
  │   Defined at drivers/crypto/Kconfig:94                                                                                                                    │
  │   Depends on: CRYPTO [=y] && CRYPTO_HW [=y] && S390                                                                                                       │
  │   Location:                                                                                                                                               │
  │     -> Cryptographic API (CRYPTO [=y])                                                                                                                    │
  │       -> Hardware crypto devices (CRYPTO_HW [=y])                                                                                                         │
  │   Selects: CRYPTO_HASH [=y]                                                                                                                               │
  │                                                                                                                                                           │
  │                                                                                                                                                           │
  │ Symbol: CRYPTO_SHA256 [=y]                                                                                                                                │
  │ Type  : tristate                                                                                                                                          │
  │ Prompt: SHA224 and SHA256 digest algorithm                                                                                                                │
  │   Defined at crypto/Kconfig:415                                                                                                                           │
  │   Depends on: CRYPTO [=y]                                                                                                                                 │
  │   Location:                                                                                                                                               │
  │     -> Cryptographic API (CRYPTO [=y])                                                                                                                    │
  │   Selects: CRYPTO_HASH [=y]                                                                                                                               │
  │   Selected by: ENCRYPTED_KEYS [=n] && KEYS [=y] && TRUSTED_KEYS [=n] || CRYPTO_DEV_PADLOCK_SHA [=n] && CRYPTO [=y] && CRYPTO_HW [=y] && CRYPTO_DEV_PADLOC │

finde ich auch nicht -.-

könnt ihr mir helfen und sagen was ich falsche mache?

fda77, 2021-02-05:
Das ist ein Fehler in Freetz. Wenn man sich die HELP von CONFIG_CIFS ansieht:
```
Selects: NLS [=y] && CRYPTO [=y] && CRYPTO_MD4 [=m] && CRYPTO_MD5 [=y] && CRYPTO_HMAC [=y] && CRYPTO_ARC4 [=y]                         
```
cmac sollte nicht angezeigt werden

closed 2021-02-05 by fda77

Conan179, 2021-02-05:
@fda77 Ich DANKE dir für deine Hilfe! Der image bau klappt wieder einwandfrei und glasht ist das image auch schon.
Also DANKE!