Freetz-NG/freetz-ng #180: tools/tic wird nicht erzeugt, später aber von run_tic.sh benutzt -> Build-Abbruch

gneiss15, 2021-02-05:
tools/tic wird nicht erzeugt, später aber von run_tic.sh benutzt -> Build-Abbruch

Rahmenbedingungen:
1. Frisch aufgesetzte VM mit Ubuntu 20.04 LTS
2. Alle benötigten Pakete installiert installiert
3. freetz-ng frisch ausgecheckt
4. make menuconfig
5. make
6. -> in Zeile 149 von run_tic.sh wird tools/tic aufgerufen, wurde aber vorher nicht "gebaut"

Im Makefile nachgesehen, Zeile 339:
tools: $(DL_DIR) $(SOURCE_DIR_ROOT) $(filter-out $(TOOLS_CONDITIONAL),$(TOOLS))
aha, offensichtlich wird DAS tool via TOOLS_CONDITIONAL ausgeblendet.
Testweise mal die Zeile geändert auf:
tools: $(DL_DIR) $(SOURCE_DIR_ROOT) $(TOOLS)
Jetzt werden zwar alle tools gebaut, aber dann läuft der build (dort oben) weiter.

Was ich jetzt nicht so ganz klar ist:
1. welches target denn tic baut
2. ob es falsch ist, dass das target welches tic baut ausgeblendet wurde, oder
3. es falsch ist das später make in Verzeichnis "target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc" aufgerufen wird

Es liegt aber offensichtlich ein Fehler in kconfig vor, der (bei meiner .config) zu diesem Verhalten führt.
Da fehlt mir aber noch einiges an Wissen um herauszufinden wo der Fehler liegt.
ich kann gerne meie .config zur Verfügung stellen, wenn das benötigt wird.

Noch ein paar weitere Hilfen zum aufspüren des Fehlers:

Via Ausgabe in Make habe ich herrausgefunden, welche targets bei mir ausgeblendet werden:
TOOLS_CONDITIONAL:  gmp-host mpc-host mpfr-host ncurses-host mklibs python-host scons-host xdelta
hier gehe ich mal davon aus, dass das ausgeblendete target ncurses-host ist.

Also, neuer Versuch. Alles auf Anfang Alles neu ausgecheckt, im Makefile die Zeile auf
tools: $(DL_DIR) $(SOURCE_DIR_ROOT) $(TOOLS) ncurses-host
geändert.

Auch jetzt läuft der build an dieser Stelle durch. Meine Vermutung stimmte also.
Es ist das target ncurses-host von tools, das gebaut werden muss.

Leider kommt dann aber später
source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2: Datei oder Verzeichnis nicht gefunden.  Schluss

Also fehlt hier wohl auch der entsprechende download der target-sourcen...

Ohne Hilfe bin ich hier erst mal "raus" :-)
... zmindest was das auspühren des eigentlichen Fehlergrundes angeht.

Nachdem ich in menuconfig unter "Shared Libraries -> ncurses -> ncurses (libncurses.so) ausgewählt habe läuft der build erst mal weiter...

Mal sehen was noch so kommt ;-)

fda77, 2021-02-05:
Ncurses wird 3x aus dem gleichen Download gebrau: nurcses, ncrusesw und ncurses-host
Hier sind die conditionalen hosts packages: tools/make/Makefile.in
Für source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/* muss man im menuconfig auswählen was gebaut werden soll.
Vermutlich hast du ein Package ausgewählt, das nicht automatisch (lib)ncurses auswählt, wahrscheinlcih eins von diesen: "grep ncurses make/*/*k"
und zwar in:
4. make menuconfig

gneiss15, 2021-02-06:
OK, mit den Hinweisen kann ich mal suchen welches Paket hier "vergisst" libncurses auszuwählen (Das ist ja der von mir vermutete Fehler).
Sobalsd ich das gefunden habe, melde ich mich mit der entspr. Korrektur zurück...
(Kann aber etwas dauern, bin mom. wo anders dran)

gneiss15, 2021-02-06:
OK, ich hatte doch mal Zeit...

zunächst einmal zu deiner Aussage:
> die gibt es bei mir nicht (run_tic.sh)
Die wird wohl erst (in source/host-tools/ncurses-6.2/misc/) gebaut wenn das aus der entspr. config.in heraus "gefordert" wird.

Zu dem was ich weiter gefunden habe:
1. In der Datei tools/make/Makefile.in werden ALLE oben schon von mir angegebenen TOOLS_CONDITIONAL geseztt.
2. Bis auf mklibs erfolgt das auch "unconditional", soll heißen: diese tools werden NIE gebaut!
3. Soweit ich das jetzt erkennen konnte ist die einzige "Besonderheit" die ich bei der .config wo das Problem auftrat ausgewählt hatte der mc (MidnightCommander).
4. der scheint auch der einzige zu sein der "ncurses-terminfo" benutzt
make/mc/mc.mk:$(PKG)_DEPENDS_ON += ncurses-terminfo glib2
5. Das der Fehler damit in Zusammenhang steht verfestigt sich, da tic der "Terminfo Compiler" ist
6. Damit müsste wohl "nur" in der tools/make/Makefile.in die Zeile 21 geändert werden.

Derzeit steht da:
TOOLS_CONDITIONAL+=ncurses-host
ifneq ($(strip $(FREETZ_STRIP_LIBRARIES)),y)
TOOLS_CONDITIONAL+=mklibs
endif

Was in etwa so sein müsste:
if "no selected pakage depends on ncurses-terminfo"
TOOLS_CONDITIONAL+=mklibs
endif
TOOLS_CONDITIONAL+=ncurses-host
ifneq ($(strip $(FREETZ_STRIP_LIBRARIES)),y)
TOOLS_CONDITIONAL+=mklibs
endif

Wobei mir derzeit noch nicht klar ist wo ich diese info her bekomme.
Wenn ich bisher alles richtig verstanden habe, wird durch die Zeile
$(PKG)_DEPENDS_ON += ncurses-terminfo glib2
in der mc.mk ja nur die target-seite abgedeckt.

Die richtige Stelle ist wohl:
make/mc/Config.in
hier muss wohl ein weiteres "select" hinein.
select FREETZ_SHARE_terminfo
Das könnte dann in obigem if verwendet werden.

Damit ich sicher bin das das auch so richtig ist, werde ich wohl:
- wieder komplett neu auschecken
- die o.a Änderungen durchführen
- dann mit der gleichen .config den make starten

Kann etwas dauern.
Melde mich wieder wenn das Ergebnis feststeht.

gneiss15, 2021-02-06:
OK, die syntax war etwas falsch (ist ja keine bash hier), aber ansonsten läuft es so wie es sollte...

Also, Zeile 21 in tools/make/Makefile.in ersetzen durch:
ifneq ($(FREETZ_SHARE_terminfo),y)
TOOLS_CONDITIONAL+=ncurses-host
endif

und der Fehler tritt nicht mehr auf.

fda77, 2021-02-06:
In der make/mc/Config.in wird doch "FREETZ_LIB_libncurses" selektiert, je nachdem was man ausgewählt hat. Und es kann auch sein dass FREETZ_LIB_libncursesw gebraucht wird. Von wo kommt denn diese "run_tic.sh"?

gneiss15, 2021-02-07:
OMG, eigentlich ein "kleines" Problem, zieht aber einiges nach sich.

Also ich hab noch mal tiefer geforscht...
Du hast recht:
> In der make/mc/Config.in wird doch "FREETZ_LIB_libncurses" selektiert, je nachdem was man ausgewählt hat.
Richtig, wenn man

fda77, 2021-02-07:
Nucrses und ncursesw unterscheidne sich im unicode/widechar support. Es können durchaus beide auf der Box sein, je nach Package. Terminfo wird nur aus ncurses gebaut, es reicht wenn es 1x da ist.
tic aus ncurses-host wird zum generieren auf dem PC gebraucht, evlt ist ein alte ncurses 5 version installiert die inkompatibel ist.

Um den Fehler nachzustellen, reicht es `Shared libraries > ncurses > terminfo database` auszuwählen

Das `$(PKG)_HOST_DEPENDS_ON += ncurses-host` wird dann nicht beachtet

In ./make/Makefile.in
```
$(pkg)-precompiled--int: $(pkg)-build-prereq $(TOOLCHAIN) $($(PKG)_HOST_DEPENDS_ON) $(patsubst %,%-precompiled,$($(PKG)_DEPENDS_ON))
```

fda77, 2021-02-07:
Wenn FREETZ_LIB_libncursesw bzw FREETZ_LIB_libncurses nicht selektiert sind, wird das `$(PKG)_HOST_DEPENDS_ON += ncurses-host` in der ncursesw.mk bzw ncurses.mk scheinbar nicht beachtet. Das wird wohl irgendwo herausgefiltert (ich hab nicht viel Ahnung von Makefiles!)
Wenn man in tools/make/Makefile.in so ändert
```
+ifneq ($(FREETZ_SHARE_terminfo),y)
 TOOLS_CONDITIONAL+=ncurses-host
+endif
```
scheint es zu funktionieren...

closed 2021-02-07 by fda77

gneiss15, 2021-02-07:
Der Kommentar [https://github.com/Freetz-NG/freetz-ng/issues/180#issuecomment-774686801](url) entspricht ja genau der von mir vorgeschlagenen Änderung 1 und dient (wie ich dort schon geschrieben habe) dazu das host-tool ncurses-host eben dann "zu bauen" bzw. nicht auszublenden, wenn terminfo angewählt ist. Also so OK. (übrigends nutzt der gesamte toolchain nur die tools aus dem Freetz-Pfad eine ggf. vorhanden "alte version auf dem PC" würde gar nicht erst genutzt).

Das 2. Problem, das terminfo nur (für das target) gebaut wird wenn ncurses angewählt ist liegt eben an der IMHO falschen Hierarchie. Die drei Pakete ncurses, ncursesw & terminfo sind eigentlich "unabhängige" teile. wobei eben für terminfo eine Abhängigkeit zu ncurses-host besteht. Leider wird aber (derzeit) terminfo nur im Makefile.in von ncurses gehandelt und das ist der Fehler.

Mit der Änderung oben wird (wie Du schon angemerkt hast jetzt zwar ncurses-host gebaut sobald FREETZ_SHARE_terminfo gesetzt ist. Allerdings wird dann immer noch nicht ncurses-terminfo gebaut (wenn nicht auch ncurses ausgewählt wurde).

Das beseitigt meine 2. vorgeschlagene Änderung in der make/libs/ncursesw/Makefile.in:
ifeq ($(strip $(FREETZ_SHARE_terminfo)),y)
ifneq ($(strip $(FREETZ_LIB_libncurses)),y) # ncurses-terminfo nicht doppelt eintragen! wenn libncurses selektiert ist wird das schon gemacht
LIBS+=ncurses-terminfo
endif
endif

Weil eben ncurses-terminfo gebaut werden muss, wenn FREETZ_SHARE_terminfo gesetzt ist, das aber bisher nur in make/libs/ncurses/Makefile.in angefordert wird. Was aber nur "abgearbeitet" wird, wenn FREETZ_LIB_libncurses gesetzt ist, was eben nicht der Fall ist, wenn nur FREETZ_LIB_libncursesw gewählt ist.

Den Fall kannst Du ebenso leicht nachstellen wie Du es oben schon für terminfo angesprochen hast.
> Um den Fehler nachzustellen, reicht es Shared libraries > ncurses > terminfo database auszuwählen...
Wenn Du nur das ausgewählt hast, sollte dann immer noch der spätere Fehler auftreten...
Nur wenn Du auch ncurses ausgewählt hast läuft alles durch.
Wenn Du aber nur noch ncursesw ausgewählt hast, fehlt immer noch terminfo.

Zum bessern Verständnis füge ich mal die build-abbrüche ohne 1. und 2. Änderung bei

gneiss15, 2021-02-07:
Fehlermeldung ohne meine 1. Änderung (in tools/make/Makefile.in)

building ... make[1]: Verzeichnis „/home/gneiss/Freetz/Ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc“ wird betreten
WHICH_XTERM=xterm-new \
XTERM_KBS=BS \
datadir=/usr/share \
/bin/sh ./gen_edit.sh >run_tic.sed
mkdir -p /home/gneiss/Freetz/Ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/tabset
echo '** adjusting tabset paths'
** adjusting tabset paths
sed -f run_tic.sed ../misc/terminfo.src >terminfo.tmp
/bin/sh ./gen-pkgconfig
DESTDIR=/home/gneiss/Freetz/Ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc \
prefix=/usr \
exec_prefix=/usr \
bindir=/usr/bin \
top_srcdir=.. \
srcdir=. \
datadir=/usr/share \
ticdir=/usr/share/terminfo \
source=terminfo.tmp \
cross_compiling=yes \
/bin/sh ./run_tic.sh
** Building terminfo database, please wait...
** creating ncurses.pc
** creating panel.pc
Running /home/gneiss/Freetz/Ng/tools/tic to install /home/gneiss/Freetz/Ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/terminfo ...

	You may see messages regarding extended capabilities, e.g., AX.
	These are extended terminal capabilities which are compiled
	using
		tic -x
	If you have ncurses 4.2 applications, you should read the INSTALL
	document, and install the terminfo without the -x option.

./run_tic.sh: 149: /home/gneiss/Freetz/Ng/tools/tic: not found
? tic could not build /home/gneiss/Freetz/Ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/terminfo
make[1]: *** [Makefile:93: install.data] Fehler 1
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
** creating menu.pc
** creating form.pc
touch pc-files
make[1]: Verzeichnis „/home/gneiss/Freetz/Ng/source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/misc“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/ncurses/ncurses.mk:83: /home/gneiss/Freetz/Ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/usr/share/tabset/std] Fehler 1
g

gneiss15, 2021-02-07:
Fehlermeldung ohne meine 2. Änderung (in make/libs/ncursesw/Makefile.in)

/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j5  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; 	if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2 \
	libs panel menu form headers
make[1]: *** source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2: Datei oder Verzeichnis nicht gefunden.  Schluss.

ERROR: Build failed.
make: *** [make/libs/ncurses/ncurses.mk:71: source/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/ncurses-6.2/lib/libncurses.so.6.2] Fehler 1

fda77, 2021-02-07:
Ich hab jetzt für eine 7590 eine .config erstellt in der so gut wie nichts selektiert ist, ausser "FREETZ_SHARE_terminfo". "make" lief durch

gneiss15, 2021-02-09:
>Kamen deine Fehler evtl davon...
Nein, ich habe bisher eigentlich immer nur "make menuconfig & make" benutzt (nur einmal bei der suche nach dem 1. Fehler explizit "make tools") sonst nichts...

Ich mache dann mal ein paar Versuche...
Das mit dem "clean" lasse ich mal, bin mir da nie sicher ob da nicht doch noch irgendwo "Reste" herrumliegen" die ein sauberes make verhindern.
Ich checke lieber immer neu aus und fange von vorne an...
Mein Rechner ist schnlee genug, so das mich das nur ein paar zusätzliche Minuten kostet. Die Sicherheit ist mir das allemal Wert.

Also hier meine Ergebnisse:
1. Test:
- freetz-ng neu ausgechecked
  da ist ja mein Änderungsvorschag zu tools/make/Makefile.in schon drin
- make menuconfig 
  Ausgewählt: 
  -- Level of user competence: Expert
  -- Hardware type: 7490
  -- Hardware series: 4&5
  -- Shared libraries > ncurses: terminfo database, unterpunkte auf default belassen
- make
Ergebnis: Alles OK

2. Test:
- make menuconfig
  zusätzlich Ausgewählt:
  -- Shared libraries > ncurses: libncursesw
- make
Ergebnis: Alles OK

Also liegt es nicht an den beiden libs....
Da ich ja schon mal ein make gemacht hatte was durchlief und später zusätlich mc ausgewählt hatte, vermute ich das dort das Problem liegt.

Also 3. Test:
- make menuconfig
  Abgewählt:
  -- Shared libraries > ncurses: libncursesw
  -- Shared libraries > ncurses: terminfo database
  neu Ausgewählt:
  -- Pakages > m > mc: , unterpunkte auf default belassen
 Damit sind die beiden oben abgewählten Libs automatisch wieder angewählt (ist ja auch richtig so)
- make
Ergebnis: Alles OK

Hm, das war es also nicht...
Dann werde ich jetzt mal nach und nach die config der bei der der Fehler auftrat anpassen...

So nach zig weiteren makes....

Der Fehler trat auf, als ich samba ausgewählt hatte. Zunächst mit den von mir ursprünglich ausgewählten sub-optionen:
FREETZ_PACKAGE_SAMBA=y
# FREETZ_SAMBA_VERSION_3_0 is not set
FREETZ_SAMBA_VERSION_3_6=y
FREETZ_PACKAGE_SAMBA_SMBD=y
FREETZ_PACKAGE_SAMBA_NMBD=y
# FREETZ_PACKAGE_SAMBA_SMBCLIENT is not set
# FREETZ_PACKAGE_SAMBA_NMBLOOKUP is not set
FREETZ_PACKAGE_SAMBA_MAX_DEBUG_LEVEL=-1

Dann habe ich zum Test mal NMBD abgewählt.
Jetzt lief alles durch.

Dann habe ich NMBD wieder ausgewählt.
... und es lief auch durch ???

Kann es sein das das ein "Timing-Problem" ist ?

Standardmäßig wird ja make mit -k 2 ausgeführt (also 2 parallele Prozesse FREETZ_JLEVEL=2). Ich habe (mein Rechner ist potent genug) das auf 5 hoch gesetzt (nicht bei diesen Tests aber vorher)
Wenn es sich wirklich um ein solches Problem handelt ist die Wahrscheinlichkeit für solche Problem ja um so höher je mehr parallele Prozesse laufen....

Egal. Ich denke es handelt sich hier um ein "Minor-"Problem. Den waren Grund herauszufinden würde sicherlich noch zig Tests erfordern und das ist es IMHO nicht Wert.

fda77, 2021-02-09:
Normal setzt man JLEVEL um 1 höher als man Cores hat, meine Freetz-VM hat 7 Cores und ich nutze JLEVEL=8
Falls du öfter "distclean" nutzt wäre es sinvoll CCACHE im menuconfig zu aktivieren, dann geht das compilieren recht flott.
Ein "Timing" Problem kann auftreten wenn irgendwo eine Abhängigkeit fehlt. Das kam bei Freetz schon öfter vor. Zb a und b brauchen c, aber nur in a ist eine Abhängigkeit definiert. Dann gehts mal und mal nicht...

Bei Samba wählt nur FREETZ_PACKAGE_SAMBA_SMBCLIENT das FREETZ_LIB_libncurses: https://github.com/Freetz-NG/freetz-ng/blob/master/make/samba/Config.in#L54
Aber abhängig ist es auf dem host eigentlich immer davon: https://github.com/Freetz-NG/freetz-ng/blob/master/make/samba/samba.mk#L51

gneiss15, 2021-02-10:
Ja, genau, ich habe (in der VM) 4 Cores, deshalb auch die 5... Da sind wir also auf dem gleichen Standpunkt ;-)

... und wenn es bei Dir "direkt" durchgelaufen ist, bei mir erst beim 2. Durchlauf muss wohl noch irgendwo etwas sein...
Wenn es aber durch ein 2. make durchläuft soll mir das auch egal sein. Ich versuche jetzt erst mal alles so hinzubekommen, das mein Box so läuft wie ich es möchte. Da hab ich noch genug zu tuen... bzw. das ist mir erst mal wichtiger.