Freetz-NG/freetz-ng #186: smartmontools compile fails with "undefined reference to `__stack_chk_guard'"

edeso, 2021-02-08:
compiling todays clone for FB7360v2 based on Fritz!OS 6.86 fails with the failure attached below

disabling stackprotector by adding 
`$(PKG)_CONFIGURE_OPTIONS += CFLAGS="$(TARGET_CFLAGS) -fno-stack-protector" CXXFLAGS="$(TARGET_CFLAGS) -fno-stack-protector"`
in `make/smartmontools/smartmontools.mk` helps but should probably only be activated for older kernels. dunno how to do that.

```
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper -DHAVE_CONFIG_H -I.  -DBUILD_INFO='"(local build)"' -DSMARTMONTOOLS_SYSCONFDIR='"/etc"' -DSMARTMONTOOLS_SMARTDSCRIPTDIR='"/etc"' -DSMARTMONTOOLS_DRIVEDBDIR='"/usr/share/smartmontools"'        -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -W -fstack-protector -MT smartctl.o -MD -MP -MF .deps/smartctl.Tpo -c -o smartctl.o smartctl.cpp
dev_jmb39x_raid.cpp: In member function 'virtual ata_device* smart_interface::get_jmb39x_device(const char*, smart_device*)':
dev_jmb39x_raid.cpp:735:98: warning: 'version' may be used uninitialized in this function [-Wmaybe-uninitialized]
   ata_device * jmbdev = new jmb39x::jmb39x_device(this, smartdev, type, version, port, lba, force);
                                                                                                  ^
mv -f .deps/utility.Tpo .deps/utility.Po
mv -f .deps/dev_jmb39x_raid.Tpo .deps/dev_jmb39x_raid.Po
mv -f .deps/dev_interface.Tpo .deps/dev_interface.Po
In file included from /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/include/uClibc++/map:24:0,
                 from smartd.cpp:32:
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/include/uClibc++/associative_base: In instantiation of 'std::__base_associative<Key, ValueType, Compare, Allocator>::__base_associative(const Compare&, const Allocator&, const key_type (*)(std::__base_associative<Key, ValueType, Compare, Allocator>::value_type)) [with Key = std::basic_string<char>; ValueType = std::pair<std::basic_string<char>, std::basic_string<char> >; Compare = std::less<std::basic_string<char> >; Allocator = std::allocator<std::basic_string<char> >; std::__base_associative<Key, ValueType, Compare, Allocator>::key_type = std::basic_string<char>; std::__base_associative<Key, ValueType, Compare, Allocator>::value_type = std::pair<std::basic_string<char>, std::basic_string<char> >]':
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/include/uClibc++/associative_base:500:25:   required from 'std::__single_associative<Key, ValueType, Compare, Allocator>::__single_associative(const Compare&, const Allocator&, const key_type (*)(std::__single_associative<Key, ValueType, Compare, Allocator>::value_type)) [with Key = std::basic_string<char>; ValueType = std::pair<std::basic_string<char>, std::basic_string<char> >; Compare = std::less<std::basic_string<char> >; Allocator = std::allocator<std::basic_string<char> >; std::__single_associative<Key, ValueType, Compare, Allocator>::key_type = std::basic_string<char>; std::__single_associative<Key, ValueType, Compare, Allocator>::value_type = std::pair<std::basic_string<char>, std::basic_string<char> >]'
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/include/uClibc++/map:106:25:   required from 'std::map<Key, T, Compare, Allocator>::map(const Compare&, const Allocator&) [with Key = std::basic_string<char>; T = std::basic_string<char>; Compare = std::less<std::basic_string<char> >; Allocator = std::allocator<std::basic_string<char> >]'
smartd.cpp:5397:25:   required from here
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/include/uClibc++/associative_base:77:11: warning: unused parameter 'A' [-Wunused-parameter]
  explicit __base_associative(const Compare& comp, const Allocator& A, const key_type (*v_to_k)(const value_type))
           ^
mv -f .deps/knowndrives.Tpo .deps/knowndrives.Po
mv -f .deps/dev_areca.Tpo .deps/dev_areca.Po
mv -f .deps/json.Tpo .deps/json.Po
mv -f .deps/nvmeprint.Tpo .deps/nvmeprint.Po
mv -f .deps/scsicmds.Tpo .deps/scsicmds.Po
mv -f .deps/scsiata.Tpo .deps/scsiata.Po
mv -f .deps/atacmds.Tpo .deps/atacmds.Po
mv -f .deps/scsiprint.Tpo .deps/scsiprint.Po
mv -f .deps/smartctl.Tpo .deps/smartctl.Po
mv -f .deps/os_linux.Tpo .deps/os_linux.Po
mv -f .deps/smartd.Tpo .deps/smartd.Po
/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper  -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -W -fstack-protector   -o smartd smartd.o atacmdnames.o atacmds.o dev_ata_cmd_set.o dev_intelliprop.o dev_interface.o dev_jmb39x_raid.o knowndrives.o nvmecmds.o scsicmds.o scsiata.o scsinvme.o utility.o     os_linux.o cciss.o dev_areca.o      
smartd.o: In function `vsyslog_lines(int, char const*, void*)':
smartd.cpp:(.text+0x110): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x118): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1a8): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1b4): undefined reference to `__stack_chk_fail'
smartd.o: In function `read_ata_error_count(ata_device*, char const*, firmwarebug_defs, bool)':
smartd.cpp:(.text+0x438): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x448): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x4dc): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x4e8): undefined reference to `__stack_chk_fail'
smartd.o: In function `read_dev_state(char const*, persistent_dev_state&)':
smartd.cpp:(.text+0xeb4): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0xebc): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x127c): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1284): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1290): undefined reference to `__stack_chk_fail'
smartd.o: In function `reset_warning_mail(dev_config const&, dev_state&, int, char const*, ...)':
smartd.cpp:(.text+0x1474): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1484): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x153c): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1548): undefined reference to `__stack_chk_fail'
smartd.o: In function `next_scheduled_test(dev_config const&, dev_state&, bool, long)':
smartd.cpp:(.text+0x164c): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1654): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1b1c): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1b24): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x1b30): undefined reference to `__stack_chk_fail'
smartd.o: In function `MailWarning(dev_config const&, dev_state&, int, char const*, ...)':
smartd.cpp:(.text+0x23b8): undefined reference to `__stack_chk_guard'
smartd.cpp:(.text+0x23c0): undefined reference to `__stack_chk_guard'
smartd.o: In function `main_worker(int, char**)':
smartd.cpp:(.text+0x9d44): undefined reference to `__stack_chk_guard'
smartd.o: In function `MailWarning(dev_config const&, dev_state&, int, char const*, ...)':
smartd.cpp:(.text+0x2b70): undefined reference to `__stack_chk_guard'
[... 546 lines cut ...]
dev_areca.cpp:(.text+0x574): undefined reference to `__stack_chk_guard'
dev_areca.cpp:(.text+0x57c): undefined reference to `__stack_chk_guard'
dev_areca.cpp:(.text+0x588): undefined reference to `__stack_chk_fail'
dev_areca.o: In function `generic_areca_device::arcmsr_ui_handler(unsigned char*, int, unsigned char*)':
dev_areca.cpp:(.text+0x5cc): undefined reference to `__stack_chk_guard'
dev_areca.cpp:(.text+0x5d4): undefined reference to `__stack_chk_guard'
dev_areca.cpp:(.text+0x750): undefined reference to `__stack_chk_guard'
dev_areca.cpp:(.text+0x75c): undefined reference to `__stack_chk_fail'
dev_areca.o: In function `generic_areca_device::arcmsr_ata_pass_through(ata_cmd_in const&, ata_cmd_out&)':
dev_areca.cpp:(.text+0x788): undefined reference to `__stack_chk_guard'
[... cut ...]
dev_areca.cpp:(.text+0xce8): undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1039: smartd] Error 1
make[2]: *** Waiting for unfinished jobs....
mv -f .deps/ataprint.Tpo .deps/ataprint.Po
make[2]: Leaving directory '/home/user/freetz-ng-7360v2/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2'
make[1]: *** [Makefile:938: all] Error 2
make[1]: Leaving directory '/home/user/freetz-ng-7360v2/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2'

fda77, 2021-02-08:
Afaik "stack-protector"-Supports depends on the used gcc version. But there is _MIN/_MAX missing for the FREETZ_TARGET_GCC_x_x: https://github.com/Freetz-NG/freetz-ng/blob/master/config/ui/toolchain.in#L315

edeso, 2021-02-08:
actually smartmontools configure tries to determine stackprotector support and determines `checking whether ...mips-linux-uclibc-g++-wrapper supports -fstack-protector... yes` s.b.
but compilation then fails with the error above. maybe some import is missing?

```
---> package/smartmontools ... preparing ... mkdir -p source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2; tools/gunzip -c dl/smartmontools-7.2.tar.gz | tools/tar-gnu -x -C source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2 --transform='s|^./\+||' --strip-components=1 
set -e; shopt -s nullglob; for i in make/smartmontools/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2 $i; done; 
configuring ... (conf_cmd() { ./configure  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; cd source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/smartmontools-7.2;  rm -f config.{cache,status};    PATH="/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin:/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" FREETZ_LIBRARY_DIR="/usr/lib/freetz" FREETZ_KERNEL_VERSION_MAJOR="3.10" FREETZ_TARGET_ARCH="mips" FREETZ_TARGET_ARCH_ENDIANNESS_DEPENDENT="mips" CC="/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CXX="/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper" CFLAGS="-march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CXXFLAGS="-march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" LDFLAGS="" PKG_CONFIG_PATH="/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/../lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/../lib/pkgconfig" GLOBAL_LIBDIR=/home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/usr/lib  CONFIG_SITE=/home/user/freetz-ng-7360v2/include/config.site/mips-linux-uclibc conf_cmd   --cache-file=/dev/null --target=mips-linux --host=mips-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --datadir=/usr/share --includedir=/usr/include --infodir=/usr/share/info --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --mandir=/usr/share/man --sbindir=/usr/sbin --sysconfdir=/etc --with-gnu-ld --disable-nls --without-nvme-devicescan   )
configure: WARNING: unrecognized options: --with-gnu-ld, --disable-nls
configure: loading site script /home/user/freetz-ng-7360v2/include/config.site/mips-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for mips-linux-strip... mips-linux-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper... gcc3
checking for mips-linux-gcc... /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc
checking whether we are using the GNU C compiler... yes
checking whether /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc accepts -g... yes
checking for /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc option to accept ISO C89... none needed
checking whether /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc understands -c and -o together... yes
checking dependency style of /home/user/freetz-ng-7360v2/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc... gcc3
[... cut ...]
local drive database:   /etc/smart_drivedb.h
smartd config file:     /etc/smartd.conf
smartd warning script:  /etc/smartd_warning.sh
smartd plugin path:     /etc/smartd_warning.d
PATH within scripts:    /usr/local/bin:/usr/bin:/bin
smartd initd script:    [disabled]
smartd save files:      [disabled]
smartd attribute logs:  [disabled]
SELinux support:        no
libcap-ng support:      no
systemd notify support: no
NVMe DEVICESCAN:        no
-----------------------------------------------------------------------------

fda77, 2021-02-08:
I dont know ... maybe the old gcc is not okay
From configure of smartmontools:
```
  # Add -Wformat=2 (GCC 3.0) -fstack-protector[-strong] (GCC 4.1[4.9]) if supported
  # and no -W or -f option was set in configure cmdline (TODO: -Wformat-signedness)
  for option in "-Wformat=2" "-fstack-protector-strong" "-fstack-protector"; do
    case " $orig_CXXFLAGS:$option" in *\ -W*:-W*|*\ -f*:-f*) continue ;; esac
    case " $CXXFLAGS:$option" in *\ -fstack-p*:-fstack-p*) continue ;; esac
    AC_MSG_CHECKING([whether $CXX supports $option])
    save_CXXFLAGS=$CXXFLAGS
    CXXFLAGS="$CXXFLAGS $option"
    AC_LINK_IFELSE([AC_LANG_PROGRAM()], [res=yes], [res=no; CXXFLAGS=$save_CXXFLAGS])
    AC_MSG_RESULT([$res])
  done
```

fda77, 2021-02-08:
https://github.com/apple/cups/issues/3198
> Actually try to link an executable to test for stack protector support,
> as gcc seems to accept the -fstack-protector argument even when it's
> compiled with --disable-libssp (but linking fails with a error
> finding -lssp_nonshared).

closed 2021-02-08 by fda77

edeso, 2021-02-10:
upstream has a more elegant fix for the gcc stackprotection detection
https://www.smartmontools.org/ticket/1439

attached patch implements it and can get dropped on the next bump.
[upstream_gcc_stackprotect_detection_fix.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/5960863/upstream_gcc_stackprotect_detection_fix.patch.txt)

it'll make the make file gcc version conditionj obsolete.

fda77, 2021-02-10:
Danke!
Man könnte auch dies als Patch nehmen: https://www.smartmontools.org/changeset/5208
Und wie in https://www.smartmontools.org/ticket/1439 steht `$(PKG)_CONFIGURE_PRE_CMDS += $(AUTORECONF)` zur smartmonools.mk hinzufügen

edeso, 2021-02-11:
hat schon nen Grund warum ich das so geliefert habe.

der zusätzliche **auto(re)conf** run ist ja nur nötig um die Änderungen aus der `configure.ac` in das `configure` Skript zu giessen. mit dem nä. Versions bump würde der dann immer noch unnötigerweise ausgeführt.
mein Patch ist ein diff nach dem **auto(re)conf** run und enthält beide Änderungen. der kann dann weg beim Version bump und alles ist wieder sauber.

aber macht wie ihr denkt. hauptsache das tut!