Freetz-NG/freetz-ng #197: host-tools Abhänigkeiten: Juis-Erkennung benötigt yourfritz-host

fda77, 2021-02-13:
@Usprünglich hier berichtet: https://github.com/PeterPawn/YourFritz/issues/37#issuecomment-778323881
Nach `distclean` kann wegen fehlendem `yourfritz-host` Juis mit `Detect the latest firmware version` nicht genutzt werden.
Wenn man dem `image:` target `youfritz-host` hinzufügt funktioniert das ganze, aber der Firmwaredownload erfolgt erst nach: `tar, avm-rle, busybox, dtc, e2fsprogs, fakeroot, find-squashfs, genext2fs, lzma465, xz, lzma2eva, patchelf, sed, sfk, squashfs2-4, uimg, yourfritz-akc-host`
Eigentlich war die ursprüngliche Idee die Firmware so früh wie möglich herunterzuladen, damit ein nicht funktionierenden Download nicht erst nach 2 Stunden auffällt!
@PeterPawn: Soll ich diese 3 cherrypicken bzw PR https://github.com/PeterPawn/YourFreetz/commits/host-tools ? xz gehört aber eigentlich zu Packer und Entpacker

PeterPawn, 2021-02-13:
> Soll ich diese 3 cherrypicken

Da sollte man erst noch ein wenig drauf herumdenken ... ich bin der Ansicht, daß Du hier eher auf eine "schnelle Lösung" setzen könntest (die aber deshalb nicht "dirty" sein muß). 

Wie schon angedeutet, ist das mit dem "bootstrapping" in Freetz schon immer etwas merkwürdig - für das Entpacken eines Pakets ist normalerweise die Variable `$(UNPACK_TARBALL_PREREQUISITES)` als Abhängigkeit definiert und die ist mit der `BusyBox` und dem `GNU-tar` für den Build-Host vorbelegt: https://github.com/Freetz-NG/freetz-ng/blob/7472a618f72985fa0c5ebb53c4dd15241320b5bf/include/make/020-helper-functions.mk#L170 (ich bleibe jetzt mal im Freetz-NG-Repo mit den Links)

Nun ist aber der Quelltext für das `GNU-tar` selbst wieder mit LZMA-Kompression gepackt: https://github.com/Freetz/freetz/blob/9208f44a00d980258fdf532957ac24dc3f03c025/tools/make/tar-host/tar-host.mk#L2 und daher bräuchte es wieder ein `unxz`, wenn man das auch mit den Freetz-Helper-Funktionen entpacken wollte: https://github.com/Freetz-NG/freetz-ng/blob/7472a618f72985fa0c5ebb53c4dd15241320b5bf/include/make/020-helper-functions.mk#L153

Das würde zu einer zirkulären Abhängigkeit führen und so wird für dieses `tar-host`-Paket beim Entpacken gar nicht auf diese Helper-Funktionen zurückgegriffen, sondern das wird direkt mit einem `tar`-Aufruf entpackt, wobei anhand der Erweiterung jetzt vom `tar` das `xz` als Filter für das Dekomprimieren vor dem Entpacken aufgerufen wird: https://github.com/Freetz/freetz/blob/9208f44a00d980258fdf532957ac24dc3f03c025/tools/make/tar-host/tar-host.mk#L15 ... denn letztlich ist das `unxz`, was im o.a. Helper-File aufgerufen werden soll, üblicherweise nichts anderes als ein Symlink auf `xz` (das man auch mit passendem Parameter (`-d` bzw. `--decompress`) aufrufen kann, wenn man damit entpacken will - ansonsten wird die Operation auch durch den Aufruf-Namen `unxz` festgelegt). 

Die Existenz von `tar` und `xz` gehört dann ja zu den Voraussetzungen in der `.build-prerequisites`: https://github.com/Freetz-NG/freetz-ng/blob/master/.build-prerequisites

Und da würde ich jetzt auch ansetzen ... anstatt im `yourfritz-host`-Paket die Abhängigkeit von `$(UNPACK_TARBALL_PREREQUISITES)` zu definieren: https://github.com/Freetz-NG/freetz-ng/blob/7472a618f72985fa0c5ebb53c4dd15241320b5bf/tools/make/yourfritz-host/yourfritz-host.mk#L33, würde ich hingehen und das Paket nicht durch den Aufruf von `UNPACK_TARBALL` entpacken lassen (dann braucht's auch die Abhängigkeit von der `BusyBox`, die ihrerseits das `unxz` bereitstellen soll, nicht länger und das `tar` ist ja schon Voraussetzung, ebenso das `xz`), sondern einen direkten `tar`-Aufruf verwenden, so wie er auch hier zu sehen ist: https://github.com/Freetz/freetz/blob/9208f44a00d980258fdf532957ac24dc3f03c025/tools/make/tar-host/tar-host.mk#L15 (natürlich mit angepaßten Namen)

Das ist dann auch wieder eine "saubere Lösung" - und sie kann erst mal in Ruhe ausgetestet werden, denn sie ändert nur sehr wenig gegenüber dem bisherigen Zustand.

Bei Gelegenheit kann man dann immer noch hingehen und die Helper-Funktion in `include/make/020-helper-functions.mk` für die `xz`-Erweiterung mal auf den Aufruf von `xz -d -c $(1)` umstellen (und das auch ohne `$(TOOLS_DIR)`) - denn es erscheint zumindest nicht schlüssig, warum das mit dem Entpacken von LZMA-Archiven beim `tar-host` funktionieren soll und danach dann nicht mehr oder gar nie wieder. Ohne die Notwendigkeit von `unxz` braucht man dann zum Entpacken von `.tar.xz` aber auch den Rest aus `$(UNPACK_TARBALL_PREREQUISITES)` nicht mehr.

Und dann kann man sogar hingehen und bei den Paketen, bei denen die Quelldatei tatsächlich ein solches mit LZMA komprimiertes TAR-File ist, die Abhängigkeit von `$(UNPACK_TARBALL_PREREQUISITES)` beim `$($(PKG)_DIR)/.unpacked`-Target (oder wie das dann im `.mk`-File auch heißen mag) entfernen und _trotzdem_ beim Aufruf von `UNPACK_TARBALL` bleiben - wenn dann tatsächlich nur ein solches Paket gebaut werden soll, entfällt das Erzeugen der `BusyBox` und des `tar-host`-Pakets so lange, bis irgendein anderes Paket dann doch wieder davon abhängig ist.

Aber das könnte dann die Änderungen in meinem `host_tools`-Branch ersetzen bzw. ist ein anderer Ansatz zur Lösung des dahinter stehenden Problems. Aber egal, wie man das (später) auch machen will - die "schnelle Lösung" für `yourfritz-host` ist (und bliebe) davon unabhängig. Wenn man das dann im `.mk`-File auch noch passend kommentiert (und im Commit-Text), weiß auch der Nächste, der das vielleicht auf eine andere Kompression umstellen will, Bescheid und kann entsprechende Vorkehrungen treffen - ja, notfalls kann er auch die Abhängigkeit einfach wieder hinzufügen.

fda77, 2021-02-15:
Das würde dann nur yourfritz-host helfen und bei jedem anderen Pakete würde dann busybox und tar dann doch wieder gebaut. Wenn bb und tar aber mit den Programmen vom Host entpackt werden können, müsste dies auch mit allen anderen funktionieren... Ich hab da aber eine ganz andere Idee wie man dies "lösen" könnte: Einfach ein download-tools Paket. Alle zu bauen dauert bei mir etwa 3 Minuten, was sich jeder sparen könnte, und auch den Platz um diese zu bauen. Alle Tool als i586 32bit statisch sind gepackt etwa 15 MB, was selbst in Deutschland halbwegs schnell herunterzuladen ist. Man könnte das dann noch zB für Raspberry/ARM erstellen, mit meinem aarch64 ist das aber nciht möglich.
Ich weiss aber noch nicht wie ich alle "$TOOL-host" targets überschreiben bzw durch eine Option deaktivieren kann, ohne jedes selbst zu verändern. Ausserdem verursacht ein statische tar ein Segfault (evtl wegen nss). Wie ds mit den libs von Fakeroot funktionieren soll weiss ich auch noch nicht, evtl wäre es ein gute Zeitpunkt zu Pseudo zu wechseln

PeterPawn, 2021-02-15:
> Das würde dann nur yourfritz-host helfen und bei jedem anderen Pakete würde dann busybox und tar dann doch wieder gebaut. 

Deshalb schrieb ich ja "schnell" und mit so wenig Änderungen wie notwendig. Das mit den `host-tools` ist jetzt schon seit x Jahren so und es hat sich niemand daran gestört (oder eher selten) und wenn Du jetzt für den Download des Images, der bei Freetz auch schon immer am Beginn erfolgt (was ebenfalls nicht notwendig wäre, die `kernel.image` würde erst gebraucht, wenn man einen eigenen Kernel linken will und dafür den AVM-Inhalt von `avm_kernel_config` benötigt und den Rest braucht man auch erst wieder, wenn's ans `fwmod` geht), zusätzliche Tools brauchst, dafür aber nicht erst die kompletten `host-tools` erstellen willst, dann wäre die Änderung in der `yourfritz-host.mk` eine sowohl schnelle, als auch eine saubere Lösung.

Den Rest solltest Du Dir gut überlegen - die `BusyBox` braucht's in diesem Schritt eigentlich noch gar nicht, wenn man die diversen Packer für die in Frage kommenden Formate zu den "Voraussetzungen" erklärt - das ist ja praktisch der Inhalt meines `host_tools`-Branches und mit einer aktuellen GNU-tar-Version, die sich auf `--format=gnu` versteht und dann solche Dinge wie PaxHeaders auch wegläßt, kann man auch das `tar` vom Host nehmen und ich kenne praktisch kein System (selbst da nicht, wo die Distro-eigene Paketverwaltung auf `cpio` setzt), wo das `tar` nicht zum Standardumfang gehören würde. Sofern man die Unterschiede im Format kennt und das `tar` mit passenden Parametern aufruft, braucht es das aus der BusyBox auch nicht zwingend zum Zusammenpacken des kompletten Firmware-Images.

Ich würde mir das jedenfalls gut überlegen mit den `host-tools` - so viel ist das tatsächlich nicht, einiges braucht obendrein passende Infos aus der AVM-Firmware (die Heuristik, die @er13 da für die `avm_kernel_config`-Größe eingebaut hat, ist m.E. auch immer noch modellabhängig - wobei ich dafür die Hand nicht ins Feuer legen würde, denn da hatte ich mich ja ausgeklinkt) und dafür jetzt ein "Download-Paket" zu schnüren, widerspricht etwas dem Grundkonzept von Freetz. Dann geht lieber gleich hin und lasse diese Tools in einer passenden Container-Umgebung erstellen (von Google Cloud über AWS bis zu irgendwelchen Build-Services der Linux-Distros - ich weiß gerade nicht, ob es den OBS für OpenSuSE noch gibt) - das geht dann wenigstens automatisch und man halst sich damit keine zusätzliche (Hand-)Arbeit auf. Das Problem der "Modellabhängigkeit" bei einigen dieser Tools kann man dann auch wieder dadurch lösen, daß man die "pro Modell" einmal mit passenden Einstellungen erstellen läßt.

Aber wenn Du schon bei "Download-Paketen" bist ... DAS wäre doch mal ein Fortschritt (auch für viele Freetz-(NG-)Benutzer), wenn da nicht jeder immer alle OpenSource-Pakete selbst übersetzen müßte. Ein `dropbear` ist nun mal für alle VR9-Modelle genauso dasselbe, wie für alle GRX-Modellen und selbst alle IPQ-Boxen - immer vorausgesetzt, die verwenden dieselben dynamischen Bibliotheken und auch die "Standard-Kernel" (wenn jemand zusätzliche Module übersetzen lassen will) sind üblicherweise dieselben, solange AVM die Kernel-Version nicht wechselt - trotzdem MUSS jeder Freetz-Benutzer den Kernel bei sich selbst kompilieren lassen und das dauert nun mal auch seine Zeit; von der Verschwendung von Ressourcen ganz zu schweigen. 

DAS wäre doch auch eine (sinnvolle) Spielwiese ... solange sich die Kernel-Konfiguration nicht ändert (und da rede ich jetzt nicht davon, daß da irgendein LKM zusätzlich gebaut werden soll), würde schon die Bereitstellung der Object-Files aus einem Kernel-Build (dann natürlich nicht unbedingt mit den Pfaden für eine spezielle C-Library garniert) einiges an Zeit sparen (auch wenn der Kernel längst nicht bei jedem Freetz-(NG-)Benutzer zu übersetzen ist) und das bißchen Zeit, was dann noch zum Linken gebraucht wird (und erst da gehen dann "Schwestermodelle" wie 7590 und 7580 wieder so richtig getrennte Wege), kann man in den Skat drücken. Dann müßte man den ganzen Schnodder tatsächlich nur noch einmal ausführen (lassen), wenn AVM mit einem neuen Kernel (oder einer umfangreich geänderten Konfiguration) um die Ecke kommt.

Da ist dann auch nichts "host-spezifisch" an den Download-Paketen ... ganz im Gegensatz zu den `host-tools` (hier geht es ja schon bei der Architektur los) und die deshalb jetzt alle statisch zu bauen, weil irgendwo ggf. eine andere `glibc` (oder eine ganz andere C-Library) verwendet werden könnte, ist auch nur bedingt zielführend. Denn - im Gegensatz zu den FRITZ!Boxen von AVM, wo der Hersteller selbst schon aufpaßt, daß die "Architektur" des Systems einigermaßen ähnlich ist - die Unterschiede zwischen verschiedenen Linux-Distributionen sind schon gewaltig (das hast Du selbst mit Deinem Fedora vs. Ubuntu gemerkt) und deren Zahl ist praktisch unüberschaubar. Das macht die Idee mit einem Download-Paket für die `host-tools` in meinen Augen zu einer Dauerbaustelle, die mehr Arbeit und Ärger einbringt, als irgendjemandem einen echten Vorteil.

Wenn Du das Problem für alle Pakete gleichzeitig (oder zumindest für viele auf einmal) lösen willst, solltest Du da noch ein paar Tage drüber nachdenken - die mit der Idee verbundene Euphorie sollte erst einmal verflogen sein, damit auch die Nachteile eine Chance haben, in Erwägung gezogen zu werden.

Daß die "Container-Idee" auch gar nicht so abwegig ist, kannst Du Dir z.B. bei den Leuten vom FKIE ansehen - die bauen für ihr `FACT` die Tools zum Entpacken der AVM-Images (im Rahmen von `fact_extractor`: https://github.com/fkie-cad/fact_extractor/blob/master) tatsächlich in einem Container und auch bei denen wird der ganze Rotz im Rahmen des `make tools` erst einmal mitgebaut, obwohl sie eigentlich nur die SquashFS-Tools mit den AVM-Spezialitäten brauch(t)en/bräuchten.

fda77, 2021-02-15:
Du findest die Idee nicht gut? Da ich für jedes Gerät ein eigenes Verzeichnis nutze würde es bei mir einiges schneller werden.  Eine Liste mit tool-Ausnahmen müsste schon sein. Die 4 libs, kconfig, fakeroot, akx.
Welche Rolle spielt denn glibc bei statischen Binaries? Zur Not müsste das mit zb Alpine Linux compiliert werden, welches musl nutzt. Wer eine andere Architektur hat muss halt wie bisher selbst bauen.
Ein vorkompilierte Kernel wäre schon nett. Aber ich denke dafür bin ich nicht der richtige. Zumal der kernel nur gebaut wird wenn man replacekernel aktiviert hat oder min 1 Modul ausgewählt hat. Und dann gibt es um die 60 Kernel in Freetz! https://github.com/Freetz-NG/freetz-ng/blob/master/config/mod/download.in#L171

PeterPawn, 2021-02-15:
> Du findest die Idee nicht gut?

Ich würde mir die Arbeit angesichts der doch minimalen Einsparungen nicht machen.

Auch bei statisch gebauten Binaries spielt die C-Library schon noch eine Rolle - immerhin wird das Binär-File ja trotzdem mit dieser C-Library gelinkt ... nur halt so, daß alle Funktionen der Library gleich in der Binärdatei enthalten sind und nicht erst auf einer Bibliothek zur Laufzeit nachgeladen werden müssen.

Trotzdem gibt es auch noch genug Unterschiede in der jeweiligen OS-Konfiguration, die auch Auswirkungen auf die C-Library haben. Nur mal als Beispiel ... bei OpenSuSE Tumbleweed (und iirc auch in den "stable releases") ist die `/etc/services` jetzt in das Verzeichnis `/usr/etc/services` gewandert und wenn irgendeine "Sockets"-Funktion der `glibc` jetzt einen Service-Namen in die passende Portnummer "übersetzen" will (https://www.gnu.org/software/libc/manual/html_node/Services-Database.html), dann muß sie (die Library) an einer anderen Stelle suchen - eben in `/usr/etc/services`. 

Und das ist nur einer der - sehr augenfälligen - Unterschiede, der mir jetzt spontan einfällt - die Distros unterscheiden sich schon sehr und die C-Library, auf der so ein System aufbaut, kann auch ziemlich unterschiedliche Konfigurationen aufweisen. Außerdem sollte man auch nicht komplett aus den Augen verlieren, daß lange nicht auf jedem 64-Bit-System der nach Deinen Plänen ja auch  notwendigen 32-Bit-Layer installiert sein muß bzw. ich würde den jetzt nicht - durch die kalte Küche - zur Voraussetzung per se machen. 

Mir ist auch bewußt, daß es noch Programme gibt (`avm_kernel_config` wäre ein solches), die ohne 32-Bit-Layer nicht arbeiten können, nur muß man das ja nicht unbedingt noch forcieren. Beim erwähnten Programm liegt es nun mal an der Arithmetik mit 32-Bit-Pointern, weil die AVM-Kernel halt auch alle 32-Bit-Adressen benutzen und das Programm halt mit den AVM-Adressen arbeitet/arbeiten muß - das KANN man irgendwann auch noch mal ändern, nur wäre es dann blöd, wenn 32-Bit auf anderen Wegen wieder "erzwungen" würden.

Wenn Du das mit dem ständigen Rebuilds (u.a. auch für die `tools`) tatsächlich vermeiden willst, würde ich Dir eher eine Art "Backup" für die Build-Artefakte empfehlen (so ähnlich, wie es beim Google Cloud Build Service möglich ist: https://cloud.google.com/build/docs/building/store-build-artifacts) - wenn der Freetz-Benutzer sich das dann "weglegen" kann auf seinem System und es nach einem frischen Checkout wiederherstellen kann (so eine Anpassung an das lokale System ist ja nach dem Klonen eines Repos auch nichts absolut Neues oder Ungewöhnliches), dann spart er auch massig Zeit. Bisher kümmert sich das `make` höchstens beim `clean`, `dirclean` oder `distclean` darum, die vorhandenen Zwischenergebnisse möglichst spurlos zu vernichten. Da wäre ein `make backup`, was genau diese Daten sichert (und sie meinetwegen dann immer noch löschen kann), schon eine Verbesserung.

Allerdings hat `git` hier tatsächlich einen Pferdefuß - es stellt bei einem frischen Checkout das Änderungsdatum einer Datei nicht wieder her - aber auch da kann man sich irgendwie helfen. Wenn man den Hash (also den Stand) des Repos zum Backup-Zeitpunkt kennt und den aktuellen eines frischen Checkouts, kann man sich die zwischenzeitlich geänderten Dateien ja ausgeben lassen und für alles das, was sich seit dem Backup nicht geändert hat, kann man ein neues Modifikationsdatum setzen, das NACH dem (aktuellen) Checkout liegt - damit ist dann `make` auch wieder zufrieden. Aber das Problem haben auch alle, die solche Build-Artefakte speichern lassen und sie danach erneut benutzen wollen - wobei es natürlich bei anderen Projekten lange nicht sooo extrem ist, wie bei Freetz(-NG), weil die eben auch mit unterschiedlichen "Teilprojekten" arbeiten, wo jedes für sich sein eigenes `make` macht und dafür auch sein eigenes `Makefile` hat. 

Das "alles in einem" ist ja auch wieder "old school" und seitdem das in Freetz mal so angelegt wurde, ist Speicherplatz für eine solches Backup (oder auch mehrere) von Build-Artefakten ständig billiger geworden. Das, was Freetz praktisch nach dem zweiten Schritt machen kann (wenn man keine Download-Toolchain verwendet), ist ja auch nichts anderes als so eine Sicherung von Build-Artefakten (ich meine hier natürlich die Download-Toolchains) ... da kann man natürlich auch noch um einiges ausweiten und es muß auch nicht immer zwingend eine "zentrale Bereitstellung" sein - so ein USB-Stick, auf dem man sich (meinetwegen sogar bei Verwendung der Freetz-VM von @gismotro) die Zwischenergebnisse eines Freetz-Builds auch mal "weglegen" kann, kostet praktisch nichts und ich wette, daß viele Freetz-Benutzer eher 10 EUR für einen passenden USB-Stick investieren würden, als bei jedem neuen Build erst mal alle Pakete erneut zu übersetzen ... und zwar auch diejenigen, bei denen das gar nicht notwendig wäre.

Die Annahme, daß sich jemand immer gleich die komplette VM weglegen kann/müßte, wäre zwar auch nicht von der Hand zu weisen ... aber auch die ganzen Live-Systeme, die von irgendeinem Stick gestartet werden, sichern ja nur das, was sie auch ändern und nicht immer gleich das komplette System auf einen Schlag.

Allerdings halte ich auch diese Backup-Thema nicht wirklich für vordringlich (es macht sogar ohne eine passende "Zerlegung" von Freetz in die "Funktionsgruppen" deutlich weniger Sinn, als nach einer solchen Aufteilung) - das wäre nur etwas, wo ich (eben auch wegen der Philosophie des "Sichern [... cut ...] sten wollen "alte" Kernel und auch die exotischen Modelle (bis hin zur 4020 oder 5530) muß man nicht alle "abdecken" wollen - es ginge darum, die am häufigsten benötigten "Build-Artefakte" irgendwo sinnvoll zu sichern (und zu verteilen), anstatt ständig dieselben Schritte (die auch bei allen Benutzern dann dasselbe Ergebnis bringen) auszuführen bzw. ausführen zu lassen.

So ... aber jetzt bin ich auch hier wieder mal raus - ich muß mich auch um meine eigenen Angelegenheiten kümmern. :sunglasses:

fda77, 2021-02-16:
Das mit Containern ist ja schön und gut, aber es müsste jemand anpassen. Bislang nutze ich nur VMs

Das mit universellen Binaries unter Linux schient sehr kompliziert zu sein... Ich bin da wohl von Windows etwas verwöhnt! Zuerst hab ich "Alpine" ausprobiert, aber da das auf Busybox setzt funktionieren fast keine Tools da die BB die Parameter nicht kennt. Man hätte zwar alles nachinstllieren oder notfalls selbst compiliren können, aber das war mir zu aufwändig. Dann hab ich mal "Void" ausprobiert, das funktioniert gar nicht. Nach der Installation kamen 5 Updates, weshalb es dann nicht mehr bootete -.-
Eigentlich hatte ich es hier schon aufgegen.
Doch dann hab ich mit mal die dl-toolchains (`dl/*r17470*`) angeschaut. Die Dateien darin sind gar nicht statisch! Dh wenn man die tools wie die toolchains baut wird es schon klappen. Über die neuen Toolchains hat sich bislang noch niemand beschwert und die sind per default an :)
Problematisch ist nur kconfig-host, wegen ncurses v5/v6

Kernel vorcompileren: Die meisten neueren Kernel sind nur mit der AVM-config hinterlegt. Dh wenn man Module in Freetz auswählt kommt nachher evtl ne Meldung dass man die Module noch im Kernel aktivieren muss. Das hat aber noch niemand gemacht und ich keine Lust dazu, das sind mir zu viele Kernel und das compilieren dauert schon recht lang (20min+). Daher macht das vorcompilieren von vor allem neuen Kernel keinen Sinn.

fda77, 2021-02-17:
dl-tools können getestet werden...

closed 2021-02-17 by fda77

fda77, 2021-02-28:
Hast du?

er13, 2021-03-07:
Es sei angemerkt, dass die Analyse von @PeterPawn aus https://github.com/Freetz-NG/freetz-ng/issues/197#issuecomment-778689994 aus meiner Sicht falsch ist.

`tar-host` setzt voraus, dass sowohl `tar` als auch der für `tar.xz` notwendige Entpacker auf dem Build-System vorhanden sind, wobei die Aufgabe, welcher Entpacker der richtige ist, beim Entpacken von `tar-host` dem `tar` des Build-Systems überlassen wird. Die Zeilen `$(if $(filter .xz .txz,$(3))` spielen beim Entpacken von `tar-host` überhaupt keine Rolle, da das Makro `UNPACK_TARBALL` für `tar-host` gar nicht verwendet wird.

Beim Entpacken von `busybox-host` wird hingegen `tar-host` von Freetz verwendet, jedoch immer noch der Entpacker des Build-Systems. Die Zeilen `$(if $(filter .bzip2 .bz2 .bz .tbz2 .tbz,$(3))` spielen beim Entpacken von `busybox-host` ebenso keine Rolle, ebenso aus dem Grund, dass das Makro `UNPACK_TARBALL` beim Entpacken von `busybox-host` gar nicht verwendet wird.

Erst mit dem nächsten `host-tool` bzw. `target-package` wird zu 100% auf die `host-tools` von Freetz gesetzt, i.e. es werden `tar-host` von Freetz und der entsprechende Entpacker (diese werden von `busybox-host` zur Verfügung gestellt) verwendet.

Es existiert gar keine zirkulare Abhängigkeit.

Überall, wo das Makro `UNPACK_TARBALL` verwendet wird, gehört auch die Abhängigkeit `UNPACK_TARBALL_PREREQUISITES` dazu. Oder andersrum die https://github.com/PeterPawn/YourFreetz/commit/ae03a7c1896748fb5bd98581a43dd36e736836d6 und https://github.com/PeterPawn/YourFreetz/commit/6566ba1f9c3525838dfb9ec1241ef2c5c27542de sind für sich alleine falsch und erst dann vielleicht richtig, wenn das Makro `UNPACK_TARBALL` mit angepasst wird (man beachte, dass im Makro für alle Entpacker explizit `$(TOOLS_DIR)/...` verwendet wird).

Das einzige, was fehlt, ist `bzip2/bunzip2` in `.build-prerequisites`(notwendig zum Entpacken von busybox-host).

fda77, 2021-03-07:
Ich hatte daran schlussendlich nichts verändert, da ich keinen Überblick davon hatte. Und stattdessen die download-tools erstellt. Diese sind .tar.xz, was in den prerequisites ist: https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites#L39
Bzip2 mach ich dann noch dazu

er13, 2021-03-07:
Sofern man möchte, dass `yourfritz-host` keine Abhängigkeiten zu `tar-host` und `busybox-host` hat und damit das einzige `host-tool` ist, welches vor dem Download der Firmware heruntergeladen und entpackt werden muss, muss man folgende 2 Änderungen in `tools/make/yourfritz-host/yourfritz-host.mk` vornehmen: die `UNPACK_TARBALL_PREREQUISITES`-Abhängigkeit entfernen und die `unpacked`-Regel so umschreiben, dass das Makro `UNPACK_TARBALL` nicht mehr verwendet wird (damit würde man analog dem `tar-host` auf tar/Entpacker des Build-Systems setzen).

In der make-Regel, in der juis verwendet wird, müsste man noch (sofern noch nicht der Fall) die `order only`-Abhängigkeit zu yourfritz-host hinzufügen.

```diff
--- a/tools/make/yourfritz-host/yourfritz-host.mk
+++ b/tools/make/yourfritz-host/yourfritz-host.mk
@@ -29,8 +29,8 @@ $(DL_DIR)/$(YOURFRITZ_HOST_SOURCE): | $(DL_DIR)
 	$(DL_TOOL) $(DL_DIR) $(YOURFRITZ_HOST_SOURCE) $(YOURFRITZ_HOST_SITE) $(YOURFRITZ_HOST_SOURCE_MD5)
 
 yourfritz-host-unpacked: $(YOURFRITZ_HOST_DIR)/.unpacked
-$(YOURFRITZ_HOST_DIR)/.unpacked: $(DL_DIR)/$(YOURFRITZ_HOST_SOURCE) | $(TOOLS_SOURCE_DIR) $(UNPACK_TARBALL_PREREQUISITES)
-	$(call UNPACK_TARBALL,$(DL_DIR)/$(YOURFRITZ_HOST_SOURCE),$(TOOLS_SOURCE_DIR))
+$(YOURFRITZ_HOST_DIR)/.unpacked: $(DL_DIR)/$(YOURFRITZ_HOST_SOURCE) | $(TOOLS_SOURCE_DIR)
+	tar -C $(TOOLS_SOURCE_DIR) $(VERBOSE) -xf $(DL_DIR)/$(YOURFRITZ_HOST_SOURCE)
 	$(call APPLY_PATCHES,$(YOURFRITZ_HOST_MAKE_DIR)/patches,$(YOURFRITZ_HOST_DIR))
 	@$(SED) -i -r -e '1 s,^($(_hash)$(_bang)[ \t]*/bin/)(sh),\1ba\2,' $(YOURFRITZ_HOST_BASH_AS_SHEBANG:%=$(YOURFRITZ_HOST_DIR)/%)
 	@$(SED) -i -r -e 's,([ \t])+$(_dollar),,' $(YOURFRITZ_HOST_STRIP_TRAILING_WHITESPACES:%=$(YOURFRITZ_HOST_DIR)/%)
```

PeterPawn, 2021-03-07:
> die Analyse von @PeterPawn [...] aus meiner Sicht falsch ist.

Dann hoffe ich meinerseits mal, daß Du das nur falsch verstanden hast und ich es nicht wirklich falsch be-/geschrieben habe.

> Die Zeilen $(if $(filter .xz .txz,$(3)) spielen beim Entpacken von tar-host überhaupt keine Rolle, da das Makro UNPACK_TARBALL für tar-host gar nicht verwendet wird.

Wo genau hätte ich das behauptet? Ich schrieb sogar explizit:

> und daher bräuchte es wieder ein unxz, wenn man das auch mit den Freetz-Helper-Funktionen entpacken wollte: 

, wo die Verwendung des Konjunktiv II (bräuchte ... wenn man ... wollte) ja klar aufzeigen sollte, daß das in Freetz derzeit nicht der Fall ist. Und auch meine Fortsetzung:

> so wird für dieses tar-host-Paket beim Entpacken gar nicht auf diese Helper-Funktionen zurückgegriffen, sondern das wird direkt mit einem tar-Aufruf entpackt, wobei anhand der Erweiterung jetzt vom tar das xz als Filter für das Dekomprimieren vor dem Entpacken aufgerufen wird

sagt - zumindest nach meinem Sprachverständnis - genau dasselbe aus, was Du mit:

> wobei die Aufgabe, welcher Entpacker der richtige ist, beim Entpacken von tar-host dem tar des Build-Systems überlassen wird. Die Zeilen $(if $(filter .xz .txz,$(3)) spielen beim Entpacken von tar-host überhaupt keine Rolle, da das Makro UNPACK_TARBALL für tar-host gar nicht verwendet wird.

hinsichtlich des "Auswahlmechanismus" für den korrekten Dekompressor und zur Frage, ob/warum die Filter im Makro für `tar-host` keine Rolle spielen, zum Ausdruck bringst.

Auch das

> Es existiert gar keine zirkulare Abhängigkeit.

kann ich eigentlich nur unterstreichen, denn es ist auch hier wieder so, daß ich in meiner Aussage den Konjunktiv verwendete und schrieb, wie das WÄRE (die verwendete Form ist "würde", abgeleitet von "sein" - Konjunktiv II/Futur I), wenn man es nicht so machte.

> Das würde zu einer zirkulären Abhängigkeit führen und so wird für dieses tar-host-Paket beim Entpacken gar nicht auf diese Helper-Funktionen zurückgegriffen

---

> Überall, wo das Makro UNPACK_TARBALL verwendet wird, gehört auch die Abhängigkeit UNPACK_TARBALL_PREREQUISITES dazu. Oder andersrum die PeterPawn@ae03a7c und PeterPawn@6566ba1 sind für sich alleine falsch und erst dann vielleicht richtig, wenn das Makro UNPACK_TARBALL mit angepasst wird (man beachte, dass im Makro für alle Entpacker explizit $(TOOLS_DIR)/... verwendet wird).

Das ist nun aber wirklich absolut unredlich ... diese beiden Commits gehören zu einem gesonderten Branch: https://github.com/PeterPawn/YourFreetz/compare/master...PeterPawn:host-tools und daß man da alle zusammen gehörenden Patches auch in ihrer Gesamtheit betrachtet, gehört für mich zur intellektuellen Redlichkeit. Wenn man - wie ich es in dem Branch gemacht habe - die Programme zum Dekomprimieren zur "Voraussetzung" erklärt, braucht kein Schwein die Abhängigkeit von `UNPACK_TARBALL_PREREQUISITES` - schließlich geht es da nur um ein `tar` und die ganzen (De-)Kompressoren. Wenn die auf dem Host bereits existieren, wie blöd muß man sein, um dann unbedingt noch die eigene Version davon bauen zu wollen. So verkorkst KANN einfach niemand sein.

---

Und daß der Dekompressor für die von Freetz verwendeten BusyBox-Quellen (für den Host) gar nicht zu den "offiziellen" Voraussetzungen gehört (was ich hier aber nicht noch einmal erwähnt habe, sondern nur in meinem Branch `host_tools` korrigierte: https://github.com/PeterPawn/YourFreetz/commit/4be3649497f294e57bb573b00935924a7e22052a#diff-d5aa28be9b4248f04591f47d6c22b7d0cb7c99b46106c406581c84e2aead975dR28), hast Du ja jetzt offenbar selbst festgestellt.

---

Sogar meine These, daß man in Freetz all die Pakete, die tatsächlich mittels LZMA-komprimierte TAR-Files als Quelle benutzen, auch problemlos ohne die Abhängigkeit von `UNPACK_TARBALL_PREREQUISITES` definieren könnte, wenn man zuvor die Verwendung von `unxz` im `UNPACK_TARBALL`-Makro durch ein `xz -d` ersetzen würde (und damit das `xz` die Option nicht kennt, müßten schon sehr viele Faktoren zusammenkommen, denn auf den allermeisten Systemen ist `unxz` auch nichts weiter als ein Symlink auf `xz`), was ja bereits (auch in Freetz) zu den Voraussetzungen zählt, steht in meiner "Analyse" erst nach einem (ausführlichen) Abschnitt:

> Bei Gelegenheit kann man dann immer noch hingehen und die Helper-Funktion in include/make/020-helper-functions.mk für die xz-Erweiterung mal auf den Aufruf von xz -d -c $(1) umstellen (und das auch [...]

(und wie man sehen kann, sogar mit dem Hinweis, daß dann die Pfadangabe über `$(TOOLS_DIR)` natürlich weg muß), in dem ich die Voraussetzungen für den Wegfall dieser Abhängigkeit aufzähle und erkläre.

Ich frage mich tatsächlich, wie man das (und Du (@er13) meintest ja nach Deinem eigenen Link auch genau den Beitrag, aus dem ich mich hier selbst zitiert habe) sooo dermaßen falsch verstehen könnte - ich bleibe jedenfalls bei meinen Aussagen und sehe mich letztlich sogar noch dadurch bestärkt, daß Du in Deiner "Kritik" all das als Begründung anführst, was ich - wenn man richtig liest - selbst geschrieben habe ... mit Ausnahme Deiner Ansichten zu `UNPACK_TARBALL_PREREQUISITES`.

Wobei ich da dann weiterhin auch auf eine stichhaltige(!) Begründung gespannt wäre, warum diese beiden Absätze in meinem Beitrag oben: 

> Bei Gelegenheit kann man dann immer noch hingehen und die Helper-Funktion in include/make/020-helper-functions.mk für die xz-Erweiterung mal auf den Aufruf von xz -d -c $(1) umstellen (und das auch [...]
[... cut ...]
kann man das in meinen Augen kaum schreiben ... das "Problem", das es hier zu lösen galt, war ja die Frage, wie man gleich am Beginn des Build-Prozesses eine JUIS-Abfrage nach der neuesten Version der AVM-Firmware machen kann, ohne zuvor erst die ganzen Tools zu bauen, was bei einer Abhängigkeit von `UNPACK_TARBALL_PREREQUISITES` nun mal der Fall wäre in einem "frischen" Checkout. 

Die Tatsache, daß Du das Ergebnis des Auscheckens eines Git-Stands bei GitHub am Ende mit LZMA komprimierst, eröffnet(e) exakt diese Möglichkeit, das `yourfritz-host`-Paket in `tools` genauso zu behandeln, wie das `tar-host` - beide brauchen zum Entpacken dieselben Programme. Punkt.

fda77, 2021-03-07:
Ich denke das ist so gewachsen, weil anfangs xz nicht so verbreitet war und nicht überall vorinstalliert. Man hat ja an den "Fehlermeldungen" der letzten Wochen gesehen wie schwer es für viele ist mit apt-get ein Programm nachzuinstallieren

@er13 
>  precompiled/downloadable tools-host für sowas einzuführen, finde ich ein Overkill.

Die dl(host)tools sind nicht nur ausschliesslich hierfür, sondern ersparen das compilierne (fast: https://github.com/Freetz-NG/freetz-ng/blob/master/tools/make/Makefile.in#L43) aller tools: https://github.com/Freetz-NG/freetz-ng/blob/master/Makefile#L361
Durch Umstllung auf 64bit konnte pseudo und fakeroot auch dazu, und es sind bei mit ca 6 Minuten Zeitersparnis

PeterPawn, 2021-03-07:
Ich vermute fast, ich komme jetzt erst dahinter, was Du eventuell meinen könntest mit den zwei verlinkten Commits aus meinem Branch ... mal vollkommen abgesehen davon, daß die hier (und in https://github.com/Freetz-NG/freetz-ng/issues/197#issuecomment-778689994) gar nicht das Thema waren und ich immer noch nicht verstehe, was die im Kontext dieses Problems hier (und meiner hier abgegebenen Beschreibung dessen, was man machen könnte für eine Lösung) überhaupt zu suchen haben.

Wenn Du in dem anderen Branch darauf hinaus willst, daß da in den zwei Makefiles auch nach meinem Commit weiterhin `UNPACK_TARBALL` verwendet wird - DAS ist tatsächlich richtig, nur stört es im Kontext dessen, warum ich den Branch überhaupt erstellt habe (und das steht nun mal in den entsprechenden Commit-Kommentaren) gar keine Rolle. Denn da ging es ausschließlich darum, daß man die Tools für die beiden SquashFS-Versionen auch aus einem "frischen Checkout" bauen kann (so steht es im Kommentar zum Branch) und da fehlt tatsächlich noch ein Patch, der das Include-File für UNPACK_TARBALL auch ändert.

Was aber bei mir bisher nicht erforderlich war/ist, weil ich - nach der Änderung der "prerequisites" - noch ein weiteres Skript eingebaut habe (nur arbeitet das halt mit `zypper`, weil ich OpenSuSE Tumbleweed verwende und taugt nicht für Ubuntu-Builds, daher ist das auch nicht im Branch), das mir erst einmal anhand der `build-prerequisites` die Symlinks zu den Tools auf dem Build-Host einträgt (bzw. die ggf. auch erst aus den Repos der Distribution nachinstalliert, falls ich mal wieder mit einem Image in einer VM arbeite, die sich beim Neustart automatisch zurückstellt), bevor ich überhaupt mit dem `make` beginne. 

Daß es bei mir vor diesem `make` ohnehin noch jede Menge weiterer Schritte gibt (ggf. bis hin zum Bereinigen einer vorhandenen Konfigurationsdatei bzw. zur Auswahl einer passenden für den geplanten Build), ist Dir vielleicht noch aus früheren Diskussionen in Erinnerung ... und es hat auch Gründe, warum ich den Branch nirgends als "allgemeinen Push-Request" "eingereicht" habe.

Aber der gesamte Branch hat eben - um das noch einmal zu betonen - mit dem Problem hier überhaupt nichts zu tun ... wenn der hier angesprochen wurde, dann nur als BEISPIEL für einen Weg, wie man das Problem mit dem endlosen Bauen der BusyBox, bevor man irgendeines der Programme aus `tools` benutzen kann, umgehen könnte - was man auch in dem Thread nachlesen kann, der diesem hier vorausging und ganz oben verlinkt ist.

Also wirf das bitte nicht alles durcheinander ... der Branch `host_tools` ist speziell für meine Bedürfnisse (und meine Umgebung) gedacht und ermöglicht es, die (für mich wichtigen) SquashFS-Tools (die ich dann in `yf_bin` bereitstelle für die Benutzung durch andere) mit der Freetz-"Toolchain" zu erstellen, ohne erst das ganze Geraffel mit `tar` und `busybox` erleiden zu müssen. 

Auch den Gedanken, daß dieser Branch die Voraussetzungen dafür schaffen soll, die Tools schneller in einem Container zu bauen (wo dann die Installation der passenden "prerequisites" genauso zum Prozess gehört, wie das Klonen des entsprechenden Repos), findet man beim Lesen direkt in den Commit-Kommentaren (https://github.com/Freetz/freetz/commit/4be3649497f294e57bb573b00935924a7e22052a):

> host-tools: change prerequisites for easier/smaller/faster builds from containers

und auch da wird beim Vorbereiten der Umgebung im Container dann dasselbe Skript benutzt zum "Zusammensuchen" der Tools (nur mit jeweils zur Distro passenden Programmen zur Paketverwaltung) und zum Verlinken dieser Programme im `tools`-Verzeichnis. 

Denn das sind dann tatsächlich (auch abseits des Makros `UNPACK_TARBALL`) zu viele Stellen in Freetz, wo auf dieses Verzeichnis zugegriffen wird, als das man das (sinnvoll/zeitnah) ändern könnte ... daher stammt - nebenbei bemerkt - auch die Idee (weiter oben hier auch erwähnt), dort anstelle der Tools erst mal nur ein Wrapper-Skript zu verlinken, was diese dann erst "on demand" baut oder aus der Paketverwaltung der Distribution nachinstalliert.

Aber eben tatsächlich erst dann, wenn sie auch gebraucht werden, weil selbst in einem (stateless) Container eigentlich nicht alle "Voraussetzungen" installiert sein müssen (ich weiß nicht mal genau, wo man ein `lunzip` finden kann für Tumbleweed) und die meisten davon gar nicht benötigt werden. Um ein `dropbear` zu bauen, brauche ich das Quellpaket, `tar`, `bunzip2`, eine Toolchain und eine passende Konfigurationsdatei ... mehr nicht. Alles andere wäre für diese Aufgabe nur unnützer Ballast und kostet nicht nur (kostbare) Rechenzeit auf einer Cloud-Instanz, sondern auch noch Datenverkehr und (zumindest vorübergehend) Speicherplatz für den Container. Die Arbeitsweise/das Ziel ist für `YourFreetz` eben ein vollkommen anderes ... auch deshalb gibt es keine PRs für Freetz aus den allermeisten Änderungen.

er13, 2021-03-08:
Ich ziehe meine Aussage, dass https://github.com/Freetz-NG/freetz-ng/issues/197#issuecomment-778689994 falsch ist, zurück. Ich habe es missverstanden - es hat sich für mich so gelesen, dass da Freetz beschrieben wird, wie es ist, in Wirklichkeit wird da beschrieben, was es nicht ist (der Konjunktiv eben). Solltest Du, @PeterPawn, Dich dadurch angegriffen gefüllt haben, so tut es mir leid.

PeterPawn, 2021-03-08:
@er13:
Nein, nicht angegriffen - ich hoffe auch, es liest sich nicht wirklich so. Ich war nur verwundert, weil selbst der Patch hier: https://github.com/Freetz-NG/freetz-ng/issues/197#issuecomment-792348260 ja genau das ist, was ich weiter oben - halt als "Textaufgabe" - beschrieben hatte. Also alles gut ...