Freetz-NG/freetz-ng #201: Wenn Dnsmasq läuft beendet sich AVM-multid Fritzbox 7590 7.2x

PhoenixG36, 2021-02-15:
Meine 7590 mit 7.21 Firmware läuft nicht mehr stabile wenn ich Dnsmasq als DNS Server nutze.
Es sieht so aus als ob dann der AVM multid beendet wird. Man kann ihn starten, aber er beendet sich gleich wieder.
Lässt man die Box so stehen, scheint sie nach kurzer Zeit zu  booten.
Daher habe ich sie aktuell nicht mehr als primere BOX.
Was wird noch benötigt um den Fehler einzugrenzen?

[FRITZ.Box_7590_(UI)_154.07.21-freetz-ng-17901-965b41254_2021-02-15_1801_support.tgz.zip](https://github.com/Freetz-NG/freetz-ng/files/5983100/FRITZ.Box_7590_.UI._154.07.21-freetz-ng-17901-965b41254_2021-02-15_1801_support.tgz.zip)

fischli, 2021-02-15:
multid übernimmt für normal alle DNS-Anfragen. Wenn ein anderer Daemon die Anfragen übernehmen soll kommt es zum Konfilkt auf UDP/53. Deswegen stirbt der sofort wieder weg weil der Socket schon in Verwendung ist. Kann man multid so konfigurieren das er kein DNS anbietet?

PhoenixG36, 2021-02-15:
Auf meiner 3490 laufen beide so mit der gleichen Konfiguration:
[image]

fischli, 2021-02-15:
Dann aber auf unterschiedlichen Ports und/oder Interfaces. Mehrere Prozesse auf dem gleichen Socket geht nicht. Was sagt "netstat -naptu"?

PhoenixG36, 2021-02-15:
das sind die Ausgaben:
root@fritz:/var/mod/root# netstat -naptu | grep multid
tcp        0      0 :::50053                :::*                    LISTEN      3446/multid
udp        0      0 0.0.0.0:50067           0.0.0.0:*                           3446/multid
udp        0      0 :::51981                :::*                                3446/multid
udp        0      0 :::37650                :::*                                3446/multid
udp        0      0 :::45620                :::*                                3446/multid
udp        0      0 :::55353                :::*                                3446/multid
udp        0      0 :::55353                :::*                                3446/multid
udp        0      0 :::55355                :::*                                3446/multid
udp        0      0 :::55355                :::*                                3446/multid
udp        0      0 :::39741                :::*                                3446/multid
udp        0      0 :::32832                :::*                                3446/multid
udp        0      0 :::49482                :::*                                3446/multid
udp        0      0 :::37453                :::*                                3446/multid
udp        0      0 :::36175                :::*                                3446/multid
udp        0      0 :::52096                :::*                                3446/multid
udp        0      0 :::50053                :::*                                3446/multid
udp        0      0 :::38828                :::*                                3446/multid
udp        0      0 :::57265                :::*                                3446/multid
udp        0      0 :::48306                :::*                                3446/multid
udp        0      0 :::46260                :::*                                3446/multid
udp        0      0 :::37312                :::*                                3446/multid
udp        0      0 :::51440                :::*                                3446/multid
udp        0      0 :::38391                :::*                                3446/multid
udp        0      0 :::37371                :::*                                3446/multid
root@fritz:/var/mod/root# netstat -naptu | grep dnsmasq
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      3359/dnsmasq
tcp        0      0 :::53                   :::*                    LISTEN      3359/dnsmasq
udp        0      0 0.0.0.0:53              0.0.0.0:*                           3359/dnsmasq
udp        0      0 :::53                   :::*                                3359/dnsmasq

fischli, 2021-02-15:
multid läuft auf 50053 und dnsmasq auf 53. Somit kein Konflikt.

fda77, 2021-02-15:
Auf meiner 7590 funkionier dnsmasq ohne Problem
`2021-02-15 18:00:38(1) [Segmentation fault] multid([6746]) CRASHED at dnsd_dot_set_enabled (/sbin/multid at 0003f088) accessing 00002f00`
Ich vermute eine Fehler von AVM. Es kann sein dass die Konfiguration der Fritzbox ducheinander gekommen ist.
Sicher mal deine Konfiguration und bei AVM einen Werksresett. Fall es hilft kannst du vermutlich die Freetz-Einstellungen zurückspielen, aber die AVM Sachen muss man neu eingeben (oder ein altes Backup benutzen)

closed 2021-02-17 by fda77