Freetz-NG/freetz-ng #202: Haproxy: can't load library 'libpcreposix.so.0'

berndy2001, 2021-02-16:
Hab heute mal meine 7590 aktualisiert und seither startet Haproxy nicht mehr. Das Freetz-Webif hat keinen Protokoll-Eintrag dazu, aber in der Comandline gibts mehr Infos dazu:
`root@fritz:/var/mod/root# haproxy`
`/usr/bin/haproxy: can't load library 'libpcreposix.so.0'`

Hängt vermutlich mit #188 zusammen. Naheliegend wäre, dass FREETZ_LIB_libpcreposix ausgewählt werden muss, das werde ich noch testen.

fda77, 2021-02-16:
Vielleicht wird in der .mk auch noch ein DEPENDS_ON benötigt damit pcre vor haproxy gebaut wird? Jenachdem wie gross die lib ist könnten du und @flosch-dev euch noch überlegen ob jeder regex braucht oder man eine Option im menuconfig macht

berndy2001, 2021-02-16:
FREETZ_LIB_libpcreposix hat geholfen, haproxy läuft nun wieder. Größe ist zwar nur 6164b, aber analog zu anderen Paketen wäre _Add/Select/Enable/with/build with regex support_ als Option im menuconfig meiner Meinung die schönere Umsetzung.
Ich für mich habe aktuell keinen Usecase für regex-Support.

flosch-dev, 2021-02-17:
optional macht denke ich Sinn, ich versuche das mal einzubauen.
Bei mir reicht allerdings "FREETZ_LIB_libpcre".  FREETZ_LIB_libpcreposix habe ich nicht aktiv und dennoch geht bei mir haproxy mit pcre support. Ich nutze 7490 mit 7.21.

fda77, 2021-02-17:
Es kann sein dass es die libpcreposix bei @berndy2001 schon gab. Evtl durch ein DEPENS_ON eines anderen Packages oder weil die mal ausgewählt war. Das configure von haproxy findet und benutzt sie. Daher ist es besser alles von "configure --help" zu deaktivieren (in der .mk) und nur was benötigt wird zu aktivieren - und die benötigten Libs mit "select" auf die Box zu bringen.
Die uclibc Version von Freetz ist bei der aktuellen 7490 und 7590 gleich.

fda77, 2021-02-17:
"pcreposix" muss man vermutlich auch auswählen. Schau doch mal auf deiner Fritzbox mit "ldd" welche Libs genutzt wetden

Aus der Makefile

```
ifeq ($(USE_STATIC_PCRE),)
# dynamic PCRE
OPTIONS_CFLAGS  += -DUSE_PCRE $(if $(PCRE_INC),-I$(PCRE_INC))
OPTIONS_LDFLAGS += $(if $(PCRE_LIB),-L$(PCRE_LIB)) -lpcreposix -lpcre
else
# static PCRE
OPTIONS_CFLAGS  += -DUSE_PCRE $(if $(PCRE_INC),-I$(PCRE_INC))
OPTIONS_LDFLAGS += $(if $(PCRE_LIB),-L$(PCRE_LIB)) -Wl,-Bstatic -lpcreposix -lpcre -Wl,-Bdynamic
endif
endif
```

fda77, 2021-02-18:
Ach, die pcreposix kommt auch aus dem pcre Package ^^
 - DEPENDS_ON braucht man damit die lib vor dem Package compiliert wird und benutzbar ist
 - REBUILD_SUBOPTS damit bei einer Änderung der Option im menuconfig das Package neu gebaut wird

berndy2001, 2021-02-18:
Hab jetzt mal per `git pull` aktualisiert und  "with regex support (PCRE)" ausgewählt. 
Haproxy läuft bei mir nicht: `/usr/bin/haproxy: can't load library 'libpcreposix.so.0'`
FREETZ_LIB_libpcreposix müsste auch selektiert werden.

fda77, 2021-02-18:
Ich hab haproxy mit pcre gebaut:
```
toolchain/build/arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.1/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-ldd packages/target-arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.1/haproxy-2.2.9/root/usr/bin/haproxy
        libpcreposix.so.0 => not found (0x00000000)
        libpcre.so.1 => not found (0x00000000)
        libc.so.0 => not found (0x00000000)
```
Es werden also beide benötigt.
Gebaut werden diese von pcre: https://github.com/Freetz-NG/freetz-ng/blob/master/make/libs/pcre/pcre.mk
Aber man muss auch beide auswählen damit sie auf die Box kommen: https://github.com/Freetz-NG/freetz-ng/blob/master/make/libs/pcre/Config.in#L7

flosch-dev, 2021-02-19:
so ganz verstehe ich das nicht, bei mir geht es ohne:

```
./toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-ldd packages/target-mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/haproxy-2.2.9/root/usr/bin/haproxy
checking sub-depends for 'not found'
checking sub-depends for 'not found'
checking sub-depends for 'not found'
checking sub-depends for 'not found'
        libssl.so.1.1 => not found (0x00000000)
        libcrypto.so.1.1 => not found (0x00000000)
        libpcre.so.1 => not found (0x00000000)
        libc.so.0 => not found (0x00000000)
        /usr/lib/freetz/ld-uClibc.so.1 => /usr/lib/freetz/ld-uClibc.so.1 (0x00000000)
```
auf der FB:
```
ldd /usr/bin/haproxy
        libssl.so.1.1 => /usr/lib/freetz/libssl.so.1.1 (0x77142000)
        libcrypto.so.1.1 => /usr/lib/freetz/libcrypto.so.1.1 (0x76f7c000)
        libpcre.so.1 => /usr/lib/freetz/libpcre.so.1 (0x76f2f000)
        libc.so.0 => /usr/lib/freetz/libc.so.0 (0x76e58000)
        libz.so.1 => /usr/lib/freetz/libz.so.1 (0x76e34000)
        ld-uClibc.so.1 => /usr/lib/freetz/ld-uClibc.so.1 (0x771b6000)

# haproxy -vv
HA-Proxy version 2.2.9-a947cc2 2021/02/06 - https://haproxy.org/
Status: long-term supported branch - will stop receiving fixes around Q2 2025.
Known bugs: http://www.haproxy.org/bugs/bugs-2.2.9.html
Running on: Linux 3.10.107 #1 SMP Mon Aug 17 17:21:20 CEST 2020 mips
Build options :
  TARGET  = custom
  CPU     = generic
  CC      = /home/build/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc
  CFLAGS  = -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -ffunction-sections -fdata-sections
  OPTIONS = USE_EPOLL=1 USE_PCRE=1 USE_OPENSSL=1
  DEBUG   =

Feature list : +EPOLL -KQUEUE -NETFILTER +PCRE -PCRE_JIT -PCRE2 -PCRE2_JIT +POLL -PRIVATE_CACHE -THREAD -PTHREAD_PSHARED -BACKTRACE -STATIC_PCRE -STATIC_PCRE2 -TPROXY -LINUX_TPROXY -LINUX_SPLICE -LIBCRYPT -CRYPT_H -GETADDRINFO +OPENSSL -LUA -FUTEX -ACCEPT4 -CLOSEFROM -ZLIB -SLZ -CPU_AFFINITY -TFO -NS -DL -RT -DEVICEATLAS -51DEGREES -WURFL -SYSTEMD -OBSOLETE_LINKER -PRCTL -THREAD_DUMP -EVPORTS

Default settings :
  bufsize = 16384, maxrewrite = 1024, maxpollevents = 200

Built with OpenSSL version : OpenSSL 1.1.1j  16 Feb 2021
Running on OpenSSL version : OpenSSL 1.1.1j  16 Feb 2021
OpenSSL library supports TLS extensions : yes
OpenSSL library supports SNI : yes
OpenSSL library supports : TLSv1.0 TLSv1.1 TLSv1.2 TLSv1.3
Built without compression support (neither USE_ZLIB nor USE_SLZ are set).
Compression algorithms supported : identity("identity")
Built with transparent proxy support using: IP_TRANSPARENT IP_FREEBIND
Built without multi-threading support (USE_THREAD not set).
Built with PCRE version : 8.43 2019-02-23
Running on PCRE version : 8.43 2019-02-23
PCRE library supports JIT : no (USE_PCRE_JIT not set)
Encrypted password support via crypt(3): no
Built with gcc compiler version 5.5.0

Available polling systems :
      epoll : pref=300,  test result OK
       poll : pref=200,  test result OK
     select : pref=150,  test result OK
Total: 3 (3 usable), will use epoll.

Available multiplexer protocols :
(protocols marked as <default> cannot be specified using 'proto' keyword)
            fcgi : mode=HTTP       side=BE        mux=FCGI
       <default> : mode=HTTP       side=FE|BE     mux=H1
              h2 : mode=HTTP       side=FE|BE     mux=H2
       <default> : mode=TCP        side=FE|BE     mux=PASS

Available services : none

Available filters :
        [SPOE] spoe
        [COMP] compression
        [TRACE] trace
        [CACHE] cache
        [FCGI] fcgi-app

```

fda77, 2021-02-19:
Wenn ich für 7590 7.1x statt wie oben 7.2x baue wird libpcreposix nicht gelinkt

arm_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-4.1 braucht libpcreposix 
mips_gcc-5.5.0_uClibc-1.0.36-nptl_kernel-3.10 nur libpcre
mips_gcc-5.5.0_uClibc-1.0.14-nptl_kernel-3.10 nur libpcre

fda77, 2021-02-21:
https://github.com/Freetz-NG/freetz-ng/commit/a174b886088434376263350fe9a19ad050ae3d32

closed 2021-02-21 by fda77