Freetz-NG/freetz-ng #203: libfakeroot Error

boba23, 2021-02-17:
Hi zusammen,

hab heute den aktuellen -NG ausgecheckt und wollte ein aktuelles Image bauen.  Bekomme jetzt massenweise folgende fakeroot fehler:

ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.

Die waren vor 1-2 Wochen bei der erfolgreich gebauten 17823 nicht da und ich konnte das Image problemlos kompilieren ....

Scheint wohl ein i386/AMD64 Thema zu sein .... kann mir hier wer helfen? Möglicher Workaround? 
Hab im Changelog gesehen, dass die Fakeroot Version aktualisiert wurde ...

Viele Grüße,

boba

fda77, 2021-02-17:
Welche CPU ist das denn? Deaktiver im menuconfig `FREETZ_HOSTTOOLS_DOWNLOAD` und führ dann `make tools-distclean` aus. Die Warnung aus https://github.com/Freetz-NG/freetz-ng/commit/fd19c8df00fa47c5ac8ebb5fdb1297b54c129759 funktioniert nicht

boba23, 2021-02-17:
Ok, versuch ich gleich ma... Is ne Intel i7-7xxx.

berndy2001, 2021-02-17:
Hatte heute was ähnliches mit libz.so; baue gerade eigene toolchain.
Du kannst du es unter Toolchain options - Toolchains - Build own toolschain auswählen.

boba23, 2021-02-17:
Build own toolchain hatte ich eh, grad gecheckt. FREETZ_HOSTTOOLS_DOWNLOAD war bei mir aber aktiviert, hab ich jetzt ma raus und lass gerade ein Image erstellen ... ma gespannt obs durch läuft.

berndy2001, 2021-02-17:
Verstehe ich nicht ganz, das ist doch entweder oder, also bei mir jedenfalls kann ich nur einen der beiden Punkte auswählen.

berndy2001, 2021-02-17:
Mit eigener toolchain bei mir:
```
user@hs:~/freetz-ng$ make
mkdir -p source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/juis_check-049839bdf5811ed0c7d9b3036249c12a294ecffe
tar cf - -C tools/yf/ juis/ | tar xf - -C source/target-mips_gcc-8.3.0_uClibc-1.0.36-nptl_kernel-4.9/juis_check-049839bdf5811ed0c7d9b3036249c12a294ecffe/
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.

STEP 1: UNPACK
source firmware file: dl/fw/FRITZ.Box_7590-07.21.image
checking signature: ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
valid
unpacking firmware image
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
unpacking filesystem image
ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
    ERROR: ld.so: object 'libfakeroot.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
    /home/user/freetz-ng/tools/unsquashfs4-avm-be: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory
ERROR: modunsqfs: Error in build/original/firmware/var/tmp/filesystem.image
make: *** [Makefile:284: firmware-nocompile] Fehler 1
```

boba23, 2021-02-17:
Nee, sind zwei völlig verschiedene Punkte ... der zweite ist nicht unter Toolchain options, sondern unter:

Build System Options -> Use precompiled i686 host tools

Übrigens, nachdem ich das aktiviert habe ist mein build gerade wieder fehlerfrei druchgelaufen. In diesem Moment fertig.

boba23, 2021-02-17:
Sorry ich meinte "deaktiviert" .... war vorher und wohl die ganze Zeit bei mir enabled.

berndy2001, 2021-02-17:
Mea culpa, das habe ich verwechselt.

boba23, 2021-02-17:
Das scheints auf jeden Fall gewesen zu sein, zumindest in meinem Fall.

fda77, 2021-02-17:
Es gibt 2 Optionen: Vorcompilierte "Tools" und "Toolchain". Tools ist neu! Die Dateien ein beiden sind "gleich" auf einem uralten Ubuntu x86 compiliert.  Der Vorteil davon ist dass man das nicht selbst compilieren muss und Zeit spart, sonst nichts. Allerdings müssen diese Binaries zum System passen...
Deshalb die Frage ob Toolchains bisher liefen und der Fehler nur in Tools liegt, oder ob keines von beiden funktioniert.
Es könnte sein dass irgendwelche 32-bit libraries nicht installiert sind. Bei Ubuntu gab es mal das Paket "ia32-libs" aber das wurde wohl irgendwie umbenannt: https://askubuntu.com/questions/107230/what-happened-to-the-ia32-libs-package

boba23, 2021-02-17:
Siehe Anhang .....
[ldd_tools.txt](https://github.com/Freetz-NG/freetz-ng/files/5998257/ldd_tools.txt)

boba23, 2021-02-17:
Ist aber jetzt dann Stand nach deaktivierung von FREETZ_HOSTTOOLS_DOWNLOAD

berndy2001, 2021-02-17:
Ohne FREETZ_HOSTTOOLS_DOWNLOAD lief der Imagebau nun auch durch.

ldd mit aktiviertem FREETZ_HOSTTOOLS_DOWNLOAD
[ldd.txt](https://github.com/Freetz-NG/freetz-ng/files/5998917/ldd.txt)

fda77, 2021-02-17:
Das solltehelfen:
```
sudo apt-get install lib32z1-dev
```

Zum testen die Option im menuconfig wieder aktivieren, dann
```
rm -rf build
make tools-distclean
make
```

fda77, 2021-02-17:
Auch https://github.com/Freetz-NG/freetz-ng/commit/b1907d74b0cb1229f972460ca971eb8b9847d6ae und https://github.com/Freetz-NG/freetz-ng/blob/master/docs/wiki/10_Beginner/install.de.md#installation-der-ben%C3%B6tigten-pakete-ubuntu

closed 2021-02-17 by fda77

berndy2001, 2021-02-18:
Danke, jetzt läufts durch. Hatte nicht daran gedacht, dass zlib ein benötigtes Paket ist bzw. wurde.
Macht es Sinn das in .build-prerequisites einzubauen?

fda77, 2021-02-18:
Ich hoffe die Erkennung funktioniert überall

fda77, 2021-02-18:
Das sollte das Problem beheben ^^