Freetz-NG/freetz-ng #210: make menuconfig meckert das execstack fehlt.

Conan179, 2021-02-26:


freetz-ng-mod, 2021-02-27:
sudo apt -y install libzstd-dev:i386 libzstd-dev cmake lib32z1-dev unar inkscape imagemagick subversion git bc wget sudo ccache gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar perl libstring-crc32-perl ruby gawk python libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib lib32stdc++6 libglib2.0-dev

fda77, 2021-02-27:
Ist aktualisiert: https://github.com/Freetz-NG/freetz-ng/commit/fdbd1c2f084bf5b25c8d63edbe4e62105f67efc4

fda77, 2021-02-27:
https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES

closed 2021-02-27 by fda77

Conan179, 2021-02-27:
@fda77 
```
diz@vm:/$ sudo apt-get update
Holen:1 http://security.ubuntu.com/ubuntu focal-security InRelease [109 kB]
OK:2 http://archive.ubuntu.com/ubuntu focal InRelease
Holen:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Holen:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]
Holen:5 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [12,7 kB]
Holen:6 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [746 kB]
Holen:7 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 c-n-f Metadata [16,2 kB]
Es wurden 1.099 kB in 4 s geholt (254 kB/s).
Paketlisten werden gelesen... Fertig
diz@vm:/$ sudo apt-get -y install sqlite3:i386
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
E: Paket sqlite3:i386 kann nicht gefunden werden.
diz@vm:/$
```
und
```
diz@vm:/$ sudo apt -y install execstack sqlite3 libsqlite3-dev libzstd-dev:i386 libzstd-dev cmake lib32z1-dev unar inkscape imagemagick subversion git bc wget sudo ccache gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar perl libstring-crc32-perl ruby gawk python libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib lib32stdc++6 libglib2.0-dev
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Hinweis: »python-is-python2« wird an Stelle von »python« gewählt.
Hinweis: »lib32ncurses-dev« wird an Stelle von »lib32ncurses5-dev« gewählt.
E: Paket libzstd-dev:i386 kann nicht gefunden werden.
diz@vm:/$
```

freetz-ng-mod, 2021-02-27:
So eine endliche Fehlermeldung hatte ich vor 2 tage auch. Nach denn ich linux (ubuntu) neu installiert habe und das was freetz will per sudo apt -y install geht allles. 

Dabei ist wichtig erst das was Freetz will und danach das was du mit anderen Programme brauchst   zu installieren

Conan179, 2021-02-27:
> Dabei ist wichtig erst das was Freetz will

Dann ist gut, das es um diese freetz-ng Anleitung geht und nicht um irgendein anderes programm
https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md

freetz-ng-mod, 2021-02-27:
poste mal deine config

Conan179, 2021-02-27:
@freetz-ng-mod bitte halte ich aus meinen Problemen hier raus, da du keinen ahnung hast, um was hier geht.

freetz-ng-mod, 2021-02-27:
gut das dein Prob ja behoben wurde.

@Conan179 wie du sagst ich weiß ja nixht mal um was es geht "Ironie aus"

PrinzVonBillAir, 2021-02-27:
> @fda77
>
> ```
> diz@vm:/$ sudo apt-get update
> Holen:1 http://security.ubuntu.com/ubuntu focal-security InRelease [109 kB]
> OK:2 http://archive.ubuntu.com/ubuntu focal InRelease
> Holen:3 [...]

Exakt vor dem gleichen Probleme stehe ich auch gerade. Ich habe aus der VM das komplette Freetz-Image gelöscht, Oracle VM geupdated auf die aktuellste Version, das neuste Freetz-Image aus dem IPFF eingespielt (mit Ubuntu 20.10 (GNU/Linux 5.8.0-26-generic x86_64), habe, wie in https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md angemerkt ZUERST sudo apt-get -y update
sudo apt-get -y upgrade
sudo apt-get -y dist-upgrade
ausgeführt und anschließend

sudo apt -y install execstack sqlite3 libsqlite3-dev libzstd-dev:i386 libzstd-dev cmake lib32z1-dev unar inkscape imagemagick subversion git bc wget sudo ccache gcc g++ binutils autoconf automake autopoint libtool-bin make bzip2 libncurses5-dev libreadline-dev zlib1g-dev flex bison patch texinfo tofrodos gettext pkg-config ecj fastjar perl libstring-crc32-perl ruby gawk python libusb-dev unzip intltool libacl1-dev libcap-dev libc6-dev-i386 lib32ncurses5-dev gcc-multilib lib32stdc++6 libglib2.0-dev 

bzw.

sudo apt-get -y install sqlite3:i386

und es kommen exakt die gleichen Fehler wie bei @Conan179

berndy2001, 2021-02-27:
siehe #208 und probiert das: https://www.heise.de/ct/hotline/Pakete-fuer-mehrere-CPU-Architekturen-in-Linux-installieren-2056403.html

Conan179, 2021-02-27:
@fda77 nachdem ist, wie oben emfohlen worden ist, 
```
dpkg --add-architecture i386
```
aus geführt hatte, klappte as kommplette instalieren von 
https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md
für Ubuntu 20.04
ikl. das wo nur per sudo apt install klappen sollte, man sollte es also in die PREREQUISITES eintragen.
Ins make menuconfig komme ich jetzt auch wieder.

fda77, 2021-02-27:
Hab ich doch vorhin schon erweitert: https://github.com/Freetz-NG/freetz-ng/commit/fcdfe56c7f1b3eb6fcb71746cc87762e30292748

Conan179, 2021-02-27:
Das hatte ich erst später gesehen gehabt, ich kann aber bestätigen das das mein problem gefixt hat.

fda77, 2021-02-27:
Bei github werden die verlinkten "refs" leider nur in Issues geschrieben, aber nicht in Diskussionen

benleb, 2021-03-05:
btw, `ldconfig` liegt in debian unter sbin...  ✌️

fda77, 2021-03-05:
Der Pfad von Programmen ist egal: https://github.com/Freetz-NG/freetz-ng/blob/master/tools/prerequisites#L14
Hauptsache es ist im $PATH

benleb, 2021-03-05:
> Der Pfad von Programmen ist egal: https://github.com/Freetz-NG/freetz-ng/blob/master/tools/prerequisites#L14
> Hauptsache es ist im $PATH

naja, fast 😅 da ein user unter debian ldconfig nicht ausführen kann (keine rechte zum erstellen von `/etc/ld.so.cache~`) und das Makefile verbietet als root zu bauen... 😄 ✌️

fda77, 2021-03-05:
Ein Programm ausgeführt und
eine Dateien erstellen
sind aber schon verschieden Dinge. Ausserdem wird nur das Cache *ausgegeben*

benleb, 2021-03-05:
😂 np, passt schon ✌️