Freetz-NG/freetz-ng #211: Anfrage wegen REETZ_IMAGE_NAME_PREFIX

freetz-ng-mod, 2021-02-27:
Hi fda,

ich habe mal eine frage. Gibt es eine Möglichkeit FREETZ_IMAGE_NAME_PREFIX an eine andere position zu setzen als ganz vorne. ZB. Vor .image

fda77, 2021-02-27:
Dann wäre es doch kein PRÄfix mehr! https://de.wikipedia.org/wiki/Pr%C3%A4fix

freetz-ng-mod, 2021-02-27:
OK !!!!

Danke für den guten WiKI link.

Ich wusste was es heißt. Gibt es eine Chance in freetz-ng ein end name zu bekommen. Da es hier nicht um den Namen geht sondern mehr um Infos was in freetz drin ist

fda77, 2021-02-27:
Wenn man eine neue Option erstellt muss man der schon einen Namen geben. Was schlägst du denn vor? Endfix find ich nicht so gut. Idefix hat auch schon ne andere Bedeutung
Da das ganze aber ne triviale Angelegenheit ist könntest du doch einen Patch/Pullrequest machen

closed 2021-02-27 by fda77