Freetz-NG/freetz-ng #217: Bekomme WARNING meldungen das was nicht in der sources ist

freetz-ng-mod, 2021-03-03:
Ich bekomme seit neusten dies Fehler beim Bauen des Images einer 7490

`
installing and stripping modules
  pl2303.ko .........................    6.13 Kb 
  usbserial.ko ......................   13.53 Kb 
  ftdi_sio.ko .......................   11.65 Kb 
  generating modules.dep
depmod: WARNING: /home/xxxx/Ubuntu/freetz-ng/74xx/build/modified/filesystem/lib/modules/3.10.107/kernel/drivers/net/nlaudio/nlaudio.ko needs unknown symbol avm_dect_if_tx
depmod: WARNING: could not open modules.builtin at /home/rxxxx/Ubuntu/freetz-ng/74xx/build/modified/filesystem/lib/modules/3.10.107: No such file or directory
WARNING: Unresolved symbols detected, not all AVM-features may work.
No current sources by AVM? Error in kernel's .config? Ask fritzbox_info@avm.de for sources!

`
Bei der 7590

`
depmod: WARNING: /home/rxxxx/Ubuntu/freetz-ng/75xx/build/modified/filesystem/lib/modules/4.9.198/kernel/drivers/net/nlaudio/nlaudio.ko needs unknown symbol avm_dect_if_tx
WARNING: Unresolved symbols detected, not all AVM-features may work.
No current sources by AVM? Error in kernel's .config? Ask fritzbox_info@avm.de for sources!

`

fda77, 2021-03-04:
-> https://github.com/Freetz-NG/freetz-ng/discussions/134

closed 2021-03-04 by fda77

freetz-ng-mod, 2021-03-06:
https://boxmatrix.info/wiki/Property:nlaudio.ko

Es gint für die 7590 und 7490 den Sourcecode