Freetz-NG/freetz-ng #219: Bluetooth module "btusb.ko" broken with FW 7.21

JakobTewes, 2021-03-07:
Hey there,

create an issue before. As it seemed to work with just commenting out "l2cap" with 7.12, I didn´t follow the trail to the end.
[old-issue](https://github.com/Freetz-NG/freetz-ng/issues/40).

Now with 7.21 my iBeacon set up via bluez-included tools hciconfig and hcitool no longer works, as the kernel module btusb won´t load.

Think, its a global problem. How to reproduce (version 7.21 current with FBF 7580):

Configure and add "bluez-utils". This adds options

`
FREETZ_PACKAGE_BLUEZ_UTILS=y
FREETZ_LIB_libbluetooth=y
FREETZ_MODULE_bluetooth=y
FREETZ_MODULE_bnep=y
FREETZ_MODULE_l2cap=y
FREETZ_MODULE_rfcomm=y`

Then make the thing

Will fail with

[image]

Old workaround was adding the configs above and then commenting out
`FREETZ_MODULE_l2cap=y`
with that (and only that way) the build succeeds.

Really in need for that in order to use Presence detection via iBeacon. Someone can dig in please? 
What additional info may i provide? Kept my build for now and can provide build logs or such info ASAP.

Thanks and regards,

Jakob

fda77, 2021-03-07:
> Think, its a global problem

No, pebkac. Just do the "yellow" on your screenshot: "make kernel-menuconfig" and select "l2cap"
If it work, upload your result as pullrequest (or patch)
There are about 50 kernel .config in Freetz and i dont want to change + test-compile every single

closed 2021-03-07 by fda77

JakobTewes, 2021-03-08:
Sorry @fda77, but this is frustrating. Last time you gave the exact same answer ;-(.
There is no „l2cap“ option in „make kernel-menuconfig“. 
The option is activated via „make menuconfig“, no corresponding option in kernel config...

JakobTewes, 2021-03-08:
Think, it is not completely hooked in (via kernel-menuconfig), so that the kernel module isn’t included correctly.

JakobTewes, 2021-03-08:
With this, the whole bluetooth subsystem doesn’t work, als l2cap is base of all bluetooth protocols

fda77, 2021-03-08:
Then .. enable the module as described above

JakobTewes, 2021-03-08:
grml....it is activated as dependency by selecting "bluez" Package and cannot be unticked.
unfortunately it seems, that the "bringing the kernel module in" thing is broken.

JakobTewes, 2021-03-08:
and agin...there is no corresponding option in kernel-menuconfig

fda77, 2021-03-08:
Sorry, i did not understand the "There is no „l2cap“ option in „make kernel-menuconfig“." correctly
Freetz-menuconfig was not updated with latest kernel versions, and there are sure more errors...
l2cap does only exist up to kernel v3:  https://cateee.net/lkddb/web-lkddb/BT_L2CAP.html 

Ther's a new BT_INTEL since 4.1: https://cateee.net/lkddb/web-lkddb/BT_INTEL.html
If it does not work now as it is, try to add "btintel" to "own modules" (and reportso it could be added)
And maybe additional broadcom or reaaltec protocol support (i have no bt)

From kerneel Kconfigs:
```
btusb.ko
config BT_HCIBTUSB
	tristate "HCI USB driver"
	select BT_INTEL

btintel.ko
config BT_INTEL

btbcm.ko
config BT_HCIBTUSB_BCM
	bool "Broadcom protocol support"
	depends on BT_HCIBTUSB
	select BT_BCM
config BT_BCM
	select FW_LOADER

btrtl.ko
config BT_HCIBTUSB_RTL
	bool "Realtek protocol support"
	depends on BT_HCIBTUSB
	select BT_RTL
config BT_RTL
	select FW_LOADER
```

JakobTewes, 2021-03-08:
Aaaah...with context, that menuconfig has not been updated for new kernel (yet), 
things look reasonable. The "bintel" thing tends to be part of the solution, 
as I see following Errors, when trying to load btusb.ko

`btusb: Unknown symbol btintel_secure_send (err 0)`
`btusb: Unknown symbol btintel_set_event_mask (err 0)`
`btusb: Unknown symbol btintel_check_bdaddr (err 0)
`btusb: Unknown symbol btintel_read_version (err 0)`
`btusb: Unknown symbol btintel_version_info (err 0)`
`btusb: Unknown symbol btintel_set_bdaddr (err 0)`
`btusb: Unknown symbol btbcm_setup_patchram (err 0)`
`btusb: Unknown symbol btintel_exit_mfg (err 0)`
`btusb: Unknown symbol btintel_set_diag (err 0)`
`btusb: Unknown symbol btintel_load_ddc_config (err 0)`
`btusb: Unknown symbol btintel_set_event_mask_mfg (err 0)`
`btusb: Unknown symbol btintel_hw_error (err 0)`
`btusb: Unknown symbol btbcm_setup_apple (err 0)`
`btusb: Unknown symbol btintel_set_diag_mfg (err 0)`
`btusb: Unknown symbol btbcm_set_bdaddr (err 0)`
`btusb: Unknown symbol btintel_enter_mfg (err 0)`
`btusb: Unknown symbol btrtl_setup_realtek (err 0)`
`

As I´m far away from beeing a kernel hacker (okay lame joke), could you please give me a hint on where to look for building and integrating own modules?

fda77, 2021-03-08:
menuconfig ... kernel modules ... own modules

From unknown symbols it seems btbcm and btrtl are also needd

Yes, the freetz-menuconfig is for all kernel versions and modules are added, deleted, renamed, changed dependencies etc