Freetz-NG/freetz-ng #221: Problem mit erweiteren DSL-Einstellungen / lesen der manuellen tunings die zuvor gesetzt wurden

Himan2001, 2021-03-09:
Seit langer Zeit gibt es das Problem das beim "DSL-Linespeed Patch" zwar die Werte extrem von links nach Rechts verschoben
werden koennen - diese werden dem setzen aber nicht in den korrekt gesetzen Stufen wieder gelesen.

Also Downstream steht immer mittig nach dem setzen. Upstream springt teilweise auch zurueck.
und bei der Stoerunterdrueckung ist es auch nicht sauber mit dem auslesen aus der ar7.cfg.

Auf einer älteren 3370 kann man gesetzt Werte teilweise korrekt lesen, was eingestellt wurde.
Auf einer 7390, 7362 und 7490 stehen manuell gestze Werte immer auf "Mittig".

Ich vermute auf Grund der Unterschiedlichen Modelle das die Werte hier eine andere "Matrix" haben innerhalb der AR7 und beim Webinterface Patch daher nicht korrek umgesetzt werden.

Weiterhin ist mir aufgefallen, das der Schritt von stufe Mitte +1 nach Mitte +2 nach recchts massiv ist.
In meinem Fall habe ich einen Unterschied von 47 MBit -> 62 MBit mit 1db (!) SNR. Zwischenwerte und kleinerer Abstufungen bis ganz nach rechts waeren hier der Hit.

Sobald der Wert Mitte +3 gesetzt wird, bzw. noch weiter nach rechts verbindet die Box nur noch mit etwa 14 MBit und der SNR geht weiter ueber 14db. Da scheints wohl ein Problem mit der Logik des DSL-Tunings zu geben ?

fda77, 2021-03-09:
Der Standardwert ist "0". Bei AVM kann man negative Werte setzen -1, -2 etc. Durch den Patch kann man Positive setzen.
Ob das klappt hängt vom Gerät und (DSL-)Firmare ab. Wenns nicht geht lass den Patch halt weg
Wie sich das auf jede Leitung auswirkirt muss jeder selbst ausprobieren.
Siehe auch https://freetz-ng.github.io/freetz-ng/patches/MODIFY_DSL_SETTINGS.html

closed 2021-03-09 by fda77

fda77, 2021-03-10:
Ich weiss nciht so recht. Der Patch ist jedenfalls dieser: https://github.com/Freetz-NG/freetz-ng/blob/master/patches/cond/831-extend-dsl_line_settings-speed_07_2X.patch  für alle 7.20++ FOS
Du kannst ja mal vergleichen was sich in der ar7.cfg ändert, einfach eine Kopie machen, Wert ändern und dann "diff". Ich glaub seit längerem wird der 10-fache Wert dort geschrieben.
Evtl fehlt im Patch auch noch etwas.
Da könnte man direkt auf der Box mit "mount -o bind" die /usr/www/all/internet/dsl_line_settings.lua beschreibar machen und ausprobieren