Freetz-NG/freetz-ng #225: PLCD-Daemon failto start

Cyklops66, 2021-03-12:
Fritzbox: 7590
Firmware: 154.07.25 rev86535
Freetz: ng-18138M-2d46848f9

Hello,

the plcd-daemon can't start. If i try it to start directly over /sbin/plcd it fails with the following error:

`terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted
`
The PLCD-Daemon is for the Linux-Users without any Windows machine relativ important to check and maintain the powerline-network.

Thank you very much and the goog work.

Greetings

fda77, 2021-03-15:
I cant tell anything about PLCD, execpt that i removed it :)
 - So its possible that some modifiction caused the crash
 - Or somehow the AVM-Configuration is bad (this happens sometime)
 
How you noticed it? Did the device crash/reboot?
To test the 2nd, create a settings backup und reset to facotry default and check it

Cyklops66, 2021-03-16:
Thank you for your answer. It is a locale problem,  If i override the localistation setting, it will start fine.
A simple workaround is to put this override in the rc.custom:

`#!/bin/sh

export LC_ALL=C; unset LANGUAGE &&
plcd

exit 0`

It is strange, because no other program is affected by this problem. But for now it is running fine.

fda77, 2021-03-16:
Maybe because PLCD does not like uft8?
https://github.com/Freetz-NG/freetz-ng/blob/master/patches/scripts/700-set_lang_to_utf8.sh
This File was added 9 years ago: https://trac.boxmatrix.info/freetz-ng/changeset/9386/freetz-ng

fda77, 2021-03-16:
AVM seems not to use  `LC_ALL`. So it should be enough to `unset LANG`. 
`LANGUAGE` is set by AVM and seems not be modified by Freetz:
```
/etc/boot.d/core/tffs:local LANGUAGE="`echo provider_default.language | ar7cfgctl -s 2>/dev/null | sed s/\\"//g`"
```

fda77, 2021-04-03:
Gibt es jemanden mit 7590 der PLCD nicht entfernt hat und das Problem auch hat?

fda77, 2021-04-22:
Factory reset should fix it

closed 2021-04-22 by fda77