Freetz-NG/freetz-ng #236: 7520: Absturz vsftpd beim Aufbau der ftp Verbindung

forenuser, 2021-03-21:
7520-Alien7530_07.25.all_freetz-ng-18157-b7e2f21c7
Minimal Image (nur vsftpd)
Der Absturz ist nicht mehr von einer aktiven DSL Verbindung abhängig und lässt sich damit leichter reproduzieren.

fda77, 2021-03-21:
@Gismotro hat doch auch eine 7520, vielleicht kann er testen ob das bei ihm auch auftritt

forenuser, 2021-03-21:
@Gismotro wäre klasse, wenn Du vsftpd bei Deiner Box testen könntest. Ich hänge meine Konfiguration mal an.
[.config.1.txt](https://github.com/Freetz-NG/freetz-ng/files/6178198/default.config.1.txt)

Gismotro, 2021-03-22:
Kann ich bestätigen.
Sobald ich die Box per TotalCommander auf Port 2121 anspreche, bootet die Box neu.
Hierbei ist es sogar egal ob der AVM-FTP  gestartet ist oder nicht.

Conan179, 2021-03-22:
ISt das etzt das selbe problem wie
https://www.digital-eliteboard.com/threads/suche-jemanden-mit-7520-bzw-7530-zur-verifikation-vsftpd.497223/
oder ein neues?

Gismotro, 2021-03-22:
@Conan179 : Möglich

@forenuser :  Was noch interessant ist:

- löschen vom AVM-FTP hilft nicht.
- Wenn ich die Box aber auf Port 21 mit root anspreche, kommt es zum reboot. Nutze ich Gismotro als Benutzer, passiert nix außer das die Rückmeldung kommt das der Nutzer sich nicht anmelden konnte (Verbindung nicht erfolgreich)

Conan179, 2021-03-22:
CommanderAdama sollte am besten auch hier schreiben, er hat ja scheinbar das selbe Problem, mit der selben
Typ von Box
https://github.com/Freetz-NG/freetz-ng/discussions/182
https://github.com/Freetz-NG/freetz-ng/discussions/179

Gismotro, 2021-03-22:
Interessant ist auch dieses : 

root@004-7520:/var/mod/root# cat /var/tmp/passwd
`asec::101:101::/nonexistent:/noshell
ftp:x:100:1:ftp:/home/ftp:/bin/false
root:x:0:0:root:/mod/root:/bin/sh
smb::100:100::/nonexistent:/noshell
ftpuser:x:102:1:ftpuser:/var/media/ftp:/bin/false
boxusr11:$6$iFU2e2pD8[...gelöscht.....]TCfPRt7Pl0:1011:0:box user:/home-not-used:/bin/sh
boxusr11int:$6$gT79ch3[....gelöscht....]$uFlvFGF6C5MJ12yDGr.:2011:0:box user:/home-not-used:/bin/sh
boxusr10:$6$7fmafNYpJ[...gelöscht...]AbUtcRGQrGGubgZbW.:1010:0:box user:/home-not-used:/bin/sh
boxusr10int:$6$4Swnrj.Z0z[....gelöscht.....]BYkJuBMYm1EB9V/pyGQs9.1:2010:0:box user:/home-not-used:/bin/sh`

Pw wechsel geht nicht: 

`root@004-7520:/var/mod/root# passwd teser3
passwd: unknown user teser3
root@004-7520:/var/mod/root# passwd tester3
passwd: unknown user tester3`

aber warum kommt dann dieses ?

`root@004-7520:/var/mod/root# adduser -h /var/media tester
adduser: group 'tester' in use
root@004-7520:/var/mod/root# adduser -h /var/media tester3
adduser: group 'tester3' in use
root@004-7520:/var/mod/root#`

fda77, 2021-03-22:
CommanderAdama ist einer der vielen Accounts von forenuser ^^

> Es werden keine neuen Benutzer in der passwd dauerhaft gespeichert

Ein bekanntes Addon-Problem https://freetz-ng.github.io/freetz-ng/ADDONS.html

>  passwd: unknown user teser3

Das liegt daran dass es laut deiner passwd-Datei diesen Benutzer nicht gibt

forenuser, 2021-03-22:
@Gismotro
CommanderAdam ist mein Ausweichaccount, wenn fda77 meinen regulären forenuser Zugang sperrt.

Das Osc** Addon verhindert das Wiederherstellen von gesicherten Einstellungen. Das wurde trotz meines Hinweises nicht vollständig behoben. Ob dies die Speicherung von Einstellungen, die im Freetz-WebIF oder in der Shell geändert wurden, habe ich nicht getestet. Ich benutze das Addon nicht mehr.

**Benutzer anlegen:**
addgroup ftp
adduser ftp -p <passwort> -g ftp
opt. passwd ftp
moduser save
modsave flash

**Benutzer überprüfen:**
cat /etc/passwd
cat /etc/shadow
cat /etc/group

Ich nehme an, dass Du versuchst die in der AVM Konfiguration hinterlegten Benutzer zu verwenden. Das funktioniert nicht.

Zu Deinen Fragen:
- löschen vom AVM-FTP hilft nicht?
- Nein.

- Es werden keine neuen Benutzer in der passwd dauerhaft gespeichert. Nach rebbot ist wieder der alte Inhalt vorhanden. (cat /var/tmp/passwd)
- s.o.

Conan179, 2021-03-22:
> Das Osc** Addon verhindert das Wiederherstellen von gesicherten Einstellungen.
Oscam, nennen wir das programm beim Namen
https://freetz-ng.github.io/freetz-ng/ADDONS

Conan179, 2021-03-22:
Sag mal, tritt dein fehler auch bei freetz(ohne-ng) auf?

forenuser, 2021-03-22:
Habe ich nicht getestet.
Bftp läuft mit freetz-ng.

forenuser, 2021-03-22:
> > Das Osc** Addon verhindert das Wiederherstellen von gesicherten Einstellungen.
> 
> Oscam, nennen wir das programm beim Namen
> https://freetz-ng.github.io/freetz-ng/ADDONS

Das Programm ist nicht der Verursacher, sondern die addon Umgebung.

> Sag mal, tritt dein fehler auch bei freetz(ohne-ng) auf?
Habe ich nicht getestet.
Bftp läuft mit freetz-ng.

Conan179, 2021-03-22:
Ok dann teste es doch mal mit freetz.

forenuser, 2021-03-22:
> dann teste es doch mal mit freetz.

Was soll das bringen?

Conan179, 2021-03-22:
Auschliesen das freetz-ng das problem ist?

forenuser, 2021-03-22:
> Auschliesen das freetz-ng das problem ist?

Das ist fast so, als würde ich vsftpd auf Debian, Ubuntu, ... testen.

Conan179, 2021-03-22:
Na wen du meinst. Bitte, ist dein problem.

fda77, 2021-03-22:
Kann schon sein dass ein Problem mit Freetz/ nicht auftritt. Das kann einerseits an den Weiterentwicklungen liegen, oder daran dass man eine paar Jahre alte FOS Version nutzt in der auch vieles anders ist.
Das Restore-Problem mit dem Addon ist weil es ohne Grund das Flash kaputtschreibt. Laut Chefentwikckler ist es aber genau so gewollt ^^
Das "Benutzer werden gelöscht" war aber was anderes, ich weiss aber nicht mehr genau was. Ich kenne das Addon auch nur vom lesen und nicht genau. Vielleicht kommt das auch nur weil modsave im falschen Moment aufgerufen wird? Ich hab jedenfalls noch Benutzer von Programmen im System die ich seit Jahren nicht mehr verwende.
Davon abgesehen muss man /etc/passwd speichern bevor man rebootet...

Gismotro, 2021-03-22:
also:
1.) ich benutze kein Addon auf meiner Box. Ich brauche kein OSCAM und warum soll ich was testen wenn alle doch sagen das es an dem Addon liegt was ich nicht benutze ..... 
2.) klar habe ich immer brave madsave all gemacht. Steht ja auch so im HowTo....
3.) Kann ich sonst noch helfen ?

forenuser, 2021-03-22:
@Gismotro
Ich verstehe das lediglich als Ausschluss einer Fehlerursache. Hast Du beide Anweisungen ausgeführt und den Inhalt der angegebenen Dateien überprüft?
moduser save
modsave flash

@fda77
Kaputtgeschrieben wird der Flash ja nicht. Nur werden nach dem Restore der Konfiguration beim Stoppen des Addons der bereits wiederhergestellte Daten im Flash wieder mit den Ram Daten überschrieben. Normale Änderungen der Konfiguration sollte demnach ohne Probleme funktionieren.

Gismotro, 2021-03-22:
Also noch mal:
Ja, ich habe die User testuser-1 bis 3 richtig angelegt und auch die passwd kontrolliert.
Bis zum ersten test per VSFTP waren auch alle drei enthalten, aber nach dem reboot sind alle Nutzer wieder raus.
Was mich aber gewundert hat, ist das ich keinen User mehr mit dem selben Namen testuser-1 anlegen kann da er angeblich in use ist, aber ein PW kann ich dem User nicht vergeben da dann kommt: unknown user.
Bei root und ftpuser kann man aber das Pw nach einem reboot ändern.

forenuser, 2021-03-22:
Kann ich bestätigen. Der angelegte Nutzer (test1) fehlt trotz korrekter Speicherung in /etc/passwd nach einem Neustart. Das in /etc/shadow gespeicherte Passwort bleibt erhalten.

fda77, 2021-03-23:
Jo, ist mal wieder alles Durcheinander hier... Ich glaube auch das ftp Problem sind 2, so sicher bin ich aber nicht.
Sind die die 4 Passwortdateien in /tmp/flash/users/ vollständig?

forenuser, 2021-03-23:
Die 4 Passwortdaten scheinen vollständig zu sein. Der angelegte Benutzer test1 ist in allen 4 Dateien enthalten.

Ich könnte zwei neue Issues eröffnen:
1. vsftpd
2. Benutzer

Dann können wir diesen Issue stilllegen.

fda77, 2021-03-24:
Und wie ist es in /etc nach "modusers load" ?

forenuser, 2021-03-24:
Es werden alle boxusr* und app* aus /etc/passwd entfernt.

fda77, 2021-03-24:
Das sollte nicht sein. Das müsste eigentlich mal funktioniert haben, evtl hat AVM was geändert

forenuser, 2021-03-26:
Wenn ich den Aufruf von modusers wie folgt ausführe, funktioniert das Skript:
```
cp /tmp/passwd /tmp/passwd.tmp
modusers load
```
Damit ist mein test1 Benutzer in /etc/passwd enthalten. Warum funktioniert dies nicht beim Neustart der Box?

forenuser, 2021-03-27:
Ich habe eine freetz Firmware gebaut und getestet. FW Version ist 07.14. Vsftpd funktioniert und gespeicherte Freetz Benutzer sind in /etc/passwd nach dem Neustart vorhanden. Das Skript modload wird wie bei freetz-ng nicht mit dem Paramter force aufgerufen, so dass darüber keine Wiederherstellung von gespeicherten freetz Benutzern beim Booten erfolgt. Bei freetz 7.14 wird im Gegensatz zu freetz-ng 7.2x libctlmgr verwendet. Darin wird modusers update aufgerufen. Im Gegensatz zu freetz-ng gibt es die Datei /tmp/passwd.avm, was darauf hinweist, dass modusers ausgeführt wurde. Vermutlich erfolgt dies durch libctlmgr.

Da bei freetz-ng 7.2x keine libctlmgr verwendet wird, müsste die Wiederherstellung der freetz Benutzer durch modload force oder auf einem anderen sinnvollen Weg erfolgen. Bei modload force müsste ein Workaround implementiert werden, da passwd.tmp nicht vorhanden ist. Damit werden die AVM Benutzer nicht in die passwd Datei übernommen.

fda77, 2021-03-29:
Achso, es betrifft die 7.1x. Hättest du gleich schreiben sollen. Da wird noch die preload-Lib genutzt

forenuser, 2021-03-29:
Nein, es betrifft die 7.2x. Bei der 7.1x mit preload-Lib funktioniert es ja. Ich denke, dass bei ARM Boxen mit uClibc weiterhin die Funktion der preload-Lib erforderlich ist, da sonst die Freetz Benutzer in /etc/passwd gelöscht werden. Oder hast Du eine andere Lösung?

fda77, 2021-03-29:
Warum soll es mit 7.1 plötzlich nicht mehr funktionieren? Und warum betrifft es nur uclibc?
Wie wärs wenn sich jemand findet der modusers so anpasst dass es funktioniert?

forenuser, 2021-03-29:
Ich habe mehrfach geschrieben, dass es mit der 7.1 FW funktioniert und das Problem bei der 7.2x Firmware bei Fritzbox 7520/7530 mit ARM Prozessor auftritt. Der Fehler hat mit modusers primär nichts zu tun und kann darin nicht gelöst werden. Kannst Du bitte mal in die libctlmgr.c reinschauen. Darin siehst Du, wofür dieses Modul verantwortlich ist. Die Funktion rename() wird durch einen speziellen Code ersetzt, worin das Umbenennen der Datei /etc/passwd.tmp in /etc/passwd abgefangen wird und stattdessen modusers update aufgerufen wir. Das wird deshalb getan, da der ctlmgr in bestimmten Situationen /etc/passwd mit den AVM Benutzern überschreib. Freetz Benutzer sind dem ctlmgr nicht bekannt. Über den Aufruf von modusers update anstelle Ausführung von rename der vom ctlmgr erzeugten passwd.tmp Datei werden die Freetz Benutzer in diesen Situationen wieder eingetragen. Damit ist dieser Hook zwingend erforderlich. Mir fällt dafür nur die Verwendung der gleichen libc Version in Freetz wie in AVM FW ein, um preload von libctlmgr wieder zu ermöglichen.
Siehst Du eine andere Möglichkeit?

fda77, 2021-03-29:
Ach du redest nicht von dern usern sondern vom ftp?????? Tud mir leid ich komm da nicht mit...

forenuser, 2021-03-29:
Das habe ich eindeutig beschrieben, oder?

Conan179, 2021-04-08:
Fürs Protokoll, der fehler wurde behoben in dem "isolate_network=NO" in die /etc/vsftpd.conf eingetragen worden ist
https://www.ip-phone-forum.de/threads/freetz-ng-fritzbox-st%C3%BCrzt-bei-verbindung-mit-vsftpd-ab.309351/page-2#post-2423680
War also kein Freetz Fehler.

fda77, 2021-04-09:
Danke. Vielleicht sollte man das dann als Default da eintragen? Hat das Nachteile auf anderen Geräten? Und welche sind genau betroffen? ARM allgemein kann es nicht sein, da auf meinem ARM Repeater das Problem nicht auftritt
Es ist aber schon ungewöhnlich das ein Linux-Programm den Kernel zum booten bringen kann! Nach den Logs sieht es für mich mehr so aus als ob da jemand mit iptables gespielt hat, was bekanntlich wegen dem PA-Aufsatz von AVM im Kernel das ganze Gerät crasht

PeterPawn, 2021-04-09:
> Nach den Logs sieht es für mich mehr so aus als ob da jemand mit iptables gespielt hat

Das macht mich jetzt wieder neugierig ... woran siehst Du das denn genau?

Außerdem wurde das Problem ja von @Gismotro bestätigt ... willst Du dem jetzt auch eine "geheime Nutzung von `iptables`" reinsingen?

Meine Bitte (im IPPF), endlich mal erwachsen zu werden, richtete sich AN ALLE, die das immer wieder nicht auf die Reihe kriegen und solange Du keine konkreten(!) Anhaltspunkte nennen kannst, die bei Dir eine Vermutung in Richtung `iptables` auslösen, ist das für mich auch nichts anderes als "nachtreten".

Hast Du eine Vorstellung, was alles an Freetz-NG geändert werden müßte, damit man da tatsächlich mit einem eigenen `iptables`-Paket auf IPQ-Boxen herumexperimentieren kann? Siehst Du in der Konfiguration, mit der @Gismotro das nachgestellt hat, irgendeinen Hinweis auf solche Anstrengungen?

---

Ja, das Problem tritt vermutlich tatsächlich nur auf ARM-basierten FRITZ!Boxen (und vielleicht auch nicht mal auf allen Modellen) auf - aber sicherlich eher durch irgendwelche AVM-Patches in der Netzwerk-Behandlung verursacht, die eben nicht berücksichtigen, daß es auch `clone()`-Aufrufe mit `CLONE_NEWNET`-Option geben könnte. Was dann auch erklären könnte, warum das auf Deinem ARM-Repeater nicht vorkommt, obwohl Du den Vergleich der Kernel-Optionen für die beiden Geräte (Repeater vs. FRITZ!Box 7520/7530) auch noch "schuldig" wärst, weil natürlich ein Kernel OHNE Namespaces dann vermutlich nicht abstürzen wird - und zusätzlich wird der Repeater ja auch kein WAN-Interface haben und schon gar keinen AVM-WAN-Stack. Ja, vermutlich gibt es auf dem Repeater nicht mal einen `ctlmgr` und der wird - wie andere vor ihm - einen `rextd` haben.

---

Auch irgendwelche Streitereien, ob das nun ein "generelles Freetz-NG-Problem" ist oder doch nur ein modellspezifisches Problem (wie sie gerade wieder im IPPF geführt wurden), tragen NULL zu einer Lösung bei - entscheidend ist nicht, wie ich das nenne, sondern wie ich es be- und verarbeite und wie ich es letztlich löse. All diese Spiegelfechtereien und Streitereien sind dabei in keinster Weise hilfreich - also verzichte(t) bitte künftig einfach darauf.

fda77, 2021-04-09:
Im Crashlog im IPPF sind diverse netfilter Dinge zu sehen.
```
<3>[ 1219.362206][F1] 3e94: [<c06b7aa0>] (0xc06b7aa0 nf_queue_nf_hook_drop+0x30/0x3c)
<3>[ 1219.362223][F1] 3eac: [<c06b6910>] (0xc06b6910 nf_unregister_net_hook+0xa8/0xc4)
<3>[ 1219.362240][F1] 3ebc: [<c06b6a84>] (0xc06b6a84 netfilter_net_exit+0x2c/0x50)
```

Welches Problem siehst du genau darin, developer-competence auszuwählen und danach iptables? Das wurde auch vor kurzem aktualisiert und läuft auf meiner 7590 (ohne state!). Im DEB wurde da auch wieder herumgebasht weil ich das iptables-cgi "gelöscht" hätte.
Für conntrack etc wären noch die Module im Kernel nötig. Der nervige Bayer kann das da er schon mal Patch für die Kernelconfig machte.  Solang ich hier der einzige bin der auf jeden Fall antworten muss hab ich keine Lust auf den. Wenn es mal mindestens ein paar Mods ( ;;;;;-) ) gibt können die sich gern mit ihm herumschlagen. Dieses ständige herumgezicke halte ich für Zeitverschwndung.
FTP hat ja ein Spezialbehandlung im conn-track wegen control+data Verbindung. Es könnte auch sein dass AVM da was aus Versehen hineincompiliert hat. Um PA komplett auszuschliessen könnte mal diesen einfach per echo (>FAQ) ausschalten und schauen ob es dann immer noch crasht. Für PA sind komplett-Crashes typisch

PeterPawn, 2021-04-09:
> Im Crashlog im IPPF sind diverse netfilter Dinge zu sehen.

Ach ... dann hat er also auch nicht nur die notwendigen Pakete erzeugt, sondern gleich noch einen eigenen Kernel generiert? Weil bei AVM ja kein `NETFILTER` im Kernel ist und damit auch kein Cleanup-Code dafür vorhanden sein kann? Oder wie willst Du mir/uns erklären, daß ein Kernel-Worker-Thread offensichtlich diesen Code ausführt? Das macht der ja gewöhnlich nur dann, wenn der Code auch im Kernel steht ... oder zumindest in irgendeinem LKM.

Wie hat er das eigentlich ohne die AVM-Quellen für die 7520/7530 gemacht? Wenn ich Freetz-NG trauen will, kann ich bei 7520/7530 mit 07.25 nicht mal als `Developer` ein "replace kernel" auswählen.

Und wie hat er dann @Gismotro dazu gebraucht, nicht mit der Konfiguration oben aus dieser Issue zu arbeiten, sondern mit einer "geheimen", in der das alles enthalten war, was Du da zu sehen glaubst? Oder stecken die beiden unter einer Decke? Wer denn dann noch alles?

---

Sorry, aber das ist als "Begründung" alles schon sehr schwach ... und so, wie er im IPPF (wohin Du ja verlinkt hast) bewegt, machst Du das tatsächlich hier in GitHub. Anstatt da einfach mal sachlich an die Dinge heranzugehen, stellst Du irgendwelche wilden Behauptungen auf, woher das Problem EIGENTLICH käme ... dann mußt Du Dich auch nicht wundern, wenn diese Kommentare niemand mehr ernst nimmt und wenn niemand mehr ernsthafte(!) Fehlermeldungen erstellen will.

Aber wie schon im IPPF geschrieben ... ich sehe mir das jetzt erst mal wieder eine Weile von der Seitenlinie an, bis sich alle wieder beruhigt haben und es hoffentlich möglich ist, sachlich (und an einer Stelle) solche Probleme zu wälzen. Auf alles andere habe ich keinen Bock und dann weiß ich mit meiner Zeit tatsächlich auch besseres anzufangen.

Conan179, 2021-04-09:
@PeterPawn forumsusre und gizmondo sind beide im digital-eliteboard Forum aktiv.

PeterPawn, 2021-04-09:
Das ist ja per se noch kein Grund, warum die beiden jetzt nicht tatsächlich mit der vorliegenden Konfiguration auf einer IPQ-Box genau dieses Problem gehabt haben könnten.

---

Und nur nebenbei ... ich lese durchaus (wenn auch nicht regelmäßig) mal im DEB mit und weiß schon, wer da jeweils aktiv ist und kann mir auch (schon aus dem verwendeten Schreibstil) bei den meisten zusammenreimen, ob und wenn ja, unter welchem Account sie auch im IPPF aktiv sind. Nur selbst schreiben werde und will ich dort auch nicht - aber das gilt auch für andere Boards ohne zweifelhafte Inhalte, wie Router-Forum oder Online-Kosten. Man kann nicht auf allen Hochzeiten gleichzeitig tanzen ... und irgendwo müssen manche Nutzer ja auch die Chance kriegen, sich über meine unmögliche Art zu echauffieren (schon deshalb lohnt der Blick nach onlinekosten.de oder auch in die Kabel-Foren).

---

Wenn ich mich im IPPF dagegen wehre, auf DEB zu verlinken, heißt das noch nicht, daß ich alles dort für unglaubwürdig und alle Protagonisten für Lügner und/oder Verbrecher halte ... auch wenn da schon (meine höchstpersönliche Einschätzung, zu der ich auch keine Namen nennen werde) deutlich mehr Unwissen vorhanden ist, was aber trotzdem niemanden abhält, große Töne zu spucken. DAS finde ich im IPPF angenehmer, weil da dann auch andere solche Leute schneller zurechtstutzen.

---

Es spielt für meine Beurteilung der Glaubwürdigkeit eines anderen also keine Rolle, wo der ansonsten noch schreibt und unter welchen Accounts ... da halte ich mich - als strenggläubiger Atheist - an das Matthäus-Evangelium: "An ihren Früchten werdet ihr sie erkennen." und die kann man dann ja an den jeweiligen Orten (je nachdem, wo sich die Leute tummeln) nachlesen.

fda77, 2021-04-09:
Ich glaub Foxy keinen Buchstaben. In seinem Log sieht man zb auch noch spuren von Oscam. Dh er hat nicht mit einer sauberen Config gestestet, sondern es war mal ein Addon installiert. Dieses ist dafür bekannt Mist in die rc.custom zu schreiben. Es werden auch custom.in gehandelt die "geheime Funktionen" im Menuconfig freischalten.
Wie dem auch sei, PA kann man ganz einfch ausschliessen indem man ihn kurz mal deaktiviert.
Ansonsten wird es hier nichts aus dem Aus zu sehen geben, da ich auf Spielchen keine Lust hab

Conan179, 2021-04-10:
Für die Akten, das 2te Problem vom USer, ist behoben, das bei seiner Box Benutzer nicht gepseichert worden ist, war nicht die schult von freetz-ng sondern, ich zitiere:
" Beim Update des lokalen Git Repo wurde die **von mir gepatchte Datei** make/mod/files/usr/bin/modpasswd nicht überschrieben. Wenn ich die Datei aus dem Freetz-NG Git verwende, funktioniert die persistente Speicherung."
https://www.ip-phone-forum.de/threads/freetz-ng-fehler-in-der-verwaltung-von-freetz-benutzern-7520-7530-7-2x-ggf-weitere-modelle-betroffen.309948/page-2#post-2423990

fda77, 2021-04-10:
Wenn PP nichts dagegen hat commit ich morgen früh, dann brauch man gar keine neuen Befehle https://github.com/Freetz-NG/freetz-ng/discussions/261

PeterPawn, 2021-04-11:
Ich spiele hier gar keine Rolle ... und meine selbst verordnete Auszeit war absolut ernst gemeint. Ich muß mich selbst auch ab und an mal runterkühlen. Gilt sinngemäß auch an allen anderen Stellen, wo man mich nach meiner Meinung fragen könnte. War allerdings auch auf Freetz-NG (hier und im IPPF) beschränkt.

Ich bin ab heute einfach noch fünf Tage krank und bettlägerig, die Augen tränen so sehr, daß ich nur schreiben und nichts lesen kann, die Buchstaben auf dem Keyboard sind gerade groß genug, um sie zu erkennen.

fda77, 2021-04-30:
Man muss wohl für dieses Gerät mit genau diesem Kernel eine zusätzliche Option in vsftpd setzen. Welche steht irgendwo im IPPF

closed 2021-04-30 by fda77

shv1, 2021-05-01:
isolate_network=NO