Freetz-NG/freetz-ng #237: 7520: Make bricht bei pycurl ab

forenuser, 2021-03-21:
mkdir -p packages/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1/root
if test -d make/python-pyopenssl/files; then    tools/tar-gnu -cf - -C make/python-pyopenssl/files --exclude=.svn --exclude=.git --exclude=.gitignore --exclude=.build-prereq-checked --exclude=.unpacked --exclude=.configured --exclude=.compiled --exclude=.installed . | tools/tar-gnu -xf - -C packages/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1; fi
---> package/python-pyopenssl ... preparing ... mkdir -p source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1; tools/gunzip -c dl/pyOpenSSL-0.13.1.tar.gz | tools/tar-gnu -x -C source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1 --transform='s|^./\+||' --strip-components=1
set -e; shopt -s nullglob; for i in make/python-pyopenssl/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1 $i; done;
    applying patch file make/python-pyopenssl/patches/010-no_tests.patch
    patching file setup.py
    ----------------------------------------------------------------------
    applying patch file make/python-pyopenssl/patches/020-openssl-1.0.2-name-clash.patch
    patching file OpenSSL/crypto/crl.c
    ----------------------------------------------------------------------
( export PYTHONPATH="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/lib/python2.7:/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi//usr/lib/python2.7/site-packages"; export PYTHONOPTIMIZE=""; export PYTHONDONTWRITEBYTECODE=1;  cd source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1;  PATH="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin:/home/freetz/7520/toolchain/build/arm_gcc-5.5.0/arm-unknown-linux-gnueabi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" FREETZ_LIBRARY_DIR="/usr/lib/freetz" FREETZ_KERNEL_VERSION_MAJOR="4.4" FREETZ_TARGET_ARCH="arm" FREETZ_TARGET_ARCH_ENDIANNESS_DEPENDENT="arm" CC="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc" CXX="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-g++-wrapper" CFLAGS="-mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" CXXFLAGS="-mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" LDFLAGS="" PKG_CONFIG_PATH="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/../lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/../lib/pkgconfig" GLOBAL_LIBDIR=/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/lib LD_RUN_PATH="/usr/lib/freetz"    /home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/bin/hostpython  ./setup.py  install --prefix=/usr --root=/home/freetz/7520/packages/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1/root  ; )
/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'zip_safe'
  warnings.warn(msg)
running install
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/OpenSSL
copying OpenSSL/__init__.py -> build/lib.linux-x86_64-2.7/OpenSSL
copying OpenSSL/tsafe.py -> build/lib.linux-x86_64-2.7/OpenSSL
copying OpenSSL/version.py -> build/lib.linux-x86_64-2.7/OpenSSL
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
building 'OpenSSL.crypto' extension
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/OpenSSL
creating build/temp.linux-x86_64-2.7/OpenSSL/crypto
/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc -fno-strict-aliasing -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -DNDEBUG -fno-inline -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=softfp -marm -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -I/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/usr/include/python2.7 -c OpenSSL/crypto/crypto.c -o build/temp.linux-x86_64-2.7/OpenSSL/crypto/crypto.o
OpenSSL/crypto/crypto.c: In function 'crypto_sign':
OpenSSL/crypto/crypto.c:626:16: error: storage size of 'md_ctx' isn't known
     EVP_MD_CTX md_ctx;
                ^
OpenSSL/crypto/crypto.c: In function 'crypto_verify':
OpenSSL/crypto/crypto.c:673:16: error: storage size of 'md_ctx' isn't known
     EVP_MD_CTX md_ctx;
                ^
error: command '/home/freetz/7520/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc' failed with exit status 1
make: *** [make/python-pyopenssl/python-pyopenssl.mk:18: packages/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/python-pyopenssl-0.13.1/root/usr/lib/python2.7/site-packages/OpenSSL/SSL.so] Error 1

fda77, 2021-03-21:
Wenn du nach dem Fehler googelst: https://stackoverflow.com/questions/42662733/evp-md-ctx-error-storage-size-of-ctx-isn-t-known/42666983

Es sieht so aus dass pyopenssl noch keine openssl 1.1 kennt, sondenr nur das seit 2 Jahren veraltete und unsupportete 1.0
Am besten machst du einen Bump davon. Wie: https://freetz-ng.github.io/freetz-ng/SUPPORT.html#package-bump vielleicht klappt das so

https://pypi.org/project/pyOpenSSL/0.14/
```
High-level wrapper around a subset of the OpenSSL library, includes
* SSL.Connection objects, wrapping the methods of Python's portable
sockets
```

Oder gleich https://pypi.org/project/pyOpenSSL/20.0.1/

forenuser, 2021-03-21:
Mit pyOpenSSL/0.14 bzw. 0.15.1 erhalte ich folgenden Fehler:
```
  File "./setup.py", line 11, in <module>
    from setuptools import setup
ImportError: No module named setuptools
```

fda77, 2021-03-22:
Dann ist beim bump was nicht okay. Evtl wird im falschen Pfad gesucht oder die Datei heisst nun anders

forenuser, 2021-03-22:
Die Zeile "setuptools import setup" ist neu in setup.py.
Für ein Hostcompiler wird empfohlen, folgendes Paket zu installieren.
sudo apt-get install -y python-setuptools
Alternativ: python -m pip install -U pip setuptools
Ich nehme an, dass setuptools für die Crosstoolchain bereitgestellt werden muss, weiß allerdings nicht, wie man das macht.

fda77, 2021-03-23:
Was passiert wenn du es tatsächlich installierst? Kann gut sein dass es nur auf dem host gebraucht wird

forenuser, 2021-03-23:
Nach Installation auf dem host ist das Verhalten unverändert.

fda77, 2021-03-24:
Bis es gebumpt ist...
Ausser in der Überschrift tauch übrigends "pycurl" nicht auf

closed 2021-03-26 by fda77