Freetz-NG/freetz-ng #274: acme.sh - missing parameter for od

flosch-dev, 2021-04-17:
acme.sh ruft in der hex_dump function "od" mit dem parameter "-A" auf. Leider fehlt der aber im aktuellen busybox applet.
https://github.com/acmesh-official/acme.sh/blob/edd46eb3d1bb3698b80d9beba79ba1be0c9b59e4/acme.sh#L633

```
[Sat Apr 17 17:40:42 CEST 2021] kSecret='[hidden](please add '--output-insecure' to see this value)'
od: invalid option -- A
BusyBox v1.32.1 (2021-04-12 21:51:35 CEST) multi-call binary.

Usage: od [-aBbcDdeFfHhIiLlOovXx] [FILE]

Print FILE (or stdin) unambiguously, as octal bytes by default
[Sat Apr 17 17:40:42 CEST 2021] kSecretH='[hidden](please add '--output-insecure' to see this value)'
Usage: _hmac hashalg secret [outputhex]
[Sat Apr 17 17:40:42 CEST 2021] kDateH
Usage: _hmac hashalg secret [outputhex]
[Sat Apr 17 17:40:42 CEST 2021] kRegionH
Usage: _hmac hashalg secret [outputhex]
[Sat Apr 17 17:40:42 CEST 2021] kServiceH
Usage: _hmac hashalg secret [outputhex]
[Sat Apr 17 17:40:42 CEST 2021] kSigningH
````

Ich benutze acme.sh schon eine Weile, auch ohne Freetz-NG und ich weiß, dass ich im Januar noch ein Cert bekam. Dieses läuft leider in 5 Tagen aus und dabei fiel mir das auf.
Im Januar hatte ich noch freetz mit 7.21 drauf. Jetzt habe ich 7.25 vom gestrigen freetz-ng master branch.
Kann man da noch was am busybox od machen oder brauchen wir od als zusätzliches packages?

berndy2001, 2021-04-17:
Wann wird die hexdump Funktion eigentlich verwendet? Habe mir acme.sh erst vor paar Tagen mit `--staging` angesehen, da gabs keine Probleme, aber vermutlich liegt das an den verwendeten Optionen.

Meine waren: `acme.sh --staging --issue --dns dns_ovh -d xxx.yy -d *.xxx.yy --home /var/media/ftp/.acme.sh`

flosch-dev, 2021-04-17:
Also ich lasse acme.sh so laufen:
`acme.sh --issue \
-d xxx.de -d *.xxx.de \
--yes-I-know-dns-manual-mode-enough-go-ahead-please \
--dns dns_aws \
--home /var/media/ftp/acme-sh/`

Habe es gerade mal mit --staging probiert, das ändert bei mir aber nichts und ich kriege weiterhin gleichen Fehler.
Bist du sicher dass bei dir ein neues cert ausgestellt wurde obwohl der parameter bei od fehlt? Ich sehe dass die function "_hex_dump" mehrmals benutzt wird.
Habe meinen Befehl gerade mal im acme-sh docker getestet und da läuft es ohne probleme durch.
Ich denke mal das schlägt bei mir wegen dem od Parameter fehl.

berndy2001, 2021-04-17:
Ja, habs gerade nochmals getestet. 

Hab in der config.in gefunden, dass "Desktop" deaktiviert ist: https://github.com/Freetz-NG/freetz-ng/blob/master/make/busybox/Config.in.busybox.1_32#L19
Kenne die Hintergründe dazu nicht.

fda77, 2021-04-19:
Hier steht ein Kommentar weshlab das deaktiviert wurde. 
https://github.com/Freetz-NG/freetz-ng/blob/master/make/busybox/generate.sh#L148
Ob das noch zutrifft weiss ich nicht

flosch-dev, 2021-04-19:
Mich wundert dass (ob) ich der Einzige mit dem Problem bei acme.sh bin, das könnte dann eventuell am dns provider (aws) liegen.
Der Kommentar liest sich, als sollte man Desktop nicht aktivieren. Vlt probiere ich damit mal die Tage rum, aber wer weiß schon was das für Auswirkungen für alle anderen (nicht meine FB 7490) hat.
Ich frage mich aber, wieso es vor Jan 2021 bei mir noch funktionierte? Da war ich noch auf dem original freetz unterwegs. Da lief acme.sh ohne Probleme, mehrere Monate auf jeden Fall. Ich hatte es in /var/media/ftp/acme-sh abgelegt.
Gibt es "od" auch als standalone binary welches man als package einbinden könnte?

fda77, 2021-04-19:
Das wurde vor 7 Jahren hinzugefügt: https://github.com/Freetz-NG/freetz-ng/commit/035fe7876671d2338f5a49eceabd16b1d7782e74

flosch-dev, 2021-04-19:
Das hatte ich mit git blame auch bemerkt und deshalb gewundert wieso es bei mir "anscheinend" im Jan 2021 noch ging. Ich hoffe mal ich erzähle da kein mist, aber ich meine dass lief alles schonmal bei mir auf der FB.

flosch-dev, 2021-04-19:
Ich hatte bisher 2.8.7 benutzt und auch gleiches Problem bemerkt. 
Der parameter "-A" ist aber auch schon einige Jahre drin: [https://github.com/acmesh-official/acme.sh/commit/4e4a6d83973fa0580ef8b2f1d7d0312fdd671543](https://github.com/acmesh-official/acme.sh/commit/4e4a6d83973fa0580ef8b2f1d7d0312fdd671543)
Keine Ahnung, bin momentan ratlos.

berndy2001, 2021-04-19:
> Das wurde vor 7 Jahren hinzugefügt: [035fe78](https://github.com/Freetz-NG/freetz-ng/commit/035fe7876671d2338f5a49eceabd16b1d7782e74)

Hat scheinbar was mit freetzmount zu tun: http://web.archive.org/web/20141024142424/http://freetz.org/ticket/2572

fda77, 2021-04-20:
Auf Boxmatrix gibt es ein Backup des kompletten Tracks: https://trac.boxmatrix.info/freetz-ng/ticket/2572
Der Commit war in 1 Ticket das sich über 500 Commits erstreckte
@flosch-dev: Vielleicht hat dein Provider etwas aktualisiert und acme nutze andere Funktionen

flosch-dev, 2021-04-20:
Ich habe gesehen, dass "od" zu coreutils gehört. https://github.com/coreutils/coreutils
Weiß nicht, ob ich der Aufgabe gewachsen bin, aber ich gucke mal ob man daraus ein package für freetz bauen kann. Am besten mit Auswahl der Binaries.

fda77, 2021-04-20:
Am besten fängt man damit an zu schauen welche Libs für ein Package gebraucht werden. Die müssen dann evtl auch noch eingebaut werden. Oder man deaktviert manches um weniger Libs zu brauchen

fda77, 2021-04-22:
https://github.com/Freetz-NG/freetz-ng/issues/274#issuecomment-822510307 ?

fda77, 2021-05-01:
Ich habe keinen Grund gefunden weshalb die ps-Optionen immer vorhanden sein müssen, diese werden nur von 3 Packages benötigt. Was so wie es jetzt ist nicht geht: acme + openvpn-cgiV1 im gleichen Image.
Falls das jemand unbedingt nutzen möchte muss dieser das "ps|grep" hier anpassen:  https://github.com/Freetz-NG/freetz-ng/blob/master/make/openvpn-cgi/files/root/etc/init.d/rc.openvpn#L8

closed 2021-05-01 by fda77

flosch-dev, 2021-05-01:
Danke!
acme.sh läuft nun, habe gerade ein neues Cert enrolled.