Freetz-NG/freetz-ng #276: FREETZ_AVMDAEMON_DISABLE_DNS & FREETZ_AVMDAEMON_DISABLE_DHCP erscheinen nicht bei der 6590

Conan179, 2021-04-18:
Ich wollte ein image für meine 6590 bauen, wo die dns und dhcp funktion des multid's ausgeschalten sind.
Die 2 patches sind laut https://github.com/Freetz-NG/freetz-ng/blob/master/config/ui/patches.in noch vorhanden, aber sie werden mir nicht angezeigt
[image]
[image]

fda77, 2021-04-19:
Da steht auch "depends on FREETZ_AVM_HAS_AVMDAEMON_PRELOAD"
Mit den neueren FOS gibt es keine libctlmgt+libmultid wegen uclibc mehr.  Daher auch keine Optionen dazu.
Als IP Client sollte AVM keine DNS+DHCP starten (ausnahmen: ein Fehler), da man dies bei der Netzwerkconfigugarion angeben muss

closed 2021-04-19 by fda77

Conan179, 2021-04-19:
Danke für die aufklärung, darauf wäre ich nicht gekommen. Ich kann zwar ein linux per terminal bedinen, aber ein profi bin ich in dem Thema leider nicht -.-
Yay, ich habs mal wieder geschaft nen fehler zu produzieren.