Freetz-NG/freetz-ng #277: 6490 7.25 - OpenVPN startet nicht

6490dk, 2021-04-19:
2021-04-18 12:55:40 Cipher negotiation is disabled since neither P2MP client nor server mode is enabled
2021-04-18 12:55:40 OpenVPN 2.5.1 i686-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [AEAD] built on Apr 5 2021
2021-04-18 12:55:40 library versions: OpenSSL 1.1.1k 25 Mar 2021, LZO 2.10
2021-04-18 12:55:40 WARNING: INSECURE cipher (BF-CBC) with block size less than 128 bit (64 bit). This allows attacks like SWEET32. Mitigate by using a --cipher with a larger block size (e.g. AES-256-CBC). Support for these insecure ciphers will be removed in OpenVPN 2.6.
2021-04-18 12:55:40 Outgoing Static Key Encryption: Cipher 'BF-CBC' initialized with 128 bit key
2021-04-18 12:55:40 WARNING: INSECURE cipher (BF-CBC) with block size less than 128 bit (64 bit). This allows attacks like SWEET32. Mitigate by using a --cipher with a larger block size (e.g. AES-256-CBC). Support for these insecure ciphers will be removed in OpenVPN 2.6.
2021-04-18 12:55:40 Outgoing Static Key Encryption: Using 160 bit message hash 'SHA1' for HMAC authentication
2021-04-18 12:55:40 Incoming Static Key Encryption: Cipher 'BF-CBC' initialized with 128 bit key
2021-04-18 12:55:40 WARNING: INSECURE cipher (BF-CBC) with block size less than 128 bit (64 bit). This allows attacks like SWEET32. Mitigate by using a --cipher with a larger block size (e.g. AES-256-CBC). Support for these insecure ciphers will be removed in OpenVPN 2.6.
2021-04-18 12:55:40 Incoming Static Key Encryption: Using 160 bit message hash 'SHA1' for HMAC authentication
2021-04-18 12:55:40 WARNING: normally if you use --mssfix and/or --fragment, you should also set --tun-mtu 1500 (currently it is 1300)
2021-04-18 12:55:41 TUN/TAP device tun0 opened
2021-04-18 12:55:41 /sbin/ip link set dev tun0 up mtu 1300
2021-04-18 12:55:41 /sbin/ip link set dev tun0 up
2021-04-18 12:55:41 /sbin/ip addr add dev tun0 local 192.168.20.1 peer 192.168.10.1
2021-04-18 12:55:41 /sbin/ip route add 192.168.10.0/24 via 192.168.10.1
2021-04-18 12:55:41 TCP/UDP: Preserving recently used remote address: [AF_INET6]xxxx:xx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:1194
2021-04-18 12:55:41 Socket Buffers: R=[176128->176128] S=[176128->176128]
2021-04-18 12:55:41 UDP link local: (not bound)
2021-04-18 12:55:41 UDP link remote: [AF_INET6]xxxx:xx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:1194
2021-04-18 12:55:41 chroot to '/var/tmp/openvpn' and cd to '/' succeeded
2021-04-18 12:55:41 GID set to openvpn
2021-04-18 12:55:41 UID set to openvpn
2021-04-18 12:55:41 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-04-18 12:55:41 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-04-18 12:55:41 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-04-18 12:55:41 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-04-18 12:55:41 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-04-18 12:55:41 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-04-18 12:55:41 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-04-18 12:55:41 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-04-18 12:55:41 RAND_bytes() failed
2021-04-18 12:55:41 Assertion failed at crypto.c:1773
2021-04-18 12:55:41 Exiting due to fatal error
2021-04-18 12:55:41 /sbin/ip route del 192.168.10.0/24
2021-04-18 12:55:41 ERROR: Linux route delete command failed: could not execute external program
2021-04-18 12:55:41 Closing TUN/TAP interface
2021-04-18 12:55:41 /sbin/ip addr del dev tun0 local 192.168.20.1 peer 192.168.10.1
2021-04-18 12:55:41 Linux ip addr del failed: could not execute external program

Conan179, 2021-04-19:
Bitte auch deine .config anhängen.

fda77, 2021-04-19:
Miste dein Konfiguration mal aus. Du nutzt alte Optionen und willst Programme aufrufen die nicht installiert sind
Oder die Kernelsourcen passen nicht mehr -> https://github.com/Freetz-NG/freetz-ng/discussions/134

closed 2021-04-19 by fda77