Freetz-NG/freetz-ng #279: gcc compile fails due to sys/sdt.h

ztNFny, 2021-04-21:
Compiling gcc-5.5.0 fails with below error.

System Info:
      Operating System: Fedora 33 (KDE Plasma)
      CPE OS Name: cpe:/o:fedoraproject:fedora:33
      Kernel: Linux 5.11.14-200.fc33.x86_64
      Architecture: x86-64

All dependencies mentioned in PREREQUISITES.md are installed:
Package rsync-3.2.3-3.fc33.x86_64 is already installed.
Package kmod-27-3.fc33.x86_64 is already installed.
Package execstack-0.5.0-19.fc33.x86_64 is already installed.
Package sqlite-3.34.1-1.fc33.i686 is already installed.
Package sqlite-devel-3.34.1-1.fc33.x86_64 is already installed.
Package libzstd-devel-1.4.9-1.fc33.i686 is already installed.
Package cmake-3.19.7-1.fc33.x86_64 is already installed.
Package zlib-devel-1.2.11-23.fc33.i686 is already installed.
Package libstdc++-devel-10.2.1-9.fc33.i686 is already installed.
Package openssl-1:1.1.1k-1.fc33.x86_64 is already installed.
Package xz-5.2.5-4.fc33.x86_64 is already installed.
Package bc-1.07.1-11.fc33.x86_64 is already installed.
Package unar-1.10.1-20.fc33.x86_64 is already installed.
Package inkscape-1.0.2-1.fc33.x86_64 is already installed.
Package ImageMagick-1:6.9.11.27-1.fc33.x86_64 is already installed.
Package subversion-1.14.1-1.fc33.x86_64 is already installed.
Package ccache-4.2-1.fc33.x86_64 is already installed.
Package gcc-10.2.1-9.fc33.x86_64 is already installed.
Package gcc-c++-10.2.1-9.fc33.x86_64 is already installed.
Package binutils-2.35-18.fc33.x86_64 is already installed.
Package autoconf-2.69-34.fc33.noarch is already installed.
Package automake-1.16.2-2.fc33.noarch is already installed.
Package libtool-2.4.6-36.fc33.x86_64 is already installed.
Package make-1:4.3-2.fc33.x86_64 is already installed.
Package bzip2-1.0.8-4.fc33.x86_64 is already installed.
Package ncurses-devel-6.2-3.20200222.fc33.x86_64 is already installed.
Package ncurses-devel-6.2-3.20200222.fc33.i686 is already installed.
Package ncurses-term-6.2-3.20200222.fc33.noarch is already installed.
Package zlib-devel-1.2.11-23.fc33.x86_64 is already installed.
Package zlib-devel-1.2.11-23.fc33.i686 is already installed.
Package flex-2.6.4-5.fc33.x86_64 is already installed.
Package bison-3.6.4-3.fc33.x86_64 is already installed.
Package patch-2.7.6-13.fc33.x86_64 is already installed.
Package texinfo-6.7-8.fc33.x86_64 is already installed.
Package gettext-0.21-3.fc33.x86_64 is already installed.
Package pkgconf-pkg-config-1.7.3-5.fc33.x86_64 is already installed.
Package ecj-1:4.19-1.fc33.noarch is already installed.
Package perl-4:5.32.1-469.fc33.x86_64 is already installed.
Package perl-String-CRC32-1.8-5.fc33.x86_64 is already installed.
Package wget-1.21.1-2.fc33.x86_64 is already installed.
Package glib2-devel-2.66.8-1.fc33.x86_64 is already installed.
Package git-2.30.2-1.fc33.x86_64 is already installed.
Package libacl-devel-2.2.53-9.fc33.x86_64 is already installed.
Package libattr-devel-2.4.48-10.fc33.x86_64 is already installed.
Package libcap-devel-2.48-2.fc33.x86_64 is already installed.
Package ncurses-devel-6.2-3.20200222.fc33.i686 is already installed.
Package glibc-devel-2.32-4.fc33.i686 is already installed.
Package libgcc-10.2.1-9.fc33.i686 is already installed.
Package ncurses-devel-6.2-3.20200222.fc33.i686 is already installed.
Package glibc-devel-2.32-4.fc33.i686 is already installed.
Package libgcc-10.2.1-9.fc33.i686 is already installed.

According to the interwebs sys/sdt.h should be provided by systemtap-sdt-devel which is also installed:
Package systemtap-sdt-devel-4.4-4.fc33.x86_64 is already installed.

End of compile log:
mips-linux-uclibc-cc   -g -O2 -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -minterlink-mips16 -O2  -g -O2 -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -minterlink-mips16 -DIN_GCC    -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include  -fPIC -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector  -fPIC -I. -I. -I../.././gcc -I/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc -I/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/. -I/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/../gcc -I/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/../include  -DHAVE_CC_TLS  -o unwind-dw2.o -MT unwind-dw2.o -MD -MP -MF unwind-dw2.dep -fexceptions -c /tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/unwind-dw2.c -fvisibility=hidden -DHIDE_EXPORTS
/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/unwind-dw2.c:41:21: fatal error: sys/sdt.h: No such file or directory
compilation terminated.
make[3]: *** [/tmp/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/shared-object.mk:14: unwind-dw2.o] Error 1

fda77, 2021-04-22:
install the missing file?

ztNFny, 2021-04-22:
> install the missing file?

It's already there as mentioned in my OP: "/usr/include/sys/sdt.h exists:
c5d4c82b2a0f3891ddf8a0484a882458 /usr/include/sys/sdt.h"

fda77, 2021-04-22:
Sry, i cant test  it as my fedroa 34beat cant build gcc5 anymore:
```
error: use of an operand of type 'bool' in 'operator++' is forbidden in C++17
```

ztNFny, 2021-06-06:
@fda77: I have the original error again with b413fb41:
/freetz-ng/source/toolchain-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/gcc-5.5.0/libgcc/unwind-dw2.c:41:21: fatal error: sys/sdt.h: No such file or directory

sdt.h exists at /usr/include/sys/sdt.h

Fedora 34, KDE, 5.11.19-300.fc34.x86_64

If that file needs to be in some specific location to be used please advice where I need to copy it.

closed 2021-06-07 by fda77