Freetz-NG/freetz-ng #294: Error in building lighttpd for 7170

hucky58, 2021-05-14:
`/media/user/d72b2992-9829-4d2b-b93b-92b730ba066f/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc -std=gnu99 -DHAVE_CONFIG_H -DHAVE_VERSIONSTAMP_H -DLIBRARY_DIR="\"/usr/lib/lighttpd\"" -DSBIN_DIR="\"/usr/sbin\"" -I. -I..    -D_REENTRANT -I/media/user/d72b2992-9829-4d2b-b93b-92b730ba066f/freetz-ng/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/usr/include -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGE_FILES    -march=4kc -mtune=4kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -W -Wshadow -pedantic -MT lighttpd-stat_cache.o -MD -MP -MF .deps/lighttpd-stat_cache.Tpo -c -o lighttpd-stat_cache.o `test -f 'stat_cache.c' || echo './'`stat_cache.c
stat_cache.c: In function 'stat_cache_init_fam':
stat_cache.c:577:2: warning: implicit declaration of function 'inotify_init1' [-Wimplicit-function-declaration]
stat_cache.c:577:26: error: 'IN_NONBLOCK' undeclared (first use in this function)
stat_cache.c:577:26: note: each undeclared identifier is reported only once for each function it appears in
stat_cache.c:577:38: error: 'IN_CLOEXEC' undeclared (first use in this function)
stat_cache.c: In function 'fam_dir_monitor':
stat_cache.c:720:25: error: 'IN_EXCL_UNLINK' undeclared (first use in this function)
make[4]: *** [Makefile:3326: lighttpd-stat_cache.o] Fehler 1
make[4]: *** Auf noch nicht beendete Prozesse wird gewartet …
mv -f .deps/lighttpd-gw_backend.Tpo .deps/lighttpd-gw_backend.Po
make[3]: *** [Makefile:1919: all] Fehler 2
make[2]: *** [Makefile:467: all-recursive] Fehler 1
make[1]: *** [Makefile:397: all] Fehler 2
make[1]: Verzeichnis „/media/user/d72b2992-9829-4d2b-b93b-92b730ba066f/freetz-ng/source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/lighttpd-1.4.59“ wird verlassen

ERROR: Build failed.
make: *** [make/lighttpd/lighttpd.mk:125: source/target-mipsel_gcc-4.6.4_uClibc-0.9.29/lighttpd-1.4.59/src/lighttpd] Fehler 1
`

fda77, 2021-05-14:
The Kernel, and/or libc of the 7170 is to old. Which was the last working lighttpd version?
You could try to disable inotify related thing wiht configure options

hucky58, 2021-05-15:
Same problem for the 7240/7290 ...
So the lighttpd package should be disabled in default settings while selecting these boxes?

fda77, 2021-05-16:
The Kernel, and/or libc of the 7270 is to old. Which was the last working lighttpd version?

gstrauss, 2021-06-11:
lighttpd developer here.

> Which was the last working lighttpd version?

lighttpd 1.4.56 added support for inotify, so I guess you want to fall back to lighttpd 1.4.55 until this issue is resolved in your build.  (see below for suggested fix)  However, there are some bugs in lighttpd 1.4.55, so the latest stable version is recommended: lighttpd 1.4.59

According to `man inotify_init`:
> inotify_init() first appeared in Linux 2.6.13; library support was added to glibc in version 2.4.  inotify_init1() was added in Linux 2.6.27; library support was added to glibc in version 2.9.

According to `man 7 inotify`:
> IN_EXCL_UNLINK (since Linux 2.6.36)

If the system does not support inotify, then the build should not detect the existence of `<sys/inotify.h>`.  If you force the build to undefine `HAVE_SYS_INOTIFY_H`, then the build should proceed.  (config.h defines HAVE_SYS_INOTIFY_H if it finds `<sys/inotify.h>`)

**The solution to this issue is probably a simple patch to lighttpd configure.ac to remove the line which looks for `sys/inotify.h` around line 109.**

According to https://uclibc.org/, uClibc has supported inotify since at least uClibc 0.9.33.1 (11 April 2012), where the changelog notes a change to `inotify_rm_watch` prototype.  uClibc probably supports inotify in earlier versions, but I did not track it further.  2012 is a *long* time ago.  2007 is even longer ago and more ancient!  6 May 2007 is when uClibc 0.9.29 was released.  Why is Freetz-NG using such an ancient version of uClibc?

If someone can provide me with more details on why `<sys/inotify.h>` is found, but the inotify prototypes and definitions are missing, I can add a patch to a future version of lighttpd which detects this better.  Is `<sys/inotify.h>` a stub file?

fda77, 2021-06-11:
During cross-compiled some features of your HOST system could be found, and not the TARGET system. Then addition patches/parameters/etc are needed. In configure of lighttp's source search for "inotify".

make/lighttpd/lighttpd.mk
```
#ifeq ($(strip $(???)),y)
$(PKG)_CONFIGURE_ENV += ac_cv_header_sys_inotify_h=no
#endif
```
configure of ``make lighttpd-recompiled` shows then  `checking for sys/inotify.h... (cached) no

> Why is Freetz-NG using such an ancient version of uClibc?
The uclibc auf AVM is REPLACED by freetz (with some more needed features), so changing the version will break all AVM closed-source binaries. On newer devices (with more flash) freetz uses its own (additional) uclibc, newer than the version of avm
The same for the kernel version

closed 2021-06-11 by fda77