Freetz-NG/freetz-ng #299: 6590: OpenVPN - Exiting due to fatal error

WXbet, 2021-05-16:
OpenVPN startet nicht auf einer 6590. Es treten diverse OpenSSL-Fehler auf:
```
2021-05-16 12:26:43 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-05-16 12:26:43 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-05-16 12:26:43 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-05-16 12:26:43 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-05-16 12:26:43 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-05-16 12:26:43 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-05-16 12:26:43 OpenSSL: error:2406C06E:lib(36):func(108):reason(110)
2021-05-16 12:26:43 OpenSSL: error:2406B072:lib(36):func(107):reason(114)
2021-05-16 12:26:43 RAND_bytes() failed
2021-05-16 12:26:43 Assertion failed at crypto.c:1773
2021-05-16 12:26:43 Exiting due to fatal error
```
Anbei das debug log von OpenVPN und die config.
- [openvpn_debug.txt](https://github.com/Freetz-NG/freetz-ng/files/6489055/openvpn_debug.txt)
- [config.txt](https://github.com/Freetz-NG/freetz-ng/files/6489056/config.txt)

Vielleicht jemand eine Idee, was zu dem Problem führt.

WXbet, 2021-05-16:
Die Hinweise ab hier https://www.ip-phone-forum.de/threads/6490-openvpn-startet-nicht-7-25.310107/post-2430388 führen zu einem Start des OpenVPN-Service auf einem Puma-Device. Nach Ausführung des Befehls
`mknod -m 444 /tmp/openvpn/dev/random c 1 8 2>/dev/null`
startet OpenVPN ordnungsgemäß.

Anbei das entsprechende debug log von OpenVPN der 6590.
- [openvpn_debug_6590.txt](https://github.com/Freetz-NG/freetz-ng/files/6489331/openvpn_debug_6590.txt)

Macht es Sinn die o.g. Zeile einfach hier https://github.com/Freetz-NG/freetz-ng/blob/19d2d42b9ce3cecafae9f6fb3a416b5f987894a3/make/openvpn-cgi/files/root/etc/init.d/rc.openvpn#L27 hinzuzufügen?

fda77, 2021-05-16:
Danke. Testen kann ich das nicht da ich keine Puma Fritzbox hab

closed 2021-05-16 by fda77