Freetz-NG/freetz-ng #301: 7581/7582: wrong compiler options

fda77, 2021-05-17:
Follow up of https://github.com/Freetz-NG/freetz-ng/issues/289

mikiantena, 2021-05-17:
@PeterPawn could you please clarify what you meant by:

> I'd like to point out again, that the (hopefully) correct compiler settings may be found at the (above) mentioned location. After adopting these settings for the appropriate models, a make call reveals, how a compiler command looks like - if someone shows the log-file of this attempt. This is in my opinion the only serious option to check, whether the used cross-compiler environment looks good or odd.

I missed this sentence completely previously.
Would it help if I provided logs of running make? If so I could provide the logs after running freshly downloaded freetz-ng sources, as well as running it after applying @fda77 patch for disabling neon. Or you want me to provide a snippet such as
```
PATH="/home/freetz/freetz-ng/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/arm-linux-uclibcgnueabi/bin:/home/freetz/freetz-ng/toolchain/build/arm_gcc-5.5.0/arm-unknown-linux-gnueabi/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/snap/bin" CC="gcc" CFLAGS="-D_GNU_SOURCE -fno-stack-protector" CXXFLAGS="-D_GNU_SOURCE -fno-stack-protector" CFLAGS_FOR_TARGET="-mcpu=cortex-a9 -mfloat-abi=soft -marm -Os -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/var/media/temp/usr/lib/freetz/ld-uClibc.so.1" CXXFLAGS_FOR_TARGET="-mcpu=cortex-a9 -mfloat-abi=soft -marm -Os -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/var/media/temp/usr/lib/freetz/ld-uClibc.so.1" gcc_cv_libc_provides_ssp=yes \
        /home/freetz/freetz-ng/source/toolchain-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/gcc-5.5.0/configure \
        --prefix=/home/freetz/freetz-ng/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/arm-linux-uclibcgnueabi/usr \
        --with-sysroot=/home/freetz/freetz-ng/source/toolchain-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/uClibc_dev/ \
        --with-newlib=yes \
        --with-headers=no \
        --build=x86_64-pc-linux-gnu \
        --host=x86_64-pc-linux-gnu \
        --target=arm-linux-uclibcgnueabi \
        --enable-languages=c \
        --disable-shared \
        --disable-threads \
        --with-gmp=/home/freetz/freetz-ng/tools/build \
        --with-mpfr=/home/freetz/freetz-ng/tools/build \
        --with-mpc=/home/freetz/freetz-ng/tools/build \
        --with-isl=no \
        --enable-option-checking --with-gnu-as --with-gnu-ld --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-libsanitizer --disable-libssp --disable-multilib --enable-tls --disable-fixed-point --with-cpu=cortex-a9 --with-abi=aapcs-linux --with-float=soft --with-mode=arm --disable-nls  --disable-decimal-float
```

The above example is from me trying to disable vfp completely, but I could re-run with the above mentioned settings later.

Best regards.

mikiantena, 2021-05-17:
Also as far as this is concerned:

> Check thoroughly, whether BusyBox and uClibc are really depending on libgcc.so - if it's true, you HAVE to rebuild the whole toolchain from scratch, because code in libgcc is platform-dependent 

and
```
file */bin/busybox|grep ARM|sort -k2
FRITZ.Box_7581-07.14/bin/busybox:                                      ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.4.0, stripped
FRITZ.Box_7582-07.14/bin/busybox:                                      ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.4.0, stripped
```

I get this for `file busybox`
```
freetz@freetz-linux:~/freetz-ng/build/modified/filesystem/bin$ file busybox 
busybox: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /var/media/temp/usr/lib/freetz/ld-uClibc.so.1, stripped
```

Ignoring the FREETZ_RPATH difference, interpreter is quoted as uClibc. (As it was for all my compiling attempts).

PeterPawn, 2021-05-17:
As written above, the original Broadcomm settings for compilations targeting BCM63138 may be found in `make.common` from open-source packages provided by AVM.

Following the steps in this make instructions, something like this can be found as compiler settings:

```
ifeq ($(ARCH),arm)
BRCM_COMMON_CFLAGS := -Os -march=armv7-a -fomit-frame-pointer -mno-thumb-interwork -mabi=aapcs-linux -marm -fno-common -ffixed-r8 -msoft-float -D__ARM_ARCH_7A__ $(BRCM_WERROR_CFLAGS) -DFRITZ_OS
#for mips' -mno-shared (do not generate posistion independent code), arm is -mno-apcs-reentrant which is the default
ifeq ($(CMS_MEM_LEAK_TRACING),y)
# These extra flags are needed by ARM stack backtrace function
BRCM_COMMON_CFLAGS += -fasynchronous-unwind-tables -rdynamic
endif
export BRCM_APP_CFLAGS :=  $(BRCM_COMMON_CFLAGS)
endif  # ARCH eq arm
```

Looks like the settings to compile applications (BCRM_APP_FLAGS due to my own interpretation) are really different from the settings used by Freetz-NG (according to a make log snippet provided at the older/other issue) - the additional `-DFRITZ_OS` at the end of the line makes this (imho) the best candidate for options used by AVM, too.

It seems, that the used settings are far more general, than the settings from current Freetz-NG checkout - in my opinion the best guess/hint, how to change the settings used by Freetz-NG to generate code runnable by FRITZ!OS.

And it's normal and expectable, that a BusyBox compiled by Freetz-NG uses `uClibc` instead of `glibc` - this should not matter in any case, as long as the correct libraries may get located and loaded. But this is only the next step ... now you should try to compile binaries runnable with stock firmware first.

mikiantena, 2021-06-07:
Hello guys,

I've been working on this whenever I have some spare time. At this point I feel like I'm stuck.

I tried getting as close as possible to AVM kernel CFLAGS, and I've paid special attention to busybox lines in my make log. At this point I made over 20 complete builds in order to try and get usable busybox. So, for example, here are the CFLAGS for busybox:

```
cmd() { PATH="/home/freetz/freetz-ng-7582/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/arm-linux-uclibcgnueabi/bin:/home/freetz/freetz-ng-7582/toolchain/build/arm_gcc-5.5.0/arm-unknown-linux-gnueabi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/snap/bin" LD_RUN_PATH="/var/media/temp/usr/lib/freetz" FREETZ_LIBRARY_DIR="/var/media/temp/usr/lib/freetz" make -j2  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };        if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/busybox-1.33.1 \
        CC="/home/freetz/freetz-ng-7582/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-gcc" CROSS_COMPILE="/home/freetz/freetz-ng-7582/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/arm-linux-uclibcgnueabi/bin/arm-linux-uclibcgnueabi-" 
EXTRA_CFLAGS="-mcpu=cortex-a9 -mfloat-abi=soft -marm -Os -pipe -ffixed-r8 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/var/media/temp/usr/lib/freetz/ld-uClibc.so.1" ARCH="arm"
ESC[48;5;56mbuildingESC[49m ... make[1]: 
Entering directory '/home/freetz/freetz-ng-7582/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.1/busybox-1.33.1'
```

As you can see, I tried excluding fpu completely by using `-mfloat-abi=soft` (which I read is the same as `-msoft-float`). I included `-ffixed-r8`, which as I've read in a few threads is pretty much standard for arm embedded devices in the industry (didn't get why not using that register is important, though).

At one point I even built it without `mcpu=cortex-a9`, but that didn't change anything.

Still the behavior I get is the same as before. busybox worked around half the time, which makes me believe it's the fpu being included only in the first core is the main culprit, as discussed earlier. Not sure how that comes together with using soft float, but it does exhibit that sporadic behavior. Busybox applets work, when they work. Like I can use bash, mkdir, rmdir, ip, and several other busybox applets that I tried (sporadically, of course).

ffixed-r8 cflag seems to make it more stable (not sure if this is only my impression or a fact)

I'm stuck right now as I can't build ltrace and/or strace. I'm trying to get further details on that illegal undefined instruction that is plaguing me around half the time when running busybox. Please see attached log file for details on the error.
[freetz_ltrace_error.log](https://github.com/Freetz-NG/freetz-ng/files/6610885/freetz_ltrace_error.log)

Right now I continue and try to modify the CFLAGS further, with `-fno-common -fomit-frame-pointer`, but I'm not too hopeful. I'm also reading AVM sources extensively to find any further hints, if there are any.

**Question 1**: Please help with strace/ltrace building if you are able to.

**Question 2**: In my `-mfloat-abi=soft`  make log, I find references to `-mfpu=vfp`, and those lines seem to completely skip the CFLAGS set by freetz. Is this normal? 
[freetz_mfpu_vfp.log](https://github.com/Freetz-NG/freetz-ng/files/6610940/freetz_mfpu_vfp.log)

**Question 3**: Is there a way to skip the whole CFLAGS config of freetz, and make it use my custom CFLAGS, as that would give me more granular control and maybe help in finding the proper build flags?

Best regards

mikiantena, 2021-06-07:
Here some more evidence that it's a core/fpu problem. busybox dmesg show multiple CPU 1 problems, most probably as a result of me running busybox, and getting illegal instruction output:
[dmesg.log](https://github.com/Freetz-NG/freetz-ng/files/6611122/dmesg.log)

fda77, 2021-06-07:
3) maybe in the file https://github.com/Freetz-NG/freetz-ng/blob/master/toolchain/make/toolchain-common.in

fda77, 2022-02-22:
Thats great! I've changed uclibc.mk part a little to keep other things like before.
Can you attach an screenshot of "avm-info"? What runs on port 81? Could this be accessed by lan?
I could add automatic changing of :81 depending on FREETZ_SYSTEM_TYPE_BCM63138
The precompiled toolchains should also be updated.

roland3939, 2022-02-23:
Oh No !!!! Use tabs instead of white spaces  !!!  to respect kconfig syntax 
your commit 0f4fac1
line 101 in file "config/avm/compiler-traits" : why did you modify indent by  white spaces  ???
this results in :

make menuconfig-nocache
config/avm/compiler-traits.in:100: syntax error
config/avm/compiler-traits.in:99: invalid statement
config/avm/compiler-traits.in:100: invalid statement
config/avm/compiler-traits.in:101: invalid statement
config/avm/compiler-traits.in:103: invalid statement
config/avm/compiler-traits.in:104: invalid statement
config/avm/compiler-traits.in:105: invalid statement
config/avm/compiler-traits.in:107: invalid statement
make: *** [Makefile:424: menuconfig-nocache] Fehler 1

fda77, 2022-02-23:
I tested again, for me it works. Is there something in your clone merged wrong by git?
Check by "git status" / "git diff"
If you have no changes you want to keep: https://github.com/Freetz-NG/freetz-ng/blob/master/README.md#delete-local-changes

Can you test if an image build with the precompiled toolchain works?

roland3939, 2022-02-24:
Let's do some feedback and follow up:
first of all, commit 0f4fac1 does NOT  cause any issues; it's fully working
issue reported in my post https://github.com/Freetz-NG/freetz-ng/issues/301#issuecomment-1048711543 was solely due to an incomplete merge in my local repo. Sorry for not checking thoroughly before.

>Can you test if an image build with the precompiled toolchain works?  

 build with  precompiled toolchain did NOT succeed for me:
```
/usr/bin/ld: /usr/lib/libzstd.so: error adding symbols: file in wrong format
```
for reference and and information, i added make log here 
[fmake_2022-02-24_15-59-40.log](https://github.com/Freetz-NG/freetz-ng/files/8134374/fmake_2022-02-24_15-59-40.log)

>Can you attach an screenshot of "avm-info"? What runs on port 81? Could this be accessed by lan?

just a brief reply; please see below:

[image]

```
root@fritz:/var/tmp/root# showportsopentolan | grep smd
[::]:81                              | TCP        | /bin/smd
[::]:443                             | TCP        | /bin/smd
root@fritz:/var/tmp/root# netstat -tupan  | grep smd
tcp        0      0 :::81                   :::*                    LISTEN      900/smd
tcp        2      0 :::443                  :::*                    LISTEN      900/smd
```

roland3939, 2022-02-26:
had a closer look at issue "build with precompiled toolchain failing" :
my host build system is
```
cat /etc/fedora-release; uname --kernel-release
Fedora release 35 (Thirty Five)
5.16.11-200.fc35.x86_64 
```
[prerequisites](https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md) require the use of  **libzstd-devel.i686** ( 32 bit library ! )  which is the root cause for failing to  build with precompiled toolchain here 
using  **libzstd-devel.x86_64** ( 64 bit library ! ) instead, image build  with "precompiled toolchain" works fine !

fda77, 2022-03-07:
Thx, updated. The precompiled from x86 to x64 some months ago

for port 81 problem -> https://github.com/Freetz-NG/freetz-ng/discussions/485

closed 2022-03-07 by fda77