Freetz-NG/freetz-ng #302: 75er Boxen: Keine autostart von SAMBA-Freigaben nach Boxstart oder Reboot

Gismotro, 2021-05-17:
Samba-Freigaben sind erst nach Neustart des Samba-smbd Dienstes wieder nutzbar.

Problem besteht m.E. scjon seit der 7.2x

Bilder und weitere Infos : https://www.ip-phone-forum.de/threads/freetz-ng-samba-share-problem.310372/post-2430635

Was kann / soll ich noch an Infos liefern um das Problem ggf. weiter eingrenzen zu können ?

Conan179, 2021-05-17:
Kan man einbauen, das der samba dienst nach ca. 60sec nach dem boot neustartet? alle @reboot sleep60 sambe restart?

fda77, 2021-05-17:
Vielleicht liegt es daran wie das Image gebaut wurde bzw von wem? Im DEB wird doch empfohlen udevmount rauszuwerfen und stattdessen das veraltete und nicht mehr compatible freetzmount irgendwie wieder "freizuschalten"...
Jedenfalls sollten bei Erkennen neuer Datenträger "udev" hooks ausgeführt werden die Samba aktualisieren. Im Webif bei den mod-logs sollte in de mod_mount.log steht ob zumindest die Hooks ausgeführt werden
Es kann auch sein dass das durch den NCQS (oder wie das genau heisst) Ersatz nicht mehr funktioniert.

Gismotro, 2021-05-18:
Ich betone nochmal : Meine Boxen laufen immer mit original Freetz-NG ohne externe addons (außer WoL, Samba u. VSFTPD)

Gismotro, 2021-05-18:
Dazu kommt das die Freigaben komischerweise schon nicht mehr erreichbar sind wenn man nur Freetz-NG flascht.
Also aktuelle 7520_7.25 Freetz-NG+Wol. Mehr ist im Image nicht aktiviert. Aus irgend einem Grund wird wohl der originale SAMBA_SMBD nicht gestartet ?

fda77, 2021-05-19:
Du kannst dich darüber aufregen soviel du willst .. bevor ich (wieder) ein Problem untersuche das es nicht gibt frag ich vorher nach, zur Not 2x. Ich hab ja nichts von "addon" geschrieben, sondern von eine obskuren Anleitung die das Problem durch herumfummeln in der .config verursacht. Und du bist ja bekannt dafür dass du immer uralte .config mitschleppst wie vor kurzem durch deinen Post zu "rerun make" offensichtlich ist...
Eine "original SAMBA_SMBD" gibt es nicht, wie oben geschrieben nutzt avm einen Ersatz für Samba.

Gismotro, 2021-05-19:
1. OK, dann ist das halt so.
2. Ich rege mich nicht auf, sondern ich wollte Dir nur zu verdeutlich das wenn ich Dir/Euch einen Bug melde, das ich eine zu 100% recoverte Box mit einem frisch ausgescheckten Freetz-NG benutze weil ich genau das von Dir beschriebene Problem ausschließen möchte.

fda77, 2021-05-19:
Von mir aus, ich benutze eh nur nmbd von Samba. Der IPPF Thread ist eh schon "gelöst". Wenn es niemanden mehr gibt den es stört.
Ausserdem weiss ich eh nicht  wie ich das Problem nachvollziehen/nachstellen könnte, das steht auch nicht im Thread

closed 2021-05-20 by Gismotro

fda77, 2021-05-20:
Ein Issue hier im Sinne von "XZ geht nicht" bringt halt nichts, solang das übliche fehlt, wie samba version, ob samba aktiviert ist, ob udevmount aktiv ist, logs davon, ob samab aktiv ist, ob smbd und/oder nmbd usw usf