Freetz-NG/freetz-ng #303: make: *** [make/linux/kernel.mk:186: source/kernel/ref-iks-7390_06.80/.modules-iks] Error 1

dekiesel, 2021-05-18:
Hi,

thank you for providing this project!

I ran into an issue building my config:
```
JUIS: Using cached value ... http://download.avm.de/fritzbox/fritzbox-7390/deutschland/fritz.os/FRITZ.Box_Fon_WLAN_7390-06.86.image ... done.
---> kernel ... modules ... cmd() { PATH="/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin:/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linu
x-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n
\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };     if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build
; fi; cmd -C source/kernel/ref-iks-7390_06.80/linux-2.6.28 CROSS_COMPILE="mips-unknown-linux-gnu-" KERNEL_MAKE_PATH="/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin:/usr/local/
sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" ARCH="mips" INSTALL_HDR_PATH=/home/david/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/linux-dev/2.6.2
8 INSTALL_MOD_PATH="/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80" V=1 modules
building ... make[1]: Entering directory '/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28'
Makefile:222: use: CROSS_COMPILE=mips-unknown-linux-gnu-
rm -f include/config/kernel.release
echo 2.6.28.10 > include/config/kernel.release
set -e; :; mkdir -p include/linux/;     (echo \#define LINUX_VERSION_CODE 132636; echo '#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))';) < /home/david/freetz-ng/source/kernel/ref-iks-7390_06.
80/linux-2.6.28/Makefile > include/linux/version.h.tmp; if [ -r include/linux/version.h ] && cmp -s include/linux/version.h include/linux/version.h.tmp; then rm -f include/linux/version.h.tmp; else :; mv -f i
nclude/linux/version.h.tmp include/linux/version.h; fi
set -e; :; mkdir -p include/linux/;     if [ `echo -n "2.6.28.10" | wc -c ` -gt 64 ]; then echo '"2.6.28.10" exceeds 64 characters' >&2; exit 1; fi; (echo \#define UTS_RELEASE \"2.6.28.10\";) < include/config
/kernel.release > include/linux/utsrelease.h.tmp; if [ -r include/linux/utsrelease.h ] && cmp -s include/linux/utsrelease.h include/linux/utsrelease.h.tmp; then rm -f include/linux/utsrelease.h.tmp; else :; m
v -f include/linux/utsrelease.h.tmp include/linux/utsrelease.h; fi
set -e; if [ -L include/asm ]; then asmlink=`readlink include/asm | cut -d '-' -f 2`; if [ "$asmlink" != "mips" ]; then echo "ERROR: the symlink include/asm points to asm-$asmlink but asm-mips was expected";
echo "       set ARCH or save .config and run 'make mrproper' to fix it"; exit 1; fi; test -e $asmlink || rm include/asm; elif [ -d include/asm ]; then echo "ERROR: include/asm is a directory but a symlink wa
s expected"; exit 1; fi
make -f scripts/Makefile.build obj=scripts/basic
if [ ! -L include/asm ]; then echo '  SYMLINK include/asm -> include/asm-mips'; if [ ! -d include/asm-mips ]; then mkdir -p include/asm-mips; fi; ln -fsn asm-mips include/asm; fi
  SYMLINK include/asm -> include/asm-mips
(cat /dev/null; ) > scripts/basic/modules.order
mkdir -p .tmp_versions ; rm -f .tmp_versions/*
make -f scripts/Makefile.build obj=scripts
make -f scripts/Makefile.build obj=scripts/mod
(cat /dev/null; ) > scripts/mod/modules.order
(cat /dev/null; ) > scripts/modules.order
make -f scripts/Makefile.build obj=.
(cat /dev/null; ) > modules.order
mkdir -p kernel/
mkdir -p arch/mips/kernel/
make -f scripts/Makefile.build obj=. missing-syscalls
  /bin/bash scripts/checksyscalls.sh mips-unknown-linux-gnu-gcc -Wp,-MD,./.missing-syscalls.d  -nostdinc -isystem /home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/lib/gcc/mips-unkno
wn-linux-gnu/4.8.5/include -D__KERNEL__ -Iinclude  -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include -include include/linux/autoconf.h -include /home/david/freetz-ng/sour
ce/kernel/ref-iks-7390_06.80/linux-2.6.28/include/linux/kconfig.h  -DNEW_CONFIG -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -
fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-error=unused-but-set-variable -Os -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -msoft-float -ffreestanding -march=24kc -U_MIPS_ISA -D_M
IPS_ISA=_MIPS_ISA_MIPS32 -Wa,-march=24kc -Wa,--trap -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include/asm/mach-fusiv -mno-branch-likely -ffreestanding -I/home/david/freet
z-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include/asm/mach-generic -D"VMLINUX_LOAD_ADDRESS=0xffffffff80010000" -Wframe-larger-than=1024 -fno-stack-protector -fomit-frame-pointer -Wdeclarati
on-after-statement -Wno-pointer-sign -fwrapv  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(missing_syscalls)"  -D"KBUILD_MODNAME=KBUILD_STR(missing_syscalls)"
<stdin>:1549:2: warning: #warning syscall shmget not implemented [-Wcpp]
<stdin>:1553:2: warning: #warning syscall shmat not implemented [-Wcpp]
<stdin>:1557:2: warning: #warning syscall shmctl not implemented [-Wcpp]
<stdin>:1685:2: warning: #warning syscall semget not implemented [-Wcpp]
<stdin>:1689:2: warning: #warning syscall semop not implemented [-Wcpp]
<stdin>:1693:2: warning: #warning syscall semctl not implemented [-Wcpp]
<stdin>:1697:2: warning: #warning syscall shmdt not implemented [-Wcpp]
<stdin>:1701:2: warning: #warning syscall msgget not implemented [-Wcpp]
<stdin>:1705:2: warning: #warning syscall msgsnd not implemented [-Wcpp]
<stdin>:1709:2: warning: #warning syscall msgrcv not implemented [-Wcpp]
<stdin>:1713:2: warning: #warning syscall msgctl not implemented [-Wcpp]
<stdin>:2145:2: warning: #warning syscall reserved177 not implemented [-Wcpp]
<stdin>:2209:2: warning: #warning syscall reserved193 not implemented [-Wcpp]
<stdin>:2293:2: warning: #warning syscall semtimedop not implemented [-Wcpp]
<stdin>:2441:2: warning: #warning syscall newfstatat not implemented [-Wcpp]
<stdin>:2717:2: warning: #warning syscall shmget not implemented [-Wcpp]
<stdin>:2721:2: warning: #warning syscall shmat not implemented [-Wcpp]
<stdin>:2725:2: warning: #warning syscall shmctl not implemented [-Wcpp]
<stdin>:2853:2: warning: #warning syscall semget not implemented [-Wcpp]
<stdin>:2857:2: warning: #warning syscall semop not implemented [-Wcpp]
<stdin>:2861:2: warning: #warning syscall semctl not implemented [-Wcpp]
<stdin>:2865:2: warning: #warning syscall shmdt not implemented [-Wcpp]
<stdin>:2869:2: warning: #warning syscall msgget not implemented [-Wcpp]
[... 3451 lines cut ...]
FREETZ_AVM_UCLIBC_0_9_33=y
FREETZ_AVM_UCLIBC_NPTL_ENABLED=y
FREETZ_AVM_SOURCE_7390_06_80=y
FREETZ_AVM_SOURCE_ID="7390_06.80"
FREETZ_AVM_SOURCE_FOR_SYSTEM_TYPE_IKS=y
FREETZ_DL_JUIS_BUILDTYPE="1"
FREETZ_DL_JUIS_COUNTRY="049"
FREETZ_DL_JUIS_LANG="de"
FREETZ_DL_JUIS_OEM="avm"
FREETZ_DL_JUIS_FLAG="empty"
FREETZ_DL_JUIS_ANNEX="B"
FREETZ_DL_JUIS_FOS="06.80-42%REV%"
FREETZ_DL_JUIS_OID="0896D7"
FREETZ_DL_JUIS_STRING="Version=${FREETZ_AVM_PROP_MAJOR}.${FREETZ_DL_JUIS_FOS} Serial=${FREETZ_DL_JUIS_OID}%SER% Name=${FREETZ_AVM_PROP_NAME} HW=${FREETZ_AVM_PROP_HWREV} OEM=${FREETZ_DL_JUIS_OEM} Lang=${FREETZ_DL_JUIS_LANG} Annex=${FREETZ_DL_JUIS_ANNEX} Country=${FREETZ_DL_JUIS_COUNTRY} Flag=${FREETZ_DL_JUIS_FLAG} Buildtype=${FREETZ_DL_JUIS_BUILDTYPE} Nonce=%NNC%"
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="7390"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="06_8X"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y
```

I am not firm enough with makefiles to see what the issue is. I hope this is the right place to ask for help.

Thanks!

fda77, 2021-05-19:
Compiling the 7390 kernel works for me. Have you changed some options in kernel-menuconfig?

dekiesel, 2021-05-20:
I cloned the repository and I am now getting the same message as in the other issue so I am closing this one, thank you for your support!

closed 2021-05-20 by dekiesel

fda77, 2021-05-20:
Ah, the other issue is about using kernel-menuconfig :D not everything cold be compiled just by selecting it.

h486qT3W-4wh63RO9-L72Ul4H6, 2022-03-23:
> أهلا،
> 
> شكرا لك على تقديم هذا المشروع!
> 
> واجهت مشكلة في إنشاء التكوين الخاص بي:
> 
> ```
> JUIS: Using cached value ... http://download.avm.de/fritzbox/fritzbox-7390/deutschland/fritz.os/FRITZ.Box_Fon_WLAN_7390-06.86.image ... done.
> ---> kernel ... modules ... cmd() { PATH="/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin:/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linu
> x-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j2  "$@"  || { printf "\n
> \\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };     if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build
> ; fi; cmd -C source/kernel/ref-iks-7390_06.80/linux-2.6.28 CROSS_COMPILE="mips-unknown-linux-gnu-" KERNEL_MAKE_PATH="/home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin:/usr/local/
> sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" ARCH="mips" INSTALL_HDR_PATH=/home/david/freetz-ng/source/toolchain-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/linux-dev/2.6.2
> 8 INSTALL_MOD_PATH="/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80" V=1 modules
> building ... make[1]: Entering directory '/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28'
> Makefile:222: use: CROSS_COMPILE=mips-unknown-linux-gnu-
> rm -f include/config/kernel.release
> echo 2.6.28.10 > include/config/kernel.release
> set -e; :; mkdir -p include/linux/;     (echo \#define LINUX_VERSION_CODE 132636; echo '#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))';) < /home/david/freetz-ng/source/kernel/ref-iks-7390_06.
> 80/linux-2.6.28/Makefile > include/linux/version.h.tmp; if [ -r include/linux/version.h ] && cmp -s include/linux/version.h include/linux/version.h.tmp; then rm -f include/linux/version.h.tmp; else :; mv -f i
> nclude/linux/version.h.tmp include/linux/version.h; fi
> set -e; :; mkdir -p include/linux/;     if [ `echo -n "2.6.28.10" | wc -c ` -gt 64 ]; then echo '"2.6.28.10" exceeds 64 characters' >&2; exit 1; fi; (echo \#define UTS_RELEASE \"2.6.28.10\";) < include/config
> /kernel.release > include/linux/utsrelease.h.tmp; if [ -r include/linux/utsrelease.h ] && cmp -s include/linux/utsrelease.h include/linux/utsrelease.h.tmp; then rm -f include/linux/utsrelease.h.tmp; else :; m
> v -f include/linux/utsrelease.h.tmp include/linux/utsrelease.h; fi
> set -e; if [ -L include/asm ]; then asmlink=`readlink include/asm | cut -d '-' -f 2`; if [ "$asmlink" != "mips" ]; then echo "ERROR: the symlink include/asm points to asm-$asmlink but asm-mips was expected";
> echo "       set ARCH or save .config and run 'make mrproper' to fix it"; exit 1; fi; test -e $asmlink || rm include/asm; elif [ -d include/asm ]; then echo "ERROR: include/asm is a directory but a symlink wa
> s expected"; exit 1; fi
> make -f scripts/Makefile.build obj=scripts/basic
> if [ ! -L include/asm ]; then echo '  SYMLINK include/asm -> include/asm-mips'; if [ ! -d include/asm-mips ]; then mkdir -p include/asm-mips; fi; ln -fsn asm-mips include/asm; fi
>   SYMLINK include/asm -> include/asm-mips
> (cat /dev/null; ) > scripts/basic/modules.order
> mkdir -p .tmp_versions ; rm -f .tmp_versions/*
> make -f scripts/Makefile.build obj=scripts
> make -f scripts/Makefile.build obj=scripts/mod
> (cat /dev/null; ) > scripts/mod/modules.order
> (cat /dev/null; ) > scripts/modules.order
> make -f scripts/Makefile.build obj=.
> (cat /dev/null; ) > modules.order
> mkdir -p kernel/
> mkdir -p arch/mips/kernel/
> make -f scripts/Makefile.build obj=. missing-syscalls
>   /bin/bash scripts/checksyscalls.sh mips-unknown-linux-gnu-gcc -Wp,-MD,./.missing-syscalls.d  -nostdinc -isystem /home/david/freetz-ng/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/lib/gcc/mips-unkno
> wn-linux-gnu/4.8.5/include -D__KERNEL__ -Iinclude  -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include -include include/linux/autoconf.h -include /home/david/freetz-ng/sour
> ce/kernel/ref-iks-7390_06.80/linux-2.6.28/include/linux/kconfig.h  -DNEW_CONFIG -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -
> fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-error=unused-but-set-variable -Os -mabi=32 -G 0 -mno-abicalls -fno-pic -pipe -msoft-float -ffreestanding -march=24kc -U_MIPS_ISA -D_M
> IPS_ISA=_MIPS_ISA_MIPS32 -Wa,-march=24kc -Wa,--trap -I/home/david/freetz-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include/asm/mach-fusiv -mno-branch-likely -ffreestanding -I/home/david/freet
> z-ng/source/kernel/ref-iks-7390_06.80/linux-2.6.28/arch/mips/include/asm/mach-generic -D"VMLINUX_LOAD_ADDRESS=0xffffffff80010000" -Wframe-larger-than=1024 -fno-stack-protector -fomit-frame-pointer -Wdeclarati
> on-after-statement -Wno-pointer-sign -fwrapv  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(missing_syscalls)"  -D"KBUILD_MODNAME=KBUILD_STR(missing_syscalls)"
> <stdin>:1549:2: warning: #warning syscall shmget not implemented [-Wcpp]
> <stdin>:1553:2: warning: #warning syscall shmat not implemented [-Wcpp]
> <stdin>:1557:2: warning: #warning syscall shmctl not implemented [-Wcpp]
[... cut ...] _JUIS_FLAG} Buildtype=${FREETZ_DL_JUIS_BUILDTYPE} Nonce=%NNC%"
> FREETZ_REPLACE_KERNEL_AVAILABLE=y
> FREETZ_REPLACE_MODULE_AVAILABLE=y
> FREETZ_TYPE_PREFIX="7390"
> FREETZ_TYPE_PREFIX_SERIES_SUBDIR="06_8X"
> FREETZ_INSTALL_BASE=y
> FREETZ_REPLACE_BUSYBOX=y
> ```
> 
> أنا لست حازمًا بما يكفي مع makefiles لمعرفة ما هي المشكلة. آمل أن يكون هذا هو المكان المناسب لطلب المساعدة.
> 
> شكرًا ممكن انهاء النتفاقيه ونهاية جميع المشاكل واسترجاع كل البيانات والحسابات العامة والخاصة والتطبيقات وألعاب الانترنت

fda77, 2022-03-26:
Have you changed the .config of the kernel?